Proxmark/proxmark3 #428: MacOS Compilation Fix

ytisf, 2017-10-18:
MacOS fails compilation due to deprecated function call in 'client/pm3_binlib.c' at line 110.
Change from `int i_read = luaL_optint(L,3,1)-1;` to `int i_read = luaL_optinteger(L,3,1)-1;` fixes the issue.

iceman1001, 2017-10-18:
strange,  since luaL_optint is a #define inside lauxlib.h ,  which defines luaL_optinteger...

Does your pm3_binlib.c has a refence to <lauxlib.h> ?   Which it should,  are you linking in another lua-lib version?

ytisf, 2017-10-20:
Yes.
It has. I used the instructions from the Wiki directly.
I guess liblua.5.3.3 changed the reference.

pwpiwi, 2017-10-20:
Proxmark source includes 5.2.2.

ytisf, 2017-10-20:
i guess it just didn't load them and overrode it with current liblua.

pwpiwi, 2017-10-20:
This could happen if CFLAGS is defined in your environment with an -I option to the Lua 5.3.3 includes. I suggest you make a pull request with your fix.

closed 2017-10-20 by pwpiwi