Proxmark/proxmark3 #431: Win compilation warning

merlokk, 2017-10-19:
There are 3 warnings:
g++ -MT obj/proxgui.o -MMD -MP -MF obj/proxgui.Td -I../include -Wall -O3 -std=c++11 -fPIC -I/qt/include -I/qt/include/QtCore -I/qt/include/QtGui -I/qt/include/QtWidgets -c -o obj/proxgui.o proxgui.cpp
proxgui.cpp:1:0: warning: -fPIC ignored for target (all code is position independent)
 //-----------------------------------------------------------------------------
 ^

is this -fPIC needs for targets other than windows?

merlokk, 2017-10-19:
https://stackoverflow.com/questions/5311515/gcc-fpic-option

pwpiwi, 2017-10-19:
Yes, see issue #270.

merlokk, 2017-10-19:
OK, then windows fixed in #432

pwpiwi, 2017-10-19:
Issue is not related to Operating System. If Qt is compiled with PIC, we need -fPIC.

merlokk, 2017-10-19:
so Gator's proxspace compiled without fPIC. And I need to check how Qt was compiled?

merlokk, 2017-10-19:
https://stackoverflow.com/questions/966960/what-does-fpic-mean-when-building-a-shared-library

merlokk, 2017-10-19:
I cant see how to check Qt fPIC option. Instead of that we can check OS or  maybe Gator's proxspace.
Do you know another way?

pwpiwi, 2017-10-19:
The only way I see is to live with the warning.

merlokk, 2017-10-19:
OK

closed 2017-10-19 by merlokk