Proxmark/proxmark3 #437: master compile error

merlokk, 2017-10-20:
https://github.com/Proxmark/proxmark3/commit/6e49717b5e025b6ac7139b00f2f39d3b22b424a1#diff-5d63a2f49fb45c3e82f01bcfa3621f95R188

so14443a.c:188:13: error: static declaration of 'iso14a_set_timeout' follows non-static declaration
static void iso14a_set_timeout(uint32_t timeout) {
^
In file included from iso14443a.c:13:0:
iso14443a.h:55:13: note: previous declaration of 'iso14a_set_timeout' was here
extern void iso14a_set_timeout(uint32_t timeout);
^
make[1]: *** [obj/iso14443a.o] Error 1
make[1]: Leaving directory `/pm3/armsrc'
make: *** [armsrc/all] Error 2

merlokk, 2017-10-20:
i have deleted the `static`

merlokk, 2017-10-20:
https://ci.appveyor.com/project/merlokk/proxmark3/build/3.0.1.56

pwpiwi, 2017-10-20:
Conflicting commits. Fixed.

closed 2017-10-20 by pwpiwi