Proxmark/proxmark3 #441: Mifare c* commands broken for gen1b cards

Fl0-0, 2017-10-24:
Gen1b c* commands does not work anymore:

```
proxmark3> hf mf csave 4
#db# halt error. response len: 1          
No chinese magic backdoor command detected          
Saving magic mifare 4K          
#db# wupC2 error          
Cant get block: 0          
#db# wupC2 error          
Cant get block: 0          
Saved to file: dump.eml          
proxmark3>
```

Fl0-0, 2017-10-24:
OK i identified the root cause:

In commit c04a4b60cefc545641582da165c42b3023f5f753: Don't do iso14443-4 select for mifare emulations on processor cards:

Not doing the iso14443-4 select for magic Gen1b cards does break the generation detection in mfCIdentify():

```
        // iso14a_card_select_t card;
	// memcpy(&card, (iso14a_card_select_t *)resp.d.asBytes, sizeof(iso14a_card_select_t));

	// uint64_t select_status = resp.arg[0];		// 0: couldn't read, 1: OK, with ATS, 2: OK, no ATS, 3: proprietary Anticollision

	// if(select_status != 0) {
		// uint8_t rats[] = { 0xE0, 0x80 }; // FSDI=8 (FSD=256), CID=0
		// c.arg[0] = ISO14A_RAW | ISO14A_APPEND_CRC | ISO14A_NO_DISCONNECT;
		// c.arg[1] = 2;
		// c.arg[2] = 0;
		// memcpy(c.d.asBytes, rats, 2);
		// SendCommand(&c);
		// WaitForResponse(CMD_ACK,&resp);
// }
```

merlokk, 2017-10-24:
also
this works
```
proxmark3> hf 14a reader
#db# ISO14443A Timeout set to 1060 (10ms)
 UID : 01 02 03 04
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands (GEN 1a): YES
```
this works with error
```
proxmark3> hf mf cgetblk 0
#db# ISO14443A Timeout set to 1060 (10ms)
No chinese magic backdoor command detected
--block number: 0
#db# ISO14443A Timeout set to 1060 (10ms)
block data:01 02 03 04 04 08 04 00 00 00 00 00 00 00 be af
```
this not woks
```
proxmark3> hf mf csetblk 1 01020304050607080910111213141516
#db# ISO14443A Timeout set to 1060 (10ms)
No chinese magic backdoor command detected
```

merlokk, 2017-10-24:
it seems that 
void MifareCIdent() from mifarecmd.c:1502
dont initialize FPGA and dont close FPGA
maybe it needs to add there initialization and etc parts? as in the other commands?

merlokk, 2017-10-24:
like that?
```
	LED_A_ON();
	LED_B_OFF();
	LED_C_OFF();
	iso14443a_setup(FPGA_HF_ISO14443A_READER_LISTEN);

	clear_trace();
	set_tracing(true);

------------code here-----------------------

	LED_B_ON();
	cmd_send(CMD_ACK,isOK,0,0,data,18);
	LED_B_OFF();

	FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);
	LEDsoff();
	}

```

Fl0-0, 2017-10-24:
I confirm @merlokk this behavior for gen1a cards. **But for gen1a csave/csave commands works** ...
Maybe it is related to FPGA initialization. 

Fixing by uncommenting changes from c04a4b6 in MifareCIdent() actually works.
Maybe it a 'quick win' fix and the underlying problem needs to be more investigated.

@merlokk could you test a fix with your FPGA init/close stuff ?

merlokk, 2017-10-24:
Ill check and publish PR. it will need to deep check...

merlokk, 2017-10-24:
It looks that some procedures go MifareCIdent() with field on and some with field off(
as i see with my changes all procedures works

merlokk, 2017-10-24:
needs to test........

merlokk, 2017-10-24:
it seems all is OK. But I dont know...   If MifareCIdent uses from scripts maybe it needs to fix or it needs to uncomments this
```
//	FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);
//	SpinDelay(100);
```

and thats all)

Fl0-0, 2017-10-24:
OK @merlokk i can do a full run of testing and validating all hf mf c* commands for both gen1a and gen1b cards tomorrow (CEST time).

Fl0-0, 2017-10-24:
I've done the tests (found the motivation to do it tonight :-) ).
Gen1a and gen1b cards c* commands works **with and without** the deactivation of the  induction field: 

```
//      FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);
//	SpinDelay(100);
```

Well done @merlokk

PR #444 (2017-10-25): identification fix for magic chinese cards
fixed #441

needs to deep check
closed 2017-10-25 by pwpiwi