Proxmark/proxmark3 #447: 14a apdu

merlokk, 2017-10-26:
I look into 14443areader and see that it have APDU exchange option.
I think about make command with APDU exchange and TLV decode result
I think it needs to have a new command because `hf 14a raw` is overloaded with functions
What do you think?
`hf 14a apdu` or `hf emv apdu`?

merlokk, 2017-10-26:
And maybe as for emv i can code PPSE and SELECT stage.

merlokk, 2017-10-26:
TLV maybe from here
https://github.com/lumag/emv-tools/blob/master/lib/tlv.c
and PSE and SELECT too)

pwpiwi, 2017-10-26:
Maybe something like a `hf 14a cmd` with options like `select`, `RATS`, `PPS`, etc.? But for emv I would suggest a separate command. Afaik, there is some emv code in icemans fork. 

TLV on the other hand is not only emv. I had hacked some TLV code in cmdhftopaz.c - which you can replace if you provide more generic tlv code.

merlokk, 2017-10-26:
Ill plain to get some portions of code from @iceman1001 fork. But it have not so general commands so I want)

iceman1001, 2017-10-26:
There is tons of EMV code from @peterfillmore forks (and branches)   sadly a mess.
First its on device,   I belived it can benefit from being on client side instead (see topaz for instance)
Second,  it doesn't have a smooth commandset like @merlokk  noticed too.

A generic TLV and EMV would be welcomed with warm arms.

I wouldn't merge EMV into 14a commands. That will be contraproductive.
I suggested a  "hf 14a info"  or  "hf mf info"  command,  where more detalied information is shown and where we could have more tests.   Instead of putting it into "14a reader".

I agree with that "14 raw"  command has too much already.

merlokk, 2017-10-27:
OK
1. int CmdHF14AReader(const char *Cmd) i will rename to `hf 14a info`
2. I will add `hf 14a reader` with anticollision only
3. Will add 'hf 14a apdu' with options to `select` and `drop fileld`
4. I add commandset `emv` with commands `hf emv apdu` (with TLV decoder), `hf emv pse`, `hf emv scan` (to scan known application IDs), `hf emv select`

and I break up it to several PR

Is this OK?

iceman1001, 2017-10-27:
Sounds like a good plan

merlokk, 2017-10-27:
about EMV: I want to use apdu command and write all the code on client side.
so maybe i spend some time to debug (and create in client) apdu command and then use it

iceman1001, 2017-10-27:
Sounds good.  There is a apdu command on device side,  I think you can call it with a flag to "raw" (might be codewise and not in client)  so you don't have to think too much about that part.
The most part is on client side,  look at all stuff peterfillmore did.  I kind of merge his stuff into my fork.  So you don't have to look two of his branches...  

Keep the best parts and make the rest great!  :)

merlokk, 2017-10-30:
I see that ISO14A_APDU its not the APDU that i need. There is something like iso7816 T=1, but I need T=CL.  And my cards cant answers to this apdu, they sends "68 81 - Logical channel not supported "
So it seems that it needs to send apdu in raw mode.

merlokk, 2017-10-31:
to work with start 14443 sequence `0x0a 0x00` (CID=0x00) we need to set CID via ATTRIB command. Will be easier to have `0x02` sequence without CID. Now it works. As I see now we dont need 14443 command chaining. So APDU command just send and receive I-Block. (`0x02 APDU 14443CRC`)

merlokk, 2017-11-02:
Part 3 added #461

merlokk, 2017-11-10:
done #465

closed 2017-11-10 by merlokk