Proxmark/proxmark3 #453: Debricking the PM3 RDV2 with BP3.5/6 on Win7/64 without any Dev tools or coding ???

jenningsreeve, 2017-10-28:
Hi all

I bricked my brand new Proxmark3 RDV2 by installing a win gui firmare only to find out it didn't work and that the repo was "bleeding edge and highly experimental". OoooucH

Rather than going 101 on the whole build scenario, I decided to explore using binaries option.
I tried numerous firmare to no end, there was always something missing from the alleged fix.

After many late nights I have worked it out and my baby is up and running. WooP WooP
If anyone is in the same position, I would be happy to share the procedure, just let me know.

Glenn

iceman1001, 2017-10-28:
...I love the sarcasm,  still how do you recover a bricked pm3 without a device like segger, buspirate?  
Feel free to update the wiki with your findings so that the community can benifit from it. There is a section of "de-bricking" already.

exploitagency, 2017-10-28:
@iceman1001 at what commit or date was there an issue?  I want to make sure I am not distributing a binary with issues, I usually try to watch until things seem stable before compiling.  But lately I have been busy with other projects mostly.  I really hate to see these people talk to you like that, they don't realize your one of top contributors to this repo as well.

jenningsreeve, 2017-10-28:
iceman1001
Sorry Iceman but it's not sarcasm, it's pure frustration.  I did not aim that at you honestly.
I realise I tried to run, and fell on my ass!. I will try walking for a bit.
If you read the header it says "with BP3.5/6".  Easy missed.
Believe me, I tried everything before buying the BP, even tried to recover it by forcing SAM-BA mode with the TST pin.
I think most have already recovered, I just posted this for any other code dodgers like me.
Cheers

exploitagency, 2017-10-28:
I use my BusPirate all the time, its a good investment.  Get the seeed studio cable with the clips on it, I think that is the main reason I reach for it when I need a certain tool.  You will surely use it again.  When I get time I will look through the commit history and make sure I am not contributing to an angry user base.

jenningsreeve, 2017-10-28:
exploitagency
Thanks for the info, I already had ribbon with clips both ends.  It was the micro surgery soldering the JTAG pins that was scary. The commit has been rectified very quickly, just bad timing on my part.  I only got PM3 for playing with RFID but I now know the PM3 schematics and boards of by heart.
Cheers

pwpiwi, 2017-10-28:
@jenningsreeve: I am interested how you succeeded with debricking. Can you add something to the Wiki or explain it here?

iceman1001, 2017-10-28:
@exploitagency   I suggest you delete all from 1st of september until 26th october to be certain. 
As of now its working, but the FPGA_HF.bit with the new FeliCa support is not the best yet but it doesnt brick :)  All HF functionality needs to be verified and I don't have time for it.

@jenningsreeve  BP in the title,  I suggest you write it out in clear text since its really not obvious which device you tried to use.  Since the wiki has instructions on how to use the buspirate to recover a bricked pm3 I can only see the sarcasm in this issue.  Frustration posts like this are better suited on the PM3 forum instead of here.   With that said,  the binary distributions by others like for instance @exploitagency  and @Gator96100  don't warn about that my fork is experimental,   all those who clone my repo can see all warnings and at least decide to use it on their own risk.  I am truly sorry for your troubles and since it gives you frustration, its better that you don't use iceman fork. I don't want you to be frustrated.

https://github.com/Proxmark/proxmark3/wiki/Debricking-Proxmark3

jenningsreeve, 2017-10-29:
https://github.com/Proxmark/proxmark3/wiki/Debricking-Proxmark3
This procedure does not work if you have no development evironment or knowledge.
I did it without compiling anything

jenningsreeve, 2017-10-29:
By that I mean there are no pre-built openocd binaries with buspirate support that have proxmark3 512K ability

exploitagency, 2017-10-29:
I just read the second part, I will include a warning now, I don't think I distributed any of the bricker binaries though, I usually test them before pushing just not this last time.

jenningsreeve, 2017-10-29:
I have a skeleton set that has only the absolute minimum progs and files required to debrick a 512K.
Buspirate 6.1 update and TDMI vcp drivers
Openocd built for Windows 7 with Buspirate support
Replacement Buspirate.cfg and at91sam7s512.cfg files that pull it all together.
No installing, no compiling, no building

jenningsreeve, 2017-10-29:
I will close now and add the info to wiki
Thanks to all for your assistance

closed 2017-10-29 by jenningsreeve