Proxmark/proxmark3 #454: lf indala is not working anymore

phiberx, 2017-10-29:
It was working welll back in the 082717 release.

now it's not working properly, tried a lf clone and then lf sea.
nothing comes up.

marshmellow42, 2017-10-29:
Please Explain

marshmellow42, 2017-10-29:
If you are using Iceman's fork then post your issue there.

closed 2017-10-29 by marshmellow42

phiberx, 2017-10-29:
hi, i believe i know what branch i am using and what i am reporting.. :)
It seems that both iceman and master branch are sharing same code base:

pm3 ~/client$ proxmark3 com5
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-89-g5191ad9-dirty-suspect 2017-10-28 17:35:01
os: master/v3.0.1-72-g14d7657-suspect 2017-08-27 07:20:25
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13

uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 197233 bytes (75%). Free: 64911 bytes (25%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> lf ind clone 40001345a
Cloning 64bit tag with UID 40001345a
#db# DONE!
proxmark3> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

proxmark3> lf read
#db# LF Sampling config:
#db#   [q] divisor:           95
#db#   [b] bps:               8
#db#   [d] decimation:        1
#db#   [a] averaging:         1
#db#   [t] trigger threshold: 0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 59 89 59 89 57 88 57 88 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
proxmark3> lf ind altdemod
Recovered 1221 raw bits, expected: 1249
worst metric (0=best..7=worst): 7 at pos 18
UID=0000000000000000000000000000001000000000000000001001101000101101 (200009a2d)
Occurrences: 19 (expected 19)
proxmark3>

marshmellow42, 2017-10-29:
Your clone values are not valid for indala.  Therefore the search will not result in a valid indala tag being found.

marshmellow42, 2017-10-29:
The old indala demod (altdemod) did not properly adhere to the indala format so some tags could be miss identified.