Proxmark/proxmark3 #455: LF indala clone and LF search does not detect indala tags

phiberx, 2017-10-29:
This is not iceman fork

pm3 ~/client$ proxmark3 com5
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-89-g5191ad9-dirty-suspect 2017-10-28 17:35:01
os: master/v3.0.1-72-g14d7657-suspect 2017-08-27 07:20:25
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13

uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 197233 bytes (75%). Free: 64911 bytes (25%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> lf ind clone 62143be57
Cloning 64bit tag with UID 62143be57
#db# DONE!
proxmark3> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

proxmark3> lf read
#db# LF Sampling config:
#db#   [q] divisor:           95
#db#   [b] bps:               8
#db#   [d] decimation:        1
#db#   [a] averaging:         1
#db#   [t] trigger threshold: 0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 5d 84 5c 83 5c 84 5b 83 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
proxmark3> lf indala demod
proxmark3>

marshmellow42, 2017-10-29:
Do not double post issues.

iceman1001, 2017-10-29:
since you closed the first issue I can see why he open a new one...

marshmellow42, 2017-10-29:
He can reopen his own items if he feels it is not resolved.  But as I explained on the other issue the commands are functioning as designed.

iceman1001, 2017-10-29:
You sure did.

phiberx, 2017-10-30:
sorry, i got theses values off the forums to test. i'll close this and that since it's functioning as designed.

closed 2017-10-30 by phiberx