Proxmark/proxmark3 #464: You've been removed from the Proxmark team Contributor

iZsh, 2017-11-09:
Hello,

I received an automated email telling me I've been removed from the proxmark team contributor (i.e., I lost write access it seems). Why? Who?

I know I haven't contributed for quite a while, but still, considering I rewrote most of the code couple of years ago, I would have expected at least a heads-up email or something.

Regards,
iZsh

W8M2Hg9lLmWqXSGC, 2017-11-09:
Hi iZsh,

Sorry. Nothing personal. Was simply clearing out inactives.

Regards,
0xFFFF

iZsh, 2017-11-10:
np

Regards,
iZsh

closed 2017-11-10 by iZsh

iceman1001, 2017-11-10:
Well,  we still do need active maintainers and the need to clean out inactives is a normal administrative process.   Lets see,  we would ask all maintainers if they still want be involved but we would need them to become more active. 

The proxmark3 repo has need of more contributing, testing of PRs,  so the need for more active maintainers is real.