Proxmark/proxmark3 #467: Discussion on Options and Parameters in Proxmark Commands

pwpiwi, 2017-11-10:
@iceman1001 wrote in #465:
>Pretty soon the discussion of a unified parameter is needed, since you seem to be fond of the prefix "-"
>while we have spend time not using it. I don't mind either, but I prefer that we have one and not three like >we have to day.
>
>no prefixed inparameters.
>letter only prefixed parameters
>"-" + letter prefixed parameters.
>Same goes for inputed "hex".. Either "405060" or "40 50 60" or both accepted.

This is indeed a topic I wanted to discuss ages ago. The proxmark way of handling parameters and options is neither consistent nor common practice. We are all used to options like `-h` and an alternative long version like `--help` for any Unix or Windows program. Why don't we adopt this common practice and start using `getopt()`  (or its sister `getopt_long()`)  as in `client\loclass\main.c` and `client\reveng\cli.c`? This is not only common practice but also avoids those ugly command line parsing we have in many places.

iceman1001, 2017-11-10:
I welcome this.

LUA script also uses "-" + letter prefix
All different command line parsing is ineffective.

I just one way,  not three.  And in a standard way so its easy for new contributors to add.  Not to worry or spend time in "how to parse commandline in proxmark".

marshmellow42, 2017-11-10:
i'm for it.  ( though i will have to learn how getopt() works.. ;)   )

iceman1001, 2017-11-10:
yeah,  getopt()  crash-course needed for me too.

pwpiwi, 2017-11-10:
https://www.gnu.org/software/libc/manual/html_node/Getopt.html#Getopt

iceman1001, 2017-11-10:
Actually,   all blogs,  forum posts will be upset but I still want it to be unified.

iceman1001, 2017-11-10:
...well... piwi...  will you start the unifying with one getopt?   The loclass, reveng has their own,  we would need to have one.

merlokk, 2017-11-10:
So, it needs to use `getopt_long`?

and this? 
"-" + letter prefixed parameters. (dont know... lets think)
Same goes for imputed "hex"..  "405060" and "40 50 60" accepted. (it will be cool because some strings in the internet with spaces, some without....  just for more convenient copy-paste)

iceman1001, 2017-11-10:
getopt_long,  whats the difference?

if we have [-h] ,  which could be confusing between being in the shell vs inside pm3 client.. What you think?

Agreed,  hex input both kinds,  with/ wo spaces,  would increase usability.

merlokk, 2017-11-11:
I don't know what the difference between getopt libs. Ill look.
We need functionality like:
SelectField=OptionExists("s") ;
GetOptionParam("a", APDU, & APDUlen)
GetLastParam(APDU, & APDUlen)

merlokk, 2017-11-11:
I have time to investigate at first part of next week. If someone can do earlier, please, do)

iceman1001, 2017-11-11:
I'm guessing,
[code]
getopt ==  "-h"
getopt_long ==  "--help"
[/code]

merlokk, 2017-11-13:
https://github.com/skeeto/optparse
https://github.com/cofyc/argparse

need too look - maybe there is cool parser on github)

pwpiwi, 2017-11-14:
skeeto makes some good points in his README.md (https://github.com/skeeto/optparse/blob/master/README.me) which discourage the usage of the original `getopt()`. cofyc prevents the same drawbacks and adds error message handling. Both licenses are compatible with GPL v2.

+1 for cofyc/argparse

merlokk, 2017-11-14:
OK,
updated list)
https://github.com/skeeto/optparse
https://github.com/cofyc/argparse
https://github.com/hroptatyr/yuck (need to look deeper)
https://github.com/argtable/argtable3 (i like it, but it needs to look)
https://github.com/jpommerening/choice - this simple with config from struct

merlokk, 2017-11-14:
and all of them dont have initialization from `char*`, just from argc arcv

pwpiwi, 2017-11-14:
Yes, a function to split the line into options and parameters is required in all cases. It would be an additipnal task of this function to distinguish a hex parameter with spaces from multiple parameters.

merlokk, 2017-11-14:
If we have `-` at the parameter's beginning - all is ok, this function can be written easily)
If not - needs to think.

guys, what are you thinking about parameters: `-` or without?

pwpiwi, 2017-11-14:
I am for the standard: - or -- for options. = or nothing for parameters. Nothing for additional arguments.

HF 14a raw -c -t 10 60 01

HF 14a raw : command
-c : option
-t: option with mandatory parameter
10: parameter
60 01 : arguments

pwpiwi, 2017-11-16:
Looks great. Has a good tutorial as well: http://www.argtable.org/tutorial/

I propose to start with `hf 14a raw`. It should be possible to refactor it without changing the syntax.

merlokk, 2017-11-17:
OK, i will try and show PR) lets look...

merlokk, 2017-11-26:
OK. lets think, test and merge.

especially needs to check if `reveng` part still works OK.

merlokk, 2018-09-28:
can be closed

closed 2018-09-28 by pwpiwi