Proxmark/proxmark3 #468: hf 14a apdu --

iceman1001, 2017-11-10:
@merlokk 

hf 14a apdu -s 0200A4040007A0000000031010

--> GetAPDUCode (sw1=104 'h', sw2=129 '\201') at emv/apduinfo.c:287  <--

(gdb) bt
#0  0xb7fdac31 in __kernel_vsyscall ()
#1  0xb6dc5ea9 in __GI_raise (sig=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0xb6dc7407 in __GI_abort () at abort.c:89
#3  0xb6e0137c in __libc_message (do_abort=2, fmt=0xb6ef8267 "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:175
#4  0xb6e916b8 in __GI___fortify_fail (msg=<optimized out>) at fortify_fail.c:37
#5  0xb6e8f8a8 in __GI___chk_fail () at chk_fail.c:28
#6  0xb6e8f088 in _IO_str_chk_overflow (fp=0xb05fe630, c=49) at vsprintf_chk.c:31
#7  0xb6e052d4 in __GI__IO_default_xsputn (f=0xb05fe630, data=0xb05fe5fe, n=2) at genops.c:455
#8  0xb6ddcc59 in _IO_vfprintf_internal (s=0xb05fe630, format=<optimized out>, ap=0xb05fe734 "") at vfprintf.c:1631
#9  0xb6e8f11d in ___vsprintf_chk (s=0xb05fe75a "8", flags=1, slen=2, format=0x81472d2 "%02X ", args=0xb05fe730 "\201") at vsprintf_chk.c:82
#10 0xb6e8f070 in ___sprintf_chk (s=0xb05fe75a "8", flags=1, slen=2, format=0x81472d2 "%02X ") at sprintf_chk.c:31
#11 0x08073687 in sprintf (__fmt=0x81472d2 "%02X ", __s=0xb05fe75a "8") at /usr/include/i386-linux-gnu/bits/stdio2.h:33
#12 GetAPDUCode (sw1=104 'h', sw2=129 '\201') at emv/apduinfo.c:287
#13 0x08073824 in GetAPDUCodeDescription (sw1=104 'h', sw2=129 '\201') at emv/apduinfo.c:313
#14 0x0807cdf1 in CmdHF14AAPDU (cmd=0xafc22934 "-s 0200A4040007A0000000031010") at cmdhf14a.c:733
#15 0x080d133a in CmdsParse (Commands=0x81a6160 <CommandTable>, Cmd=<optimized out>) at cmdparser.c:66
#16 0x0807cf19 in CmdHF14A (Cmd=0xafc2292f "apdu -s 0200A4040007A0000000031010") at cmdhf14a.c:942
#17 0x080d133a in CmdsParse (Commands=0x81a6040 <CommandTable>, Cmd=<optimized out>) at cmdparser.c:66
#18 0x0807a889 in CmdHF (Cmd=0xafc2292b "14a apdu -s 0200A4040007A0000000031010") at cmdhf.c:1092
#19 0x080d133a in CmdsParse (Commands=0x81a7a80 <CommandTable>, Cmd=<optimized out>) at cmdparser.c:66
#20 0x080d1b41 in CommandReceived (Cmd=0xafc22928 "hf 14a apdu -s 0200A4040007A0000000031010") at cmdmain.c:188

iceman1001, 2017-11-10:
This char buffer is too small..
https://github.com/Proxmark/proxmark3/blob/master/client/emv/apduinfo.c#L281

pwpiwi, 2017-11-10:
Indeed. Did coverty find this?

iceman1001, 2017-11-10:
nop.  That would be me. Wanting to test APDU and managed to trigger it when on a linux box which has gdb :)
I will need to send these into coverty after the other PR gets merged and I mangle it into my fork.    What a time to be alive in the proxmark world.  Emv, Apdu, felica,  ...

merlokk, 2017-11-10:
strange. there are two uint8: (uint8_t sw1, uint8_t sw2)
i can raise size of buffer, but 2 uint8 in hex = 4 symbols.

iceman1001, 2017-11-10:
and you have spaces inside? and a \0 string terminator`?

merlokk, 2017-11-10:
maybe space for string terminator. yes, thanks

closed 2017-11-10 by iceman1001

Giorgamose, 2025-03-12:
Can you please write me a command to emulate URL on pm3

commit 77499f6f (2017-11-10): fix #468
commit c425cc7e (2017-11-10): Merge pull request #472 from merlokk/master

fix #468