Proxmark/proxmark3 #469: pm3 lf interference?

marshmellow42, 2017-11-10:
for some time i've noticed some sort of residual interference on the lf side.  maybe my device is dying but i don't think so (checked on 2 pm3's)

when reading with no card on the antenna there is a pattern that is clear: 
[image]

with weaker antennas it is not as clear (but still present), and with this 40+v antenna it is pretty strong.
(making the checks for "Just Noise" fail)

to reproduce i just run `lf read` - `data plot` with no tag on the antenna.

when i get time i will attempt to go back in time and identify where it first appears, because i know i didn't have this problem a few years ago...

iceman1001, 2017-11-10:
I don't have that,   but then I'm also experimenting with a better noise-detection since the current one is a bit limited.

marshmellow42, 2017-11-10:
what does a lf plot look like for you with no tag near the antenna?

marshmellow42, 2017-11-10:
with a weaker antenna i see:
[image]

marshmellow42, 2017-11-10:
strong antenna's tune:
[image]

weaker antenna's tune:
[image]

pwpiwi, 2017-11-10:
My noise looks more random:
[image]

antenna tune:
```
proxmark3> hw tune

Measuring antenna characteristics, please wait.........
# LF antenna: 41.11 V @   125.00 kHz
# LF antenna: 42.62 V @   134.00 kHz
# LF optimal: 45.10 V @   127.66 kHz
# HF antenna:  0.28 V @    13.56 MHz
# Your HF antenna is unusable.
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

proxmark3>
```

Are you picking up some external signal? What if you are moving (far away)?

iceman1001, 2017-11-10:
Mine goes between -2 and -10  with an amplitude of 8.  

[image]

```
Measuring antenna characteristics, please wait......          
# LF antenna: 18,43 V @   125.00 kHz          
# LF antenna: 16,23 V @   134.00 kHz          
# LF optimal: 18,84 V @   126,32 kHz          
# HF antenna: 13,58 V @    13.56 MHz          
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

```

iceman1001, 2017-11-10:
it could be bad antenna cable,  bad capacitors, bad connectors,
since its you, I guess you already test it

marshmellow42, 2017-11-10:
I went to different rooms, changed cables, pm3s.  Only common element is ms surface pro...    I'll try a different computer soon

merlokk, 2017-11-11:
Have you checked the power of proxmark and maybe power near antenna amplifier?

marshmellow42, 2017-11-12:
well it appears the interference is location dependent as in a different building i get:
[image]

so evidently the hvac or other large electronic device in the other building produces a patterned signal interference and a field large enough to be consistent throughout the building... 

sorry for wasting your time...

closed 2017-11-12 by marshmellow42