Proxmark/proxmark3 #476: mifare 1k return nak 0x01.

maxben14, 2017-11-11:
I'm doing an emulator on my developed device. I try to test my emulator on a proxmark. My proxmark returns nak when reading the block of the card.
I put my device on proxmark and I try in proxmark read block.
proxmark3> hf mf rdbl 0 a a0a1a2a3a4a5
--block no:0, key type:A, key:a0 a1 a2 a3 a4 a5
#db# Cmd Error: 09
#db# Read block error
isOk:00
proxmark3> hf list 14a
Recorded Activity (TraceLen = 156 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       4596 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL       
      10676 |      16564 | Tag | 01  02  03  04  04                                              |     |
      19072 |      29600 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
      30788 |      34308 | Tag | 08  b6  dd                                                      |     |
      35968 |      40672 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
     110420 |     115092 | Tag | 00  00  00  00                                                  |     |
     124800 |     134112 | Rdr | c1 f4!  5f  1b 25! a0! 69! 61!                                  | !crc| DEC(244)       
     312932 |     317604 | Tag | 62  92  09  6c                                                  |     |
     323584 |     328352 | Rdr |5e! 2a!  b2 a3!                                                  | !crc| ?
     376644 |     376836 | Tag | 01

I correct understand that answer from Tag(my device - emulator 1k) 01 is parity or CRC error ?
And what mean Cmd Error: 09 ?
I show this with mfkey64
mfkey64 01020304 00000000 c1f45f1b 25a06961 6292096c 5e2ab2a3 01
MIFARE Classic key recovery - based on 64 bits of keystream
Recover key from only one complete authentication!

Recovering key for:
   uid: 01020304
    nt: 00000000
  {nr}: c1f45f1b
  {ar}: 25a06961
  {at}: 6292096c
{enc0}: 5e2ab2a3
{enc1}: 01

LFSR successors of the tag challenge:
  nt' : 00000000
  nt'': 00000000
Time spent in lfsr_recovery64(): 0.18 seconds

Keystream used to generate {ar} and {at}:
   ks2: 25a06961
   ks3: 6292096c

Decrypted communication:
{dec0}: 300002a8
{dec1}: 89

Found Key: [a0a1a2a3a4a5]

My device successfully received and decrypted the read command and sends a 16 byte response with the crc and parity bits. But how me see what data get proxmark from my device ?

iceman1001, 2017-11-11:
... since @pwpiwi  's updates,  the access rights are now enforced on emulator. 
Which sectortrailor did you use?

maxben14, 2017-11-11:
I use only key 1k in my code device.

merlokk, 2017-11-11:
`hf list 14a` must record 16 bytes you have sent. So there is a problem in transmitting. Maybe timings or something like that.
Also, to be sure you can sniff communication between some reader (ACR or like that) and your emulator with proxmark and look deeper

merlokk, 2017-11-11:
As I see proxmark received only 1 byte, but you send 16?
P.S. `01` looks strange, because it comes with good parity. look at previous line...

iceman1001, 2017-11-11:
.....your emulator device is sending  0x09 (decrypted)  | 0x01 (encrypted)...
so your emulator has some kind of problem.   This is not a PM3 issue.

maxben14, 2017-11-11:
@merlokk , i try sniff communication between my device and acr122 with proxmark3.
Now i get 16 byte 01 or 00 from acr122.
i try in acr122 script tools command:
[1]                  > ff 00 00 00 0f d4 40 01 60 00 a0 a1 a2 a3 a4 a5 01 02 03 04 
                     < D5 41 00 90 00 

[2]                  > ff 00 00 00 05 d4 40 01 30 00
                     < D5 41 00 90 00 
And log from proxmark
proxmark3> hf 14a snoop
#db# cancelled by button
#db# COMMAND FINISHED
#db# maxDataLen=3, Uart.state=0, Uart.len=0
#db# traceLen=256, Uart.output[0]=00000092
proxmark3> hf list 14a
Recorded Activity (TraceLen = 256 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |       1056 | Rdr | 26                                                              |     | REQA
       2228 |       4596 | Tag | 04  00                                                          |     |
     346080 |     348544 | Rdr | 93  20                                                          |     | ANTICOLL       
     349716 |     355604 | Tag | 01  02  03  04  04                                              |     |
     845888 |     856416 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
     857588 |     861108 | Tag | 08  b6  dd                                                      |     |
    1162000 |    1166704 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
    1236452 |    1241124 | Tag | 00  00  00  00                                                  |     |
    1242528 |    1251840 | Rdr |14!  24 fe!  06  b2  ca  90  00                                  | !crc| ?
    1427844 |    1432516 | Tag | a0  63  20  a0                                                  |     |
    1673888 |    1678656 | Rdr |92! f2! 0e! 47!                                                  | !crc| ?
    1728836 |    1729028 | Tag | 01                                                              |     |
    1738756 |    1738948 | Tag | 01                                                              |     |
    1748788 |    1748980 | Tag | 01                                                              |     |
    1769156 |    1769348 | Tag | 01                                                              |     |
    1797124 |    1797316 | Tag | 01                                                              |     |
    1806516 |    1806708 | Tag | 01                                                              |     |
    1835076 |    1835268 | Tag | 01                                                              |     |
    1855140 |    1855332 | Tag | 01                                                              |     |
    1865044 |    1865300 | Tag |00!                                                              |     |
    1874884 |    1875140 | Tag |00!                                                              |     |
    1884564 |    1884756 | Tag | 01

maxben14, 2017-11-11:
@iceman1001 , my device send 16 byte this
uint8_t block0[] = { 0x04, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0a, 0x04, 0x00, 0x08, 0x00, 0x00 };
In my massive haven't 09 byte.

merlokk, 2017-11-11:
There is a sample how it needs to be
https://github.com/Proxmark/proxmark3/wiki/Mifare-Tag-Ops

here reader asked yor emulator
`1673888 | 1678656 | Rdr |92! f2! 0e! 47! | !crc| ?`
and it must response with 18 bytes, but it responses with strange single bytes with pauses.
you can calc pause: 1738756 - 1729028 * 7.4*10-8 - it too big

so there are 2 possibilities:
1. your emulator sent this single bytes with pauses
2. your emulator sent all 18 bytes, but with error in frequency or modulation

merlokk, 2017-11-11:
P.S. try to send 4 bytes and look on proxmark side. maybe you have buffer overflow

maxben14, 2017-11-11:
@merlokk , i try send only 4 bytes on comand 0x30. Now error code 0x08.
proxmark3> hf mf rdbl 0 a a0a1a2a3a4a5
--block no:0, key type:A, key:a0 a1 a2 a3 a4 a5
#db# Cmd Error: 08
#db# Read block error
#db# READ BLOCK FINISHED
isOk:00
proxmark3> hf list 14a
Recorded Activity (TraceLen = 156 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2212 |       4580 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL       
      10660 |      16548 | Tag | 01  02  03  04  04                                              |     |
      19072 |      29600 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
      30756 |      34276 | Tag | 08  b6  dd                                                      |     |
      35968 |      40672 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
     110420 |     115092 | Tag | 00  00  00  00                                                  |     |
     124800 |     134112 | Rdr | c1 f4!  5f  1b 25! a0! 69! 61!                                  | !crc| DEC(244)       
     312948 |     317620 | Tag | 62  92  09  6c                                                  |     |
     323584 |     328352 | Rdr |5e! 2a!  b2 a3!                                                  | !crc| ?
     376548 |     376804 | Tag |00!

maxben14, 2017-11-11:
In sniff acr122 by 4 bytes i see 2-3 answer from tag 00 or 01.
153048700 |  153053404 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
  153123152 |  153127824 | Tag | 00  00  00  00                                                  |     |
  153129228 |  153138540 | Rdr |9b! 63!  d0  d1  f2  bb  36  72                                  | !crc| ?
  153320160 |  153324832 | Tag | 12  e6  46  3f                                                  |     |
  153451004 |  153455708 | Rdr |3d! c8!  10  28                                                  | !crc| ?
  153513632 |  153513888 | Tag |00!                                                              |     |
  153523472 |  153523728 | Tag |00!                                                              |     |
  153533376 |  153533568 | Tag | 01                                                              |     |
  175438908 |  175439900 | Rdr | 52                                                              |     | WUPA
  175514940 |  175515932 | Rdr | 52

merlokk, 2017-11-11:
it looks like you have problems with modulation.

merlokk, 2017-11-11:
#db# Cmd Error:  - the proxmark deciphers your answer and put it here

maxben14, 2017-11-11:
@merlokk , if this problem in timing how can I check that my device have problem with timing and my emulator code is correct? How do I know if there is an error in my code or is it just a problem with timings.

merlokk, 2017-11-11:
ok, you need to learn how to use raw data sniffer http://www.proxmark.org/forum/viewtopic.php?id=2738 (or another link...)
or you need to have Digital Storage Oscilloscope or something like to catch what you do in field.

maxben14, 2017-11-11:
@merlokk , i try do plot.
proxmark3> data samples
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
proxmark3> data plot
[image]
And how to stretch the graph ?
I want to highlight on it the reception, the transfer, using the documentation Iso14443 on Miller and Manchester

maxben14, 2017-11-11:
I found mistake in my algoritm.
If in function modulate manchester i do 
For i 0 to 18.
    crypto1 operation for i byte
    send encrypt byte
This method don't work.

If i do
crypto1 operations for all massive 18 byte and after
For i 0 to 18
    send encrypt byte
This method work good.
Why my first method don't work ?

merlokk, 2017-11-11:
It not works coz your processor too slow

iceman1001, 2017-11-11:
This is not an issue.  Start a thread on the forum instead.

closed 2017-11-11 by iceman1001

maxben14, 2017-11-11:
@merlokk, i use attiny4313 mk, can you advise me which microcontroller to put instead of this ?

maxben14, 2017-11-11:
Now about timing i get this.
 35968 |      40672 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
     111076 |     115748 | Tag | 00  00  00  00                                                  |     |
     125568 |     134880 | Rdr |be!  30 3d!  41  a1 93! 1d! ef!                                  | !crc| ?
     314372 |     319044 | Tag | 22  25  62  4e                                                  |     |
     324992 |     329760 | Rdr | c4 51! ae! 1d!                                                  | !crc| ?
     545892 |     566756 | Tag |f0! 3c! 5c!  04 dc!  d2 2d! 01! 6c!  ba ad!  89  9e  9a  57 dd!  |     |
            |            |     |5b!  c4                                                          | !crc|
     579968 |     584736 | Rdr |c1! 9a! 86! 8e!     

Can say this timing very bad ?

merlokk, 2017-11-12:
cpu
https://www.ti.com/product/am3358
board
https://beagleboard.org/black
http://www.myirtech.com/list.asp?id=494

there is a realtime kernel that have driver for linux.

maxben14, 2017-11-13:
@merlokk , my code from the moment the command is received 60 xx to the command Nt spends 5 ms, by datasheet it takes no more than 1 ms. I need a processor faster 5 times, can you advise cpu from the atmega series?

merlokk, 2017-11-13:
atmega too slow... maybe stm32f4xx or f7xx

merlokk, 2017-11-13:
it depends. look at mifare card (not at documentation) and try to make the clone. And you see all that i try to told you...

maxben14, 2017-11-14:
@merlokk, if not at datasheet, where ? You advise make clone how flat card ?

merlokk, 2017-11-14:
Try to read card with proxmark, look at timings and behavior and try to copy this, not the data see)