Proxmark/proxmark3 #479: hf mf nested crashes

kingbobi, 2017-11-12:
Hello,

I have problems executing the following command successfully. 
```
hf mf nested 1 0 A ffffffffffff d
```
When the command is executed, proxmark3 crashes after a short time.
When I use it on a non-empty card it does not calculate all required keys.

Here is an example output (with gdb):
```
gdb --args ./proxmark3 /dev/pm3-0 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
...
Reading symbols from ./proxmark3...done.

(gdb) run
Starting program: /home/king/git/proxmark3/client/proxmark3 /dev/pm3-0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe6fe1700 (LWP 25668)]
[New Thread 0x7fffe67e0700 (LWP 25669)]
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:18
os: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200250 bytes (38%). Free: 324038 bytes (62%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          

proxmark3> hf mf nested 1 4 A 385432bca5e9 d
--nested. sectors:16, block no:  4, key type:A, eml:n, dmp=y checktimeout=471 us          
Testing known keys. Sector count=16          
nested...          
-----------------------------------------------          
uid:245c7340 trgbl=0 trgkey=0          
[New Thread 0x7fffe5fdf700 (LWP 25675)]
[New Thread 0x7fffe57de700 (LWP 25676)]
[Thread 0x7fffe57de700 (LWP 25676) exited]
[Thread 0x7fffe5fdf700 (LWP 25675) exited]
-----------------------------------------------          
uid:245c7340 trgbl=0 trgkey=1          
[New Thread 0x7fffe57de700 (LWP 25677)]
[New Thread 0x7fffe5fdf700 (LWP 25678)]
[Thread 0x7fffe5fdf700 (LWP 25678) exited]
[Thread 0x7fffe57de700 (LWP 25677) exited]
Found valid key:322c9cbbe53f          
-----------------------------------------------          
uid:245c7340 trgbl=4 trgkey=0          
[New Thread 0x7fffe5fdf700 (LWP 25679)]
[New Thread 0x7fffe57de700 (LWP 25680)]
[Thread 0x7fffe57de700 (LWP 25680) exited]
[Thread 0x7fffe5fdf700 (LWP 25679) exited]
-----------------------------------------------          
uid:245c7340 trgbl=4 trgkey=1          
[New Thread 0x7fffe57de700 (LWP 25681)]
[New Thread 0x7fffe5fdf700 (LWP 25682)]
[Thread 0x7fffe5fdf700 (LWP 25682) exited]
[Thread 0x7fffe57de700 (LWP 25681) exited]
-----------------------------------------------          
uid:245c7340 trgbl=8 trgkey=0          
[New Thread 0x7fffe5fdf700 (LWP 25683)]
[New Thread 0x7fffe57de700 (LWP 25684)]
[Thread 0x7fffe5fdf700 (LWP 25683) exited]
[... 99 lines cut ...]

Thanks for your help!
kingbob_

My setup:
Linux x64
4.10.0-38-generic
```
proxmark3> hw version
[[[ Cached information ]]]
          
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:18
os: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200250 bytes (38%). Free: 324038 bytes (62%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory 
```

iceman1001, 2017-11-12:
-->  "Printing keys to binary file dumpkeys.bin..."

https://github.com/proxmark/proxmark3/blob/master/client/cmdhfmf.c#L816

So, when it writes to file,  it breaks.

kingbobi, 2017-11-12:
No, even if I don't write the keys to a file (without d flag) the program crashes. I think that when the command is executed, the crash is intercepted and finally  the calculated keys are printed or the file is written. 

```
gdb --args ./proxmark3 /dev/pm3-0 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
...
Reading symbols from ./proxmark3...done.
(gdb) run
Starting program: /home/king/git/proxmark3/client/proxmark3 /dev/pm3-0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe6fe1700 (LWP 31122)]
[New Thread 0x7fffe67e0700 (LWP 31123)]
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:18
os: master/v3.0.1-174-gc0e852f-suspect 2017-11-11 13:29:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200250 bytes (38%). Free: 324038 bytes (62%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf mf nested 1 0 A e251a9da734d
--nested. sectors:16, block no:  0, key type:A, eml:n, dmp=n checktimeout=471 us          
Testing known keys. Sector count=16          
nested...          
-----------------------------------------------          
uid:245c7340 trgbl=0 trgkey=0          
[New Thread 0x7fffe5fdf700 (LWP 31132)]
[New Thread 0x7fffe57de700 (LWP 31133)]
[Thread 0x7fffe5fdf700 (LWP 31132) exited]
[Thread 0x7fffe57de700 (LWP 31133) exited]
-----------------------------------------------          
...
...
...
-----------------------------------------------          
uid:245c7340 trgbl=56 trgkey=1          
[New Thread 0x7fffe5fdf700 (LWP 32216)]
[New Thread 0x7fffe57de700 (LWP 32217)]
[Thread 0x7fffe57de700 (LWP 32217) exited]
[Thread 0x7fffe5fdf700 (LWP 32216) exited]

-----------------------------------------------
Nested statistic:
Iterations count: 211          
Time in nested: 170,867 (0,810 sec per key)          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  000000000000  | 0 |  000000000000  | 0 |          
|001|  385432bca5e9  | 1 |  20050d3ef4c5  | 1 |          
|002|  773d520cd2e2  | 1 |  012387edfa37  | 1 |          
|003|  000000000000  | 0 |  000000000000  | 0 |          
|004|  000000000000  | 0 |  73fdf90acdc2  | 1 |          
|005|  73fee908ada2  | 1 |  000000000000  | 0 |          
|006|  1723ad983cf7  | 1 |  043d560cdd24  | 1 |          
|007|  1723ad983cf7  | 1 |  043d560cdd24  | 1 |          
|008|  1723ad983cf7  | 1 |  043d560cdd24  | 1 |          
|009|  000000000000  | 0 |  000000000000  | 0 |          
|010|  000000000000  | 0 |  000000000000  | 0 |          
|011|  000000000000  | 0 |  000000000000  | 0 |          
|012|  000000000000  | 0 |  c60aa7b3d2fd  | 1 |          
|013|  84676fb9dccd  | 1 |  b5bba2210f35  | 1 |          
|014|  000000000000  | 0 |  000000000000  | 0 |          
|015|  1723ad983cf7  | 1 |  043d560cdd24  | 1 |          
|---|----------------|---|----------------|---|          
*** stack smashing detected ***: /home/king/git/proxmark3/client/proxmark3 terminated

Thread 2 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe6fe1700 (LWP 31122)]
0x00007ffff5953428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c:  no such file or directory.
(gdb) bt
#0  0x00007ffff5953428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff595502a in __GI_abort () at abort.c:89
#2  0x00007ffff59957ea in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff5aad45f "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff5a3711c in __GI___fortify_fail (msg=<optimized out>, msg@entry=0x7ffff5aad441 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff5a370c0 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x0000000000448820 in CmdHF14AMfNested (Cmd=<optimized out>) at cmdhfmf.c:841
#6  0x000000000047b657 in CmdsParse (Commands=Commands@entry=0x7140a0 <CommandTable>, Cmd=<optimized out>, Cmd@entry=0x7fffe0015ce6 "nested 1 0 A e251a9da734d")
    at cmdparser.c:70
#7  0x000000000044daf2 in CmdHFMF (Cmd=0x7fffe0015ce6 "nested 1 0 A e251a9da734d") at cmdhfmf.c:2610
#8  0x000000000047b657 in CmdsParse (Commands=Commands@entry=0x713660 <CommandTable>, Cmd=<optimized out>) at cmdparser.c:70
#9  0x00000000004395f1 in CmdHF (Cmd=<optimized out>) at cmdhf.c:720
#10 0x000000000047b657 in CmdsParse (Commands=Commands@entry=0x715f20 <CommandTable>, Cmd=<optimized out>, Cmd@entry=0x7fffe0015ce0 "hf mf nested 1 0 A e251a9da734d")
    at cmdparser.c:70
#11 0x000000000047bc9d in CommandReceived (Cmd=Cmd@entry=0x7fffe0015ce0 "hf mf nested 1 0 A e251a9da734d") at cmdmain.c:196
#12 0x00000000004088d2 in main_loop (script_cmds_file=<optimized out>, script_cmd=0x0, usb_present=<optimized out>) at proxmark3.c:180
#13 0x00007ffff6603e3c in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#14 0x00007ffff797b6ba in start_thread (arg=0x7fffe6fe1700) at pthread_create.c:333
#15 0x00007ffff5a253dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

```

merlokk, 2017-11-12:
Strange...
sysdeps/unix/sysv/linux/raise.c:  no such file or directory.

iceman1001, 2017-11-12:
I can verify this bug,  I get same crash.
it is triggered when leaving CmdHFNested  (return 0)

merlokk, 2017-11-14:
@kingbobi, all is OK?

kingbobi, 2017-11-14:
Yes the error has been fixed, but the command still could not calculate all keys. However, this could be fixed by increasing the variable NESTED_SECTOR_RETRY in client/cmdhfmf. c. (see #483 )

With this fix everything works fine. Thank you guys

closed 2017-11-22 by pwpiwi