Proxmark/proxmark3 #485: Proxmark slowness issue

Kaicastledine, 2017-11-15:
Hello,

I've noticed with my proxmark that performance degrades after a few commands.
Im using the CLI on Mac OSX - High seirra 10.13.1

[image]

When using a Mifare s50 card 

hf search - Takes up to 30 seconds
hf mf - takes up to 1 min 23 seconds to lists commands
hf mf chk = ? - takes up to 1 min 57 seconds
hf mf nested 1 0 B ffffffffffff d - Takes 15 min 29 seconds ! 

Any ideas how I can troubleshoot what's causing the slowness ?

pwpiwi, 2017-11-15:
Duplicate of #283?

Kaicastledine, 2017-11-16:
Indeed, I've just read the whole issue post.
Thanks for the reference. 

Will keep posted on the other thread.

pwpiwi, 2017-11-22:
Closed in favor of #283

closed 2017-11-22 by pwpiwi