Proxmark/proxmark3 #489: Proxmark don't check parity bits.

maxben14, 2017-11-18:
I'm developing a classic 1k emulator. I'm writing the firmware for atmega and testing the results on the proxmark.
The Proxmark does not check the parity bits in the card responses during the block read operation.

Example read block:
 hf mf rdbl 0 a a0a1a2a3a4a5
--block no:0, key type:A, key:a0 a1 a2 a3 a4 a5
#db# READ BLOCK FINISHED
isOk:01 data:dc cf ee 01 fc 08 04 00 62 63 64 65 66 67 68 69

    0 |        992 | Rdr | 52                                                              |     | WUPA
       2244 |       4612 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL       
      10692 |      16580 | Tag | 01  02  03  04  04                                              |     |
      19072 |      29600 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
      30788 |      34308 | Tag | 08  b6  dd                                                      |     |
      35968 |      40672 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
      41924 |      46596 | Tag | 00  00  00  00                                                  |     |
      56320 |      65632 | Rdr | c1 f4!  5f  1b 25! a0! 69! 61!                                  | !crc| DEC(244)       
      78164 |      82836 | Tag | 62  92  09  6c                                                  |     |
      88704 |      93472 | Rdr |5e! 2a!  b2 a3!                                                  | !crc| ?
     123236 |     144100 | Tag |54! d5!  e4  c6  e2  7c  a6  f7 f8! a5! f7! c6!  94 28! 5a! 31!  |     |
            |            |     | 42 be!                                                          | !crc|  // Here proxmark don't check parity correct bits.
     157312 |     162080 | Rdr |30!  ef 54! be!       

I try from my emulator on comand read block send encrypt answer 16 bytes and crc 2 bytes with parity random bits and proxmark don't show me error parity bits.

pwpiwi, 2017-11-20:
Correct. Parity errors are only displayed in `hf list` (and you should ignore them for encrypted data). Tag answers are not checked for parity errors.

iceman1001, 2017-11-30:
close?

pwpiwi, 2017-11-30:
Let's keep this as a reminder (like #12).

merlokk, 2017-11-30:
yes. as for me - I dont want to make proxmark a "standard reader" (i made some of them) Its just reader with big "field of look" and development tool
and as for me dont need to check all the conditions.

pwpiwi, 2017-11-30:
... and as a development tool I would expect that it is able to analyze incoming data for parity errors. There should always be an option to ignore it though.

iceman1001, 2017-11-30:
agree with @pwpiwi