Proxmark/proxmark3 #492: How in android with nfc do Nr const ?

maxben14, 2017-11-22:
How in android with nfc do Nr const ?
Can this do in android firmware ?

pwpiwi, 2017-11-22:
This is the proxmark github repository.

maxben14, 2017-11-22:
@pwpiwi , i understand, but me interest, i do emulator 1k and my emulator send At in reader about 700 microsec, in datasheet max 1 ms, proxmark read my device good, but android function Authentificate return false. Do you know about timing in android ?

pwpiwi, 2017-11-23:
No. And afaik the other maintainers in this group develop proxmark software on and for Linux and Windows. Some Mac users are around as well. This is definitely the wrong group to ask for Android development.

maxben14, 2017-11-23:
@pwpiwi , why does no one make the firmware of the proxmark for android? It is very uncomfortable to approach the turnstile with the proxmark along with the laptop, android is more convenient.

pwpiwi, 2017-11-23:
There are forks with changes for android. We had provided the links already.

iceman1001, 2017-11-24:
This Github repository is for the Proxmark device.  Not for Android apps,  nor iphone apps,  or other devices like chameleon mini,  emutag,   or other libs like libnfc or libfreefare.  
closed.

closed 2017-11-24 by iceman1001