Proxmark/proxmark3 #495: Unknown byte in hf 14a apdu

ikarus23, 2017-11-28:
Hi! Thanks for adding the APDU feature. I gave it a try but I noticed an issue.  Not sure if I'm doing something wrong.

I want to send an APDU like `0a0000a4040c...` But when I've looked at the trace there is always a `02` in front of my APDU. Did I miss something? Is this suppose to happen?

```
proxmark3> hf 14a apdu -s 0a0000a4040c....
>>>>[sel ] 0a 00 00 a4 04 0c 09 d2 76 00 00 67 30 01 02 00           
<<<< 67 00           
APDU response: 67 00 - Wrong length          
proxmark3> hf list 14a
Recorded Activity (TraceLen = 158 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
iso14443a - All times are in carrier periods (1/13.56Mhz)          
iClass    - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |        992 | Rdr | 52                                                              |     | WUPA          
       2228 |       4596 | Tag | 04  00                                                          |     |           
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL          
      10676 |      16564 | Tag | ef  5d  4a  33  cb                                              |     |           
      19328 |      29856 | Rdr | 93  70  ef  5d  4a  33  cb  06  57                              |  ok | SELECT_UID          
      31028 |      34548 | Tag | 28  b4  fc                                                      |     |           
      36352 |      41120 | Rdr | e0  80  31  73                                                  |  ok | RATS          
      46644 |      62900 | Tag | 0c  78  80  84  02  80  31  c0  73  d6  20  c0  a3  58          |  ok |           
     128512 |     150496 | Rdr | 02  0a  00  00  a4  04  0c  ...                                 |     |           
            |            |     | 00  4c  81                                                      |  ok | ?          
     363956 |     369844 | Tag | 02  67  00  f1  38                                              |     |
```

merlokk, 2017-11-28:
Its from iso14443 APDU low level. Just look into iso14443

merlokk, 2017-11-28:
P. S. Tag answers wrong Length. You need to have correct Length in APDU

ikarus23, 2017-11-28:
I just copied the APDU from a comunication I sniffed and skipped the CRC. In that sniffed communication no APDU starts with `02`. I get `9000` from the card if I also skip the first two bytes of the original communication (`0a00`). But the trace still shows `0200` instead of the original sniffed `0a00`.

I guess I have to look a the ISO standard... ;)

merlokk, 2017-11-28:
If skip first 2 bytes - your APDU is correct.
First 2 bytes - from ISO 14443, s-block and cid

merlokk, 2017-11-28:
You can check your APDU in command hf 14a raw. It will work

merlokk, 2017-11-28:
https://stackoverflow.com/questions/9513383/evrc-smart-cards

ikarus23, 2017-11-28:
@merlokk Thanks for all the help. It worked to recreate the sniffed command with `hf 14a raw`.

```
proxmark3> hf 14a apdu -s 00a4040c...      
>>>>[sel ] 00 a4 04 0c 09 d2 76 00 00 67 30 01 02 00           
<<<< 90 00           
APDU response: 90 00 - Command successfully executed (OK).          
proxmark3> hf 14a raw -c -s 0a0000a4040c...
Card selected. UID[4]:          
EF 5D 4A 33           
received 6 bytes:          
0A 00 90 00 F3 93  
```
Well It also worked with `hf 14a apdu` but with `hf 14a apdu` I can recreate the 100% identical communication.  As you said, the only difference is the first byte and that has nothing to do with APDU (ISO 7816-4) but ISO 14443A.

closed 2017-11-28 by ikarus23

iceman1001, 2017-11-29:
apdu encapsulation..  
0x02 doesnt use CID
0x0a use CID 

your sniffed data: 

0a00 00a4040c 
   |      |----------  apdu  cmd 
   | -------encapsulation with CID.

When implementing APDU on pm3,  we decided to skip CID for starters. 
and the command should encapsulate it for you.   Leaving us only to need the command.

ie your command  00 a4 04 0c .. .. ..  
become 02 00 a4 00 0c .. .. ..

And here is where it all goes bad, because ppl sniff and sees what you see and thinks its the "raw command"...    Its just down to not knowing what you are trying to do.  ;)     I guess this means there will be some RTFM posts on the forum.

ikarus23, 2017-11-29:
@iceman1001 Thanks for the detailed explanation! Yeah, I guess it always comes down to RTFM.  :) I mean I somehow expected it to be some part of the encapsulation because I already did know that the CRC is part of this... 

Maybe we can change the help text from
>Send ISO 14443-4 APDU to tag

to something like

> Send an ISO 7816-4 APDU encapsulated in a ISO 14443-4 frame

But in the end you always have to know your stuff, if you are looking at sniffed traces. I did know that it was a APDU communication, but I didn't know enough about the wrapping... ;)

iceman1001, 2017-11-29:
Sounds like an execellent idea

merlokk, 2017-11-29:
maybe this?
Send an ISO 7816-4 APDU via ISO 14443-4 block transmission protocol

ikarus23, 2017-11-29:
> Send an ISO 7816-4 APDU via ISO 14443-4 block transmission protocol (T=CL)

I like it! I would skipped the "(T=CL)" because proxmark3 help texts are as short as possible.

Also, I would have changed it right away. But although being a member of "Proxmark" I'm no longer allowed to push to the "proxmark3" repository (since 2017-11-07). I don't know why :-(

iceman1001, 2017-11-30:
....and that is because we are revising the maintainers.  We want active maintainers, so those who hasn't being active has been put on hold.

pwpiwi, 2017-11-30:
See issue #464.

ikarus23, 2017-11-30:
Well, do as it pleases you :). But in this case you have to change the help text yourself.

iceman1001, 2017-11-30:
You are more than welcome to become active again but you seem to have been busy with MCT and not here.

ikarus23, 2017-11-30:
Hehe, thank you. But I can't promise anything. You are right. For me MCT does indeed have a higher priority than the Proxmark3 (I love both projects). If you only want "really" active people, I might be the wrong guy (lack of spare time). I can't compete with @iceman1001, @pwpiwi, @marshmellow42 or any of you guys. ;) But hey, my commit history of small fixes got me the 7th pace on the contributer list. Yay :P

Jokes apart: If you want me to fix some small things from time to time, you can add me again. If not, I'm ok with it too! In that case I might just add a PR.

commit 606eddca (2017-11-30): added hint on `hf 14a apdu`. close #495
commit d2172825 (2017-11-30): Merge pull request #502 from merlokk/apduhelp

added hint on `hf 14a apdu`. close #495