Proxmark/proxmark3 #496: Compile fullimage.elf with debug symbols

SpenserStyles, 2017-11-29:
How do I compile the fullimage.elf with debug symbols?  I have added -g flag to CFLAGS in Makefile.common re-compiled it but GDB still complains that their are no debugging symbols in the fullimage.elf file.

pwpiwi, 2017-11-30:
After compiling and linking the elf file goes through several modifications to include the fpga binaries and to compress the .data section. See armsrc/Makefile:

```
$(OBJDIR)/fullimage.stage1.elf: $(VERSIONOBJ) $(OBJDIR)/fpga_all.o $(THUMBOBJ) $(ARMOBJ)
	$(CC) $(LDFLAGS) -Wl,-T,ldscript,-Map,$(patsubst %.elf,%.map,$@) -o $@ $^ $(LIBS)

$(OBJDIR)/fullimage.nodata.bin: $(OBJDIR)/fullimage.stage1.elf
	$(OBJCOPY) -O binary -I elf32-littlearm --remove-section .data $^ $@

$(OBJDIR)/fullimage.nodata.o: $(OBJDIR)/fullimage.nodata.bin
	$(OBJCOPY) -O elf32-littlearm -I binary -B arm --rename-section .data=stage1_image $^ $@

$(OBJDIR)/fullimage.data.bin: $(OBJDIR)/fullimage.stage1.elf
	$(OBJCOPY) -O binary -I elf32-littlearm --only-section .data $^ $@

$(OBJDIR)/fullimage.data.bin.z: $(OBJDIR)/fullimage.data.bin $(FPGA_COMPRESSOR)
	$(FPGA_COMPRESSOR) $(filter %.bin,$^) $@

$(OBJDIR)/fullimage.data.o: $(OBJDIR)/fullimage.data.bin.z
	$(OBJCOPY) -O elf32-littlearm -I binary -B arm --rename-section .data=compressed_data $^ $@

$(OBJDIR)/fullimage.elf: $(OBJDIR)/fullimage.nodata.o $(OBJDIR)/fullimage.data.o
	$(CC) $(LDFLAGS) -Wl,-T,ldscript,-e,_osimage_entry,-Map,$(patsubst %.elf,%.map,$@) -o $@ $^
```
From the `objcopy` documentation:
> When objcopy generates a raw binary file, it will essentially produce a memory dump of the contents of the input object file. All symbols and relocation information will be discarded.

closed 2017-12-08 by iceman1001

sampurndash, 2022-02-20:
Hi [SpenserStyles](https://github.com/SpenserStyles), 

Did You find the way to add debug information . I am facing some issues in debugging the Proxmark3

Gator96100, 2022-02-20:
> Hi [SpenserStyles](https://github.com/SpenserStyles),
> 
> Did You find the way to add debug information . I am facing some issues in debugging the Proxmark3

The [RRG repo](https://github.com/RfidResearchGroup/proxmark3) supports firmware and client debugging in VSCode.
If there is a need to get it working on this repo look at how it was done:
https://github.com/RfidResearchGroup/proxmark3/issues/1138
https://github.com/RfidResearchGroup/proxmark3/pull/1146

sampurnd, 2022-06-21:
Hi @Gator96100 ,
I followed the following steps in Ubuntu,

1)Download and install [Visual Studio Code](https://code.visualstudio.com/) 

2)Download and install [J-Link Software and Documentation pack](https://www.segger.com/downloads/jlink) 

3)On some systems `arm-none-eabi-gdb` was replaced with `gdb-multiarch`. In order to use the J-Link debugger you need to link `arm-none-eabi-gdb` to `gdb-multiarch`:
```sh
ln -s /usr/bin/gdb-multiarch /usr/bin/arm-none-eabi-gdb
```

4)Setup the Visual Studio Code configuration, by going into your project folder and run:
```sh
./.vscode/setup.sh
```

5)now launch Visual Studio Code and open your project folder..

Still not able to set the debug environment for firmware . I have also connected the JTAG. Can you please let me know the next steps .

sampurnd, 2022-06-21:
Thanks @Gator96100 ,
I am able to debug the firmware .