Proxmark/proxmark3 #497: Issues with after --head flash

Kaicastledine, 2017-11-29:
Hello,

So I've had the USB power issues on mac as previously posted.

Went and reflashed using  the --head brew install.
(Uninstalled previous one first)
[image]

[image]

Now I get the following issues - All tested with a Mifare S50

[image]

[image]

[image]

Any ideas or maybe a re-flash ?

Kaicastledine, 2017-11-29:
Just rand a mifare-autopwn script and it's crashed the proxmark and rebooted it 
[image]

Kaicastledine, 2017-11-29:
Atleast the USB power/performance issue seems to of gone :)

Kaicastledine, 2017-11-29:
Some more inconstant  HF search results, S50 card not moved 
[image]

Kaicastledine, 2017-11-29:
This is with a different card - Magic UID
[image]

Kaicastledine, 2017-11-29:
LF search results in proxmark crash 
[image]

iceman1001, 2017-11-29:
That UID is sure much longer than 10 bytes...
Congrats, I never seen such mess anytime and I'm good at breaking my pm3 client.

pwpiwi, 2017-11-30:
Can you please provide the rest of the crash report as well (lines 13 and following)?

iceman1001, 2017-11-30:
for the UID, to break,   and print 256 values,  it needs uidlen to be 0xff,  but uidlen is populated from usbcommand response.  So I'm guessing but the iso14443a select,  is sending wrong data.

around here:  https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c#L2042

which would make sense since that one piece of code that recently been changed.

Kaicastledine, 2017-11-30:
I've run the same commands using the same tag without issue on the stable build.

Indeed I thought it was a bit of a laugh last night.

I've re-flashed with the stable build and these issues go away but get the USB Power/performance issue over time.

Any ideas ?

Kaicastledine, 2017-11-30:
Here is the crash reportProcess:               proxmark3 [39630]
Path:                  /usr/local/bin/proxmark3
Identifier:            proxmark3
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [34781]
Responsible:           proxmark3 [39630]
User ID:               501

Date/Time:             2017-11-29 18:59:03.665 +0000
OS Version:            Mac OS X 10.13.1 (17B48)
Report Version:        12
Anonymous UUID:        7887CD69-07E0-1A0F-1A9A-F0367945F7EC

Sleep/Wake UUID:       2C4D67FB-4ED7-4771-9DF6-6DDDFB3896B4

Time Awake Since Boot: 180000 seconds
Time Since Wake:       2900 seconds

System Integrity Protection: enabled

Crashed Thread:        2  WorkerThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 00000001096a5000-000000010975a000 [  724K] r-x/rwx SM=COW  /usr/local/Cellar/proxmark3/HEAD-653e5ed/bin/proxmark3

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff78439e76 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff78439390 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff50d125d5 __CFRunLoopServiceMachPort + 341
3   com.apple.CoreFoundation      	0x00007fff50d11927 __CFRunLoopRun + 1783
4   com.apple.CoreFoundation      	0x00007fff50d10fa3 CFRunLoopRunSpecific + 483
5   com.apple.HIToolbox           	0x00007fff50030866 RunCurrentEventLoopInMode + 286
6   com.apple.HIToolbox           	0x00007fff500305d6 ReceiveNextEventCommon + 613
7   com.apple.HIToolbox           	0x00007fff50030354 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff4e32da23 _DPSNextEvent + 2085
9   com.apple.AppKit              	0x00007fff4eac2e6c -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 3044
10  com.apple.AppKit              	0x00007fff4e322831 -[NSApplication run] + 764
11  libqcocoa.dylib               	0x000000010dbab05d 0x10db84000 + 159837
12  org.qt-project.QtCore         	0x000000010ae30d0e QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 398
13  org.qt-project.QtCore         	0x000000010ae35621 QCoreApplication::exec() + 369
14  proxmark3                     	0x00000001096fd709 ProxGuiQT::MainLoop() + 335
15  proxmark3                     	0x00000001096a8982 main + 1204
16  libdyld.dylib                 	0x00007fff782f3145 start + 1

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff78439e76 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff78439390 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff50d125d5 __CFRunLoopServiceMachPort + 341
3   com.apple.CoreFoundation      	0x00007fff50d11927 __CFRunLoopRun + 1783
4   com.apple.CoreFoundation      	0x00007fff50d10fa3 CFRunLoopRunSpecific + 483
5   com.apple.AppKit              	0x00007fff4e46b2d1 _NSEventThread + 184
6   libsystem_pthread.dylib       	0x00007fff7857d6c1 _pthread_body + 340
7   libsystem_pthread.dylib       	0x00007fff7857d56d _pthread_start + 377
8   libsystem_pthread.dylib       	0x00007fff7857cc5d thread_start + 13

Thread 2 Crashed:: WorkerThread

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff78443142 __select + 10
1   proxmark3                     	0x00000001096a8d30 uart_receive + 143
[... 321 lines cut ...]
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       191.6M       11 
__TEXT                           162.2M      259 
__UNICODE                          560K        2 
mapped file                       52.7M       22 
shared memory                     1828K       15 
===========                     =======  ======= 
TOTAL                            665.8M      668 

Model: MacBookPro11,2, BootROM MBP112.0142.B00, 4 processors, Intel Core i7, 2.2 GHz, 16 GB, SMC 2.18f15
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.37.2.1a3)
Bluetooth: Version 6.0.0f1, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: PM3
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

pwpiwi, 2017-11-30:
> I've run the same commands using the same tag without issue on the stable build.

What does that mean? The issue is gone? Or is "stable build" another version?

Kaicastledine, 2017-11-30:
Not using the --head version the proxmark doesnt have the same issues.

Only when using the --head version do I get those bugs

Kaicastledine, 2017-11-30:
More specifically 

```
for stable release

brew install proxmark3
for latest non-stable from GitHub (use this if previous command fails)

brew install --HEAD proxmark3
```

Kaicastledine, 2017-11-30:
At work atm, what's the best way to check and I will post it this evening.

pwpiwi, 2017-11-30:
Frankly I don't know if this can be achieved with homebrew.

On a commandline I would checkout version after version going back from HEAD. Compile and test until it works as expected.

Kaicastledine, 2017-11-30:
Sure will see what the results are and send them over.

pwpiwi, 2018-02-01:
@Kaicastledine: are you still on? Is your issue gone? If not, can you please test PR #558 if it helps?

Kaicastledine, 2018-02-14:
Hello, Yes never got fair with testing previously.

Will get that tested and come back to you, thanks :)

Kaicastledine, 2018-03-14:
Finally updated it ! Will get it tested this week and report back.

Kaicastledine, 2018-03-15:
Right so issues when scanning mifare s50 

```
proxmark3> hf search
          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
No chinese magic backdoor command detected          
PRNG UID: Reply timeout.          
Prng detection error.         
```

when scanning EM4xxx   ( crashed proxmark)

```
proxmark3> lf search
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
Segmentation fault: 11
KCMP:HEAD-51d51c6 kaic$ 
```

only a couple of quick tests and the proxmark crashed and shutdown so I will try some more soon and see what's going on.

iceman1001, 2018-03-15:
@Kaicastledine   your email?  or send me a mail on  iceman @@ iuse  se  ..

pwpiwi, 2018-03-16:
There are some indications that the issue is with the Qt integration. Does commit  36b1cd work for you?

Kaicastledine, 2018-03-18:
Just emailed you iceman,  Ill check it out now

Kaicastledine, 2018-03-18:
@pwpiwi  Can't seem to find 36b1cd , can you post a link to the commit please ?

pwpiwi, 2018-03-18:
Sure: https://github.com/Proxmark/proxmark3/commit/36b1cdd1b4064e1a506b5bab0d00e61b91cb10b2

Kaicastledine, 2018-03-18:
Iceman headless , 3rd command lagged then 4th timed out 

```
Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-741-g4c666e38 2018-03-18 15:04:36
      os: iceman/master/ice_v3.1.0-741-g4c666e38 2018-03-18 15:04:39
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 255169 bytes (49) Free: 269119 bytes (51)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
          
pm3 --> hf search
          
 UID : E4 A0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to magic commands: NO          
Prng detection: WEAK          
Valid ISO14443-A Tag Found
          
pm3 --> hf search
          
 UID : E4 A0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to magic commands: NO          
Prng detection: WEAK          
Valid ISO14443-A Tag Found
          
pm3 --> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag
          
False Positives ARE possible
          
Checking for known tags:
          
Valid Indala ID Found!
          
pm3 --> lf search
command execution time out          
NOTE: some demods output possible binary
  if it finds something that looks like a tag
          
False Positives ARE possible
          
Checking for known tags:
          
Valid Indala ID Found!
          
pm3 --> hf search
          
timeout while waiting for reply.          
no known/supported 13.56 MHz tags found
          
pm3 --> 

```

Kaicastledine, 2018-03-18:
@pwpiwi  Just saw the https://github.com/Proxmark/proxmark3/pull/585  post.

Had this been added to a commit if not what shall I test and I will reflash the proxmark back from the now Iceman install.

iceman1001, 2018-03-18:
Interesting,  I get a sneaky feeling its the swapping between fpga images,   HF/LF.
I pushed the #585 into iceman fork already.

pwpiwi, 2018-03-18:
OK. Seems that trying different commits doesn't lead to anything. I even have the impression that some commits work for some people and segfault for others. Are you able to run a client on OSX with a debugger (e.g. dbg)? It would be quite helpfull to know at which line or at least in which function it segfaults.

Something along these lines should work:
* Optional (gives us information on source file and line number): Compile with an additional -g option
* `gdb`
* `run Proxmark3 /dev/cu.usbmodem1421`
And after it segfaults
* `bt`

Kaicastledine, 2018-03-18:
Tried getting it to work with gdb, think the syntax is correct however get this ?

```
KCMP:bin kaic$ sudo gdb --args ./proxmark3 /dev/cu.usbmodem1421
Password:
GNU gdb (GDB) 8.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin17.3.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./proxmark3...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/local/Cellar/proxmark3/HEAD-51d51c6/bin/proxmark3 /dev/cu.usbmodem1421
[New Thread 0xc03 of process 96287]
During startup program terminated with signal ?, Unknown signal.
(gdb) 
```

Followed this guide and now get another error
https://stackoverflow.com/questions/40110435/unknown-ending-signal-when-using-debugger-gdb

```
(gdb) run
Starting program: /usr/local/Cellar/proxmark3/HEAD-51d51c6/bin/proxmark3 /dev/cu.usbmodem1421
[New Thread 0x1303 of process 96504]
[New Thread 0x1103 of process 96504]
During startup program terminated with signal SIGTRAP, Trace/breakpoint trap.

```

gdb might not easily work on my mac, will keep trying.

Any other suggestions at all for a debugger ?

Kaicastledine, 2018-03-18:
Got gdb working via this
https://gist.github.com/gravitylow/fb595186ce6068537a6e9da6d8b5b96d

```
Valid EM410x ID Found!          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
[New Thread 0xc07 of process 97465]
[New Thread 0xd03 of process 97465]
[New Thread 0x1503 of process 97465]
[New Thread 0x2903 of process 97465]

Thread 3 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xc07 of process 97465]
0xffffffffffffffff in ?? ()
(gdb) bt
#0  0xffffffffffffffff in ?? ()
#1  0xd7dfe6f0faffffff in ?? ()
#2  0xa8aeb2b8bdc3c9cf in ?? ()
#3  0x00000000173f6da6 in ?? ()
#4  0x0000000000000000 in ?? ()
(gdb) 
```

micolous, 2018-03-19:
You want `lldb`, not `gdb` for debugging stuff on OSX these days.

```
$ lldb ./client/proxmark3 /dev/....
(lldb) run
(proxmark3) do things that crash it
(lldb) bt
```

I'm unsure if the codesigning / security lockdown in current OSX kernels is preventing `gdb` from loading debugging symbols, or if `homebrew` is stripping out all the debug symbols. I haven't tried to run PM3 from `homebrew` for a while.

iceman1001, 2018-03-19:
...running gdb on ubuntu17.04,   gets lots of crashes for readline after some few commands..

`*** Error in `/home/osboxes/pm3/client/proxmark3': free(): invalid pointer: 0x00007fffd4004100 ***

Thread 6 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe35ec700 (LWP 7631)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff58e6f5d in __GI_abort () at abort.c:90
#2  0x00007ffff592f28d in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff5a56528 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff593664a in malloc_printerr (action=<optimized out>, str=0x7ffff5a52eae "free(): invalid pointer", ptr=<optimized out>, ar_ptr=<optimized out>)
    at malloc.c:5426
#4  0x00007ffff593873e in _int_free (av=0x7ffff5c88c20 <main_arena>, p=<optimized out>, have_lock=0) at malloc.c:4175
#5  0x00007ffff593d44e in __GI___libc_free (mem=<optimized out>) at malloc.c:3145
#6  0x00007ffff7bbba86 in _rl_free_history_entry () from /lib/x86_64-linux-gnu/libreadline.so.7
#7  0x00007ffff7bbbaca in _rl_start_using_history () from /lib/x86_64-linux-gnu/libreadline.so.7
#8  0x00007ffff7b9fc7f in rl_initialize () from /lib/x86_64-linux-gnu/libreadline.so.7
#9  0x00007ffff7b9ff87 in readline () from /lib/x86_64-linux-gnu/libreadline.so.7
#10 0x000055555557149f in main_loop (script_cmds_file=0x0, script_cmd=0x0, usb_present=<optimized out>) at proxmark3.c:270
#11 0x00007ffff65fbe3c in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#12 0x00007ffff79737fc in start_thread (arg=0x7fffe35ec700) at pthread_create.c:465
#13 0x00007ffff59c2b5f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) fr 10
#10 0x000055555557149f in main_loop (script_cmds_file=0x0, script_cmd=0x0, usb_present=<optimized out>) at proxmark3.c:270
270						cmd = readline(PROXPROMPT);
(gdb) list
265							pthread_mutex_unlock(&print_lock);
266						}
267						
268					} else {		
269						// read command from command prompt
270						cmd = readline(PROXPROMPT);
271					}
272				}
273			}
274			
`

Kaicastledine, 2018-03-20:
@micolous 

I know SIP causes a lot of issues with gdb. I'll check the other debugger and see how much info that shows.

pwpiwi, 2018-03-21:
Interesting. `readline()` calls `rl_initialize()` (which it should only do when called for the first time) and then frees unallocated memory. Probably somehow readline's internal memory structures have been corrupted by other pieces of code writing to it. Or it is a readline bug.

iceman1001, 2018-03-21:
remove the redundant rl_initialize call in micolous patch inside proxmark3.c  and the crashes dissappears.

pwpiwi, 2018-03-21:
Which @micolous patch? PR #585 is not yet committed.

iceman1001, 2018-03-21:
yeah,  (https://github.com/Proxmark/proxmark3/pull/585)  (https://github.com/Proxmark/proxmark3/pull/463#issuecomment-374898106)     Since kaicastledine is trying it out, I posted,  but with hindsight posting the other thread would have been better.  Now just hope kaiscastledine is not trying out that other one while testing for this one..

micolous, 2018-03-21:
OK, I'll remove that call for now. I put that in because of another client crash that I observed by trying to give libedit equivalent functionality.

pwpiwi, 2018-03-21:
I have committed #585 now (without the `rl_initialize()`)

Kaicastledine, 2018-04-25:
Right got a new mac ... Going to set things back up and get some tests done.

Whats the current score as I've seen theres been a few updates.

pwpiwi, 2018-04-26:
Status is still unclear. @micolous added some patches for OSX readline compatibility mode. @iceman1001 reported crashes with Ubuntu when initializing readline with `rl_initialize()`, but no crashes when initializing with `using_history()` (which calls `rl_initialize()`).

We are still waiting for someone (you?) who can confirm the crashes on OSX **and** use a debugger to identify where exactly it crashes.

Kaicastledine, 2018-04-26:
Alright, I've just had my mac replaced so will get the debugger etc set back up and run a few tests.

Will post back soon!

Kaicastledine, 2018-04-26:
Thanks @micolous  Got lldb working with no issues !

lldb proxmark commands
```
KCMP-2:**HEAD-61aaee3** kaicas$ **lldb -- Proxmark3 /dev/cu.usbmodem1421**
(lldb) target create "Proxmark3"
Current executable set to 'Proxmark3' (x86_64).
(lldb) settings set -- target.run-args  "/dev/cu.usbmodem1421"
(lldb) **run**
Process 770 launched: '/usr/local/bin/Proxmark3' (x86_64)
2018-04-26 10:05:11.862208+0100 Proxmark3[770:37331] MessageTracer: load_domain_whitelist_search_tree:73: Search tree file's format version number (0) is not supported
2018-04-26 10:05:11.862235+0100 Proxmark3[770:37331] MessageTracer: Falling back to default whitelist
```
Proxmark starting 

```
Prox/RFID mark3 RFID instrument          
bootrom: master/-suspect 2018-04-25 23:44:17
os: master/-suspect 2018-04-25 23:44:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200160 bytes (38). Free: 324128 bytes (62).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory  
```    
    
LF search crashing 

```
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
proxmark3> lf search
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
^[[AProcess 770 stopped
* thread #10, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x000000010006139e Proxmark3`getCommand + 154
Proxmark3`getCommand:
->  0x10006139e <+154>: retq   

Proxmark3`UsbCommandReceived:
    0x10006139f <+0>:   pushq  %rbp
    0x1000613a0 <+1>:   movq   %rsp, %rbp
    0x1000613a3 <+4>:   pushq  %r14
Target 0: (Proxmark3) stopped.
(lldb) bt
* thread #10, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x000000010006139e Proxmark3`getCommand + 154
```

Kaicastledine, 2018-04-26:
Restarted proxmark and run lf search commands again

```
KCMP-2:HEAD-61aaee3 kaicas$ lldb -- Proxmark3 /dev/cu.usbmodem1421
(lldb) target create "Proxmark3"
Current executable set to 'Proxmark3' (x86_64).
(lldb) settings set -- target.run-args  "/dev/cu.usbmodem1421"
(lldb) run
Process 775 launched: '/usr/local/bin/Proxmark3' (x86_64)
2018-04-26 10:09:13.157519+0100 Proxmark3[775:42946] MessageTracer: load_domain_whitelist_search_tree:73: Search tree file's format version number (0) is not supported
2018-04-26 10:09:13.157555+0100 Proxmark3[775:42946] MessageTracer: Falling back to default whitelist
Prox/RFID mark3 RFID instrument          
bootrom: master/-suspect 2018-04-25 23:44:17
os: master/-suspect 2018-04-25 23:44:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200160 bytes (38). Free: 324128 bytes (62).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
[... 406 lines cut ...]
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
Process 775 stopped
* thread #10, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x222)
    frame #0: 0x0000000000000222
error: memory read failed for 0x200
Target 0: (Proxmark3) stopped.
(lldb) bt
* thread #10, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x222)
  * frame #0: 0x0000000000000222
```

Kaicastledine, 2018-04-26:
HF search starts doing some weird stuff but doesnt crash the app

(```
lldb) run
There is a running process, kill it and restart?: [Y/n] y
Process 775 exited with status = 9 (0x00000009) 
Process 779 launched: '/usr/local/bin/Proxmark3' (x86_64)
2018-04-26 10:11:21.209568+0100 Proxmark3[779:45810] MessageTracer: load_domain_whitelist_search_tree:73: Search tree file's format version number (0) is not supported
2018-04-26 10:11:21.209596+0100 Proxmark3[779:45810] MessageTracer: Falling back to default whitelist
Prox/RFID mark3 RFID instrument          
bootrom: master/-suspect 2018-04-25 23:44:17
os: master/-suspect 2018-04-25 23:44:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200160 bytes (38). Free: 324128 bytes (62).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          

Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          

Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          

Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          

Valid ISO14443A Tag Found - Quiting Search
          
[... 423 lines cut ...]
PRNG data error: Wrong length: 0          
Prng detection error.          

Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
SAK incorrectly claims that card doesn't support RATS          
 ATS : 78 d0 ea 68           
       -  TL : length is 120 bytes          
ATS may be corrupted. Length of ATS (4 bytes incl. 2 Bytes CRC) doesn't match TL          
       -  T0 : TA1 is present, TB1 is NOT present, TC1 is present, FSCI is 0 (FSC = 4294967280)          
       - TA1 : different divisors are NOT supported, DR: [4], DS: [4, 8]          
       - TC1 : NAD is NOT supported, CID is NOT supported          
       -  HB : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
Chinese magic backdoor command (GEN 1b) detected          
PRNG data error: Wrong length: 0          
Prng detection error.          

Valid ISO14443A Tag Found - Quiting Search
```

Kaicastledine, 2018-04-26:
Switching between LF search + HF search 

```
lldb) run
Process 782 launched: '/usr/local/bin/Proxmark3' (x86_64)
2018-04-26 10:15:57.124548+0100 Proxmark3[782:52564] MessageTracer: load_domain_whitelist_search_tree:73: Search tree file's format version number (0) is not supported
2018-04-26 10:15:57.124578+0100 Proxmark3[782:52564] MessageTracer: Falling back to default whitelist
Prox/RFID mark3 RFID instrument          
bootrom: master/-suspect 2018-04-25 23:44:17
os: master/-suspect 2018-04-25 23:44:19
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200160 bytes (38). Free: 324128 bytes (62).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
EM410x pattern found:           

EM TAG ID      : 41008F16EF          

Possible de-scramble patterns          
Unique TAG ID  : 8200F168F7          
HoneyWell IdentKey {          
DEZ 8          : 09377519          
DEZ 10         : 0009377519          
DEZ 5.5        : 00143.05871          
DEZ 3.5A       : 065.05871          
DEZ 3.5B       : 000.05871          
DEZ 3.5C       : 143.05871          
DEZ 14/IK2     : 00279182251759          
DEZ 15/IK3     : 000558361569527          
DEZ 20/ZK      : 08020000150106081507          
}
Other          : 05871_143_09377519          
Pattern Paxton : 1101223151 [0x41A354EF]          
Pattern 1      : 3100597 [0x2F4FB5]          
Pattern Sebury : 5871 15 988911  [0x16EF 0xF 0xF16EF]          

Valid EM410x ID Found!          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          

Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
[... 132 lines cut ...]
False Positives ARE possible
          

Checking for known tags:
          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
Process 782 stopped
* thread #11, name = 'WorkerThread', stop reason = signal SIGABRT
    frame #0: 0x00007fff53d8ab6e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff53d8ab6e <+10>: jae    0x7fff53d8ab78            ; <+20>
    0x7fff53d8ab70 <+12>: movq   %rax, %rdi
    0x7fff53d8ab73 <+15>: jmp    0x7fff53d81b00            ; cerror_nocancel
    0x7fff53d8ab78 <+20>: retq   
Target 0: (Proxmark3) stopped.
(lldb) bt
* thread #11, name = 'WorkerThread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff53d8ab6e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff53f55080 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff53ce624d libsystem_c.dylib`__abort + 144
    frame #3: 0x00007fff53ce6af8 libsystem_c.dylib`__stack_chk_fail + 205
    frame #4: 0x000000010004f4e6 Proxmark3`downloadSamplesEM + 136
```

pwpiwi, 2018-04-27:
Excellent! I think I know what is going on and why it crashes. Will try to find some time over the weekend for a fix. (I still have no idea what causes the dramatic USB speed slowdown).

iceman1001, 2018-04-27:
Could it be the sample_buf is getting overwritten in cmdmain.c when downloading from device since there is no array length checks there?

pwpiwi, 2018-04-27:
No. It is that after a timeout (`WaitForCommandTimeout()`) it is possible to continue writing into `sample_buf`. If this happens to be an automatic variable (i.e. a stack location) which is already released, it overwrites other variables and the return address.

iceman1001, 2018-04-27:
isn't sample_buf a pointer, used as global variable?   What is a automatic variable in C?

pwpiwi, 2018-04-27:
A variable declared inside a function and not `static`. `sample_buf` can point to it.

iceman1001, 2018-04-27:
aha,  you talk about local scope

pwpiwi, 2018-04-28:
No, it is not about local scope. Variables declared `static` inside a function have local scope too. But they have a memory reserved which remains valid after the function returns. The memory of an automatic variable is on stack and is reused for other purposes after the function returns.

pwpiwi, 2018-05-03:
@Kaicastledine: could you test the fix in PR #597 ?

Kaicastledine, 2018-06-27:
Apologies for the delay had to shelf the proxmark stuff but got some time again !

Latest test HEAD-fdee1ff

Lf read - has some issue after repeated use 

```
KCMP-2:HEAD-fdee1ff kaicas$ Proxmark3 /dev/cu.usbmodem1421
Prox/RFID mark3 RFID instrument          
bootrom: master/-suspect 2018-06-27 14:54:05
os: master/-suspect 2018-06-27 14:54:06
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 200278 bytes (38). Free: 324010 bytes (62).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf read
#db# LF Sampling config:           
#db#   [q] divisor:           95           
#db#   [b] bps:               8           
#db#   [d] decimation:        1           
#db#   [a] averaging:         1           
#db#   [t] trigger threshold: 0           
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample          
#db# buffer samples: 00 00 00 00 cd ff ff ff ...          
Reading 39999 bytes from device memory
          
Data fetched          
Samples @ 8 bits/smpl, decimation 1:1           
proxmark3> lf read
#db# LF Sampling config:           
#db#   [q] divisor:           95           
#db#   [b] bps:               8           
#db#   [d] decimation:        1           
#db#   [a] averaging:         1           
#db#   [t] trigger threshold: 0           
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample          
#db# buffer samples: 00 00 00 00 cd ff ff ff ...          
Reading 39999 bytes from device memory
          
Data fetched          
Samples @ 8 bits/smpl, decimation 1:1           
proxmark3> lf read
#db# LF Sampling config:           
#db#   [q] divisor:           95           
#db#   [b] bps:               8           
#db#   [d] decimation:        1           
#db#   [a] averaging:         1           
#db#   [t] trigger threshold: 0           
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample          
#db# buffer samples: 00 00 00 00 ce ff ff ff ...          
Reading 39999 bytes from device memory
          
Data fetched          
Samples @ 8 bits/smpl, decimation 1:1           
proxmark3> lf read
#db# LF Sampling config:           
#db#   [q] divisor:           95           
#db#   [b] bps:               8           
#db#   [d] decimation:        1           
#db#   [a] averaging:         1           
#db#   [t] trigger threshold: 0           
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample          
#db# buffer samples: 00 00 cd ff ff ff ff ff ...          
Reading 39999 bytes from device memory
[... 587 lines cut ...]
Valid ISO14443A Tag Found - Quiting Search
          
proxmark3> hf search
          
 UID : e4 a0 66 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
SAK incorrectly claims that card doesn't support RATS          
 ATS : 93 3d a9 1c           
       -  TL : length is 147 bytes          
ATS may be corrupted. Length of ATS (4 bytes incl. 2 Bytes CRC) doesn't match TL          
       -  T0 : TA1 is present, TB1 is present, TC1 is NOT present, FSCI is 13 (FSC = 4294967295)          
       - TA1 : different divisors are NOT supported, DR: [2], DS: [4]          
       - TB1 : SFGI = 12 (SFGT = 16777216/fc), FWI = 1 (FWT = 8192/fc)          
       -  HB : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
Chinese magic backdoor command (GEN 1b) detected          
PRNG data error: Wrong length: 0          
Prng detection error.          

Valid ISO14443A Tag Found - Quiting Search
          
```

Let me know if theres anything else that I can do :)

pwpiwi, 2018-07-04:
The good news: no more crashes. But we still have the underlying issue of a sudden slowdown in USB communications resulting in timeouts (usually indicated by "Waiting for a response from Proxmark").

pwpiwi, 2018-07-13:
With the memory corruption being fixed I would like to close this issue because the remaining problems are most probably a duplicate of #283 - but I would like to confirm. @Kaicastledine, can you please compile without Qt (i.e. no graph support) and confirm that the issues are then gone?

Kaicastledine, 2018-07-28:
Yeah sure, how can I compile without Qt and I'll get it tested ?

pwpiwi, 2018-07-28:
In client/Makefile remove (or comment) all the following lines:
```
# Check for correctly configured Qt5
QTINCLUDES = $(shell pkg-config --cflags Qt5Core Qt5Widgets 2>/dev/null)
QTLDLIBS = $(shell pkg-config --libs Qt5Core Qt5Widgets 2>/dev/null)
MOC = $(shell pkg-config --variable=host_bins Qt5Core)/moc
UIC = $(shell pkg-config --variable=host_bins Qt5Core)/uic
ifeq ($(QTINCLUDES), )
# if Qt5 not found check for correctly configured Qt4	
	QTINCLUDES = $(shell pkg-config --cflags QtCore QtGui 2>/dev/null)
	QTLDLIBS = $(shell pkg-config --libs QtCore QtGui 2>/dev/null)
	MOC = $(shell pkg-config --variable=moc_location QtCore)
	UIC = $(shell pkg-config --variable=uic_location QtCore)
else
	CXXFLAGS += -std=c++11 -fPIC
endif
ifeq ($(QTINCLUDES), )
# if both pkg-config commands failed, search in common places
	ifneq ($(QTDIR), )
		QTINCLUDES = -I$(QTDIR)/include -I$(QTDIR)/include/QtCore -I$(QTDIR)/include/QtGui
		QTLDLIBS = -L$(QTDIR)/lib -lQtCore4 -lQtGui4
		ifneq ($(wildcard $(QTDIR)/include/QtWidgets),)
			QTINCLUDES += -I$(QTDIR)/include/QtWidgets
			QTLDLIBS = -L$(QTDIR)/lib -lQt5Widgets -lQt5Gui -lQt5Core
			CXXFLAGS += -std=c++11 -fPIC
		endif
		MOC = $(QTDIR)/bin/moc
		UIC = $(QTDIR)/bin/uic
	endif
endif
```
Then `make clean` and `make all` again. You should get a proxmark client without grafics. And it should not show the weird behaviour. Please confirm.

[2 more replies]

closed 2018-10-02 by pwpiwi