Proxmark/proxmark3 #499: antenna outputs after power on

iceman1001, 2017-11-30:
Its been brought to my attention that the proxmark3 is 

1. power out 13.56Mhz when started (initialised).  This output is not stopped until arbitary command is sent.   Swapping from 13.56 -> LF is risk for damaging the driver chip.  Current is large and board have a lot of heat.
2. 'hf 15' commands leaves antenna on when executed.  ('hf 14a' commands stops)
3. PoD - power on delay in latest source.  Latest source is very slow from start.  I belive this is related with flashing delay issues mentioned.

I suggested a fix for 1.   While it seems a call to FpgaDownloadAndGo() powers the antenna. This is think is strange. 
Solution:  Turn off antenna afterward afterwards

Fix for 2,  would be looking at all places when a command is finished, it needs to turn off antenna.

iceman1001, 2017-11-30:
or, when thinking on it a bit,  add the FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);   inside FpgaDownloadAndGo,  in the end.

That will make sure no misstaken powersurges will arise from calls to FpgaDownloadAndGo()

pwpiwi, 2017-11-30:
After so many years someone notices that FPGA is not initialized after downloading the configuration!

I agree that the fix should go into FpgaDownloadAndGo().

Re damaging driver: This should never happen. The drivers have a resistor of at least 33Ohms connected to their output. Hence maximum current is 5V/33Ohms = 150mA. There are at least three drivers connected to the 33Ohms resistor, i.e. the current is max 50mA per driver which is the allowed maximum.

iceman1001, 2017-11-30:
So you knew about it and didn't think it was interesting to fix?  
The idea is that the antenna is not powered when swapping between HF - LF in order to be kind to the hardware.

The damaging of drivers,  should not, but could.  There might be hardware revisions which is different.
New revisions which has different antenna strengths compared to old models....

pwpiwi, 2017-11-30:
> So you knew about it and didn't think it was interesting to fix?

No, I didn't notice yet. But it is such a basic thing that I am astonished that no one stumbled over it yet.

> There might be hardware revisions which is different.
 New revisions which has different antenna strengths compared to old models....

That would be bad engineering. I am sure that it is not by accident that the value of these resistors result in exactly the maximum allowed output current. They are meant to protect the drivers.

iceman1001, 2017-11-30:
ok, I misunderstood you.  
I didn't realized it until it was brought to my attention and I look inside DownloadFpga() where is sets all to HIGH. Not sure what is needed to initialise the fpga properly.  Its outside my knowledge. This is your field.

Of course, a new revision would have matching resistors and drivers for the new load.

iceman1001, 2017-11-30:
Partly fixed.

What is needed is the tiresum work of looking att all places where the fpga is turned on, and make sure its turned off once its finished.  That is if the command "should" turn off.  Like "hf 14a raw" will need to keep it running..

iceman1001, 2018-02-07:
Another part in making the device behave nice.
https://github.com/Proxmark/proxmark3/pull/566

iceman1001, 2018-02-08:
after this one,  the following commands leave power on.

- hf 15 reader
- hf iclass reader

pwpiwi, 2018-02-09:
Which may be an indicator that it was intended like this. Do you remember our discussion when we changed `hf 14a reader` to behave like this? I still think that the intention was to start a "session" with the reader commands. You then would send commands to the card and see the responses and finally send a "power off" command.

iceman1001, 2018-02-09:
yes,  I agree,  but then we added the **hf search** which doesn't act like that, but we use these "readers" command like a select command,   hence the  **hf 14a raw** can up with option to leave field on/off.  Which I think is an ok solution.    The 15, iclass, reader command acts inbetween here.   There is no clear behavior between  sending  raw commands and act as a simple reader, reading uid and report back.

That is the underlaying problem which needs to be adressed.  I have divided much of that in iceman fork,to have a clear definition between  "reader" / "info" / "raw"  commands.

iceman1001, 2018-02-12:
I think I just leave this one be.   This issue is to general and concerns serveral ones.
I close it and if the need to address the other issues arises,  then a new dedicated issue is better.

closed 2018-02-12 by iceman1001