Proxmark/proxmark3 #505: Build fails with AVX512 (again)

uzlonewolf, 2017-12-02:
Issue #386 has reappeared due to commit de1e68d34bcf7e2d3396f32c4ebc9dab839c9774 .

Still Fedora 26,
```
$ uname -a
Linux uzlonewolf.d 4.11.8-300.fc26.x86_64 #1 SMP Thu Jun 29 20:09:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,objc,obj-c++,fortran,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 7.1.1 20170622 (Red Hat 7.1.1-3) (GCC) 

$ gcc --version
gcc (GCC) 7.1.1 20170622 (Red Hat 7.1.1-3)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ gcc -dumpversion
7
```

Before blowing up with a bunch of "undefined reference to `*_AVX512'" later in the compile, make spits out
```
proxmark3]$ make
make -C client  all
make[1]: Entering directory '/home/uzlonewolf/proxmark/proxmark3/client'
/bin/sh: -c: line 0: syntax error near unexpected token `)'
/bin/sh: -c: line 0: `[ 70101-3) -ge 40900 ] && echo "True" '
Compiling liblua, using platform linux
cd ../liblua && make linux
...
```
The offending Makefile section is
```
GCC_VERSION := $(shell gcc --version | awk '/gcc/{print $$NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/')
CLANG_VERSION := $(shell gcc --version | awk '/Apple LLVM version/{print $$4;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/')
ifneq ($(CLANG_VERSION), )
    SUPPORTS_AVX512 :=  $(shell [ $(CLANG_VERSION) -ge 80000 ] && echo "True" )
endif
ifneq ($(GCC_VERSION), )
    SUPPORTS_AVX512 :=  $(shell [ $(GCC_VERSION) -ge 40900 ] && echo "True" )
endif
```
Perhaps another regex that strips everything after 3 groups of numbers?  Just stripping everything except [0-9\\.] won't work as i.e. 4.1.1-3 will get turned into 401013 and wrongly pass the -ge.  That whole line feels like a bad kludge to me, there's got to be a better way of doing this...

micolous, 2017-12-03:
This looks like a better fix: https://stackoverflow.com/questions/28939652/how-to-detect-sse-avx-avx2-avx-512-availability-at-compile-time

Then we should be able to do `-march=native`.

But we'd need to take care on platforms where binaries get redistributed and not built from source (eg: Windows), as if the compiler had a new Intel CPU, the binaries wouldn't work on older (and non-Intel) CPUs.

rtomsick, 2017-12-05:
Is there a workaround for this for those of us who have no other machines other than those running Fedora 26+?

uzlonewolf, 2017-12-05:
Edit client/Makefile
Replace lines 195 - 202 with the single line "SUPPORTS_AVX512 := True"

```
diff -u client/Makefile.orig client/Makefile
--- client/Makefile.orig	2017-12-02 13:35:23.568065444 -0800
+++ client/Makefile	2017-12-04 20:36:24.154593133 -0800
@@ -192,14 +192,8 @@
 			$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX.o) \
 			$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)
 
-GCC_VERSION := $(shell gcc --version | awk '/gcc/{print $$NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/')
-CLANG_VERSION := $(shell gcc --version | awk '/Apple LLVM version/{print $$4;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/')
-ifneq ($(CLANG_VERSION), )
-	SUPPORTS_AVX512 :=  $(shell [ $(CLANG_VERSION) -ge 80000 ] && echo "True" )
-endif
-ifneq ($(GCC_VERSION), )
-	SUPPORTS_AVX512 :=  $(shell [ $(GCC_VERSION) -ge 40900 ] && echo "True" )
-endif
+SUPPORTS_AVX512 := True
+
 HARD_SWITCH_NOSIMD = -mno-mmx -mno-sse2 -mno-avx -mno-avx2
 HARD_SWITCH_MMX = -mmmx -mno-sse2 -mno-avx -mno-avx2
 HARD_SWITCH_SSE2 = -mmmx -msse2 -mno-avx -mno-avx2
```

iceman1001, 2017-12-05:
What is your output in these two variables?
GGC_VERSION
CLANG_VERSION

uzlonewolf, 2017-12-05:
GCC_VERSION 70101-3)
CLANG_VERSION

iceman1001, 2017-12-05:
try that one...

https://github.com/iceman1001/proxmark3/commit/66e0e18a35dc62c85506da2d3bbea3786a4ed373

iceman1001, 2017-12-05:
hm,...  still,
on ubuntu 16.0.4 and GCC 5.4.0  I still get it wrong since the --version looks differently.

pwpiwi, 2017-12-05:
@micolous:
> This looks like a better fix: https://stackoverflow.com/questions/28939652/how-to-detect-sse-avx-avx2-avx-512-availability-at-compile-time

This is compile time detection (which we use as well). But we need to know before calling gcc if it supports `-mavx512` and `-mnoavx512`

And no, we don't want to use `-march=native` because the distributed binaries would then run on specific machines only,

@uzlonewolf: yes, it is a kludge. Unfortunately `--dumpversion` is unknown by older gcc releases. And the output of  `--version` follows no standard. Currently we therefore need to adapt the version detection whenever someone decides to add a new flavor to the `gcc --version` output. 

Possible solution: In the end we are not really interested in the gcc version. We just want to know if it supports `-mavx512`. We therefore could simply run gcc with an empty input file and see if it throws an error. Thoughts?

uzlonewolf, 2017-12-05:
How old of a GCC do we need to support?  2.0 (~25 years ago) supports -dumpversion.  In the gcc sources ```gcc -dumpversion | awk -F. '{print $1}'``` is used to get the major version; if we really want to keep looking at the version number then we should probably use that - no/error output means 0.x or 1.x, otherwise we now have the major version number.  If the major number is 4 then we can use the full output of -dumpversion to get the full version number as the shortened major-only response wasn't added until gcc 7.0.

That said, yes, I agree running a test with -mavx512 is a much better solution.

uzlonewolf, 2017-12-05:
```echo | gcc -x c -mavx512f -c -o /dev/null - 2>/dev/null && echo True``` ?

pwpiwi, 2017-12-05:
Try PR #507

uzlonewolf, 2017-12-06:
PR #507 works great, no issues.

pwpiwi, 2017-12-06:
Unfortunately doesn't work for everyone. PR #508 hopefully fixes the fix

iceman1001, 2017-12-06:
#508   worked for me on Ubuntu16.04 and mingw

iceman1001, 2017-12-08:
@uzlonewolf   did it work?  time to close?

uzlonewolf, 2017-12-11:
Yup, works great

closed 2017-12-11 by uzlonewolf