Proxmark/proxmark3 #510: HF MF MIFARE

iceman1001, 2017-12-06:
@pwpiwi   
The refactoring you did some months ago,  I've been trying to sort out why iceman fork has problems with it and I stumble on a strange behavior.

Its around here:
https://github.com/Proxmark/proxmark3/blob/8afa9712f7597da04338e61cfdbce9c219b0840c/client/mifarehost.c#L181

When it collects the list of potential keys, fine.   However,  it _always_ want to execute twice and compare with another list of key candidates.  Making this current implementation take longer time to execute.

Normal keys:
If first key list has keys,  it should try to authenticate them.  If that fails, it should go and get another bunch of keys.   Where this match up should work.

Unusual keys,
like those with parity=0,  which _needs_ two run times of keylists.

iceman1001, 2017-12-07:
Not to mention the bug with sync-ing a prng which is not following normal speed or reason.. triggering the WDT when the sync_time gets huge because of it.

PRNG speeds 0.617s  for a complete sequence,  hwoever its been brought to my attention some clones doens't follow that speed but are much faster.  There are three types,  2x , 4x, 8x  the speed in cycling a prng sequence.   This makes it hard to "lock" into with sync_time variable.

pwpiwi, 2017-12-07:
Doesn't matter. After 0.617s the faster PRNGs have completed 2, 4, or 8 complete cycles which will result in the same nonce.

pwpiwi, 2017-12-07:
And re always executing twice you may be right. Need to have a closer look.

iceman1001, 2017-12-07:
Of course you would need to look, but my claim is correct for unnessesary loop.

The faster prng, is a different story,  I haven't pinpointed down the exact problem or why,   but my best guess is the nt_distance starts adding up,  when unsynced. 
When in sync the faster prng iteration wouldn't be a problem I agree.

pwpiwi, 2018-01-22:
Unnecessary extra loop is not fixed.

pwpiwi, 2018-02-05:
No issue here:
```

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
.............Found 41 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff
```

Why do our setups behave so different?

iceman1001, 2018-02-05:
I wasn't clear enough.  The attack works some times and then we have the WDT crash.
it works against normal (weak) s50,  clone s50 with strange prng,     its only zero parity (all nack) clones which behavies inconsistent.  Where its about 50/50 in successful execution vs crash

pwpiwi, 2018-02-05:
I cannot confirm. No WDT crash within 11 consecutive runs. But a few wrong PRNG classifications.
```
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
......Found 9 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
......Found 2 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
.....Found 2 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
......Found 3 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
....Found 3 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
[... 24 lines cut ...]
Attack will take a few seconds longer because we need two consecutive successful runs.
......Found 9 possible keys. Trying to authenticate with each of them ...

Found valid key:ffffffffffff

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
..Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
......Found a possible key. Trying to authenticate...

Found valid key:ffffffffffff

proxmark3>
```

iceman1001, 2018-02-05:
First time,  WDT..  I have another key than 0xFFFFFFFFFFFF..  

```
pm3 ~$ git pull
Already up-to-date.
pm3 ~$ cd client
pm3 ~/client$ ./proxmark3.exe com3
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-301-gc839fa2-dirty-suspect 2018-02-01 09:36:19
os: master/v2.2.0-807-gc839fa2-dirty-suspect 2018-02-05 19:08:18
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 199376 bytes (38%). Free: 324912 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...Parity is all zero. Most likely this card sends NACK on every failed authentication.
Attack will take a few seconds longer because we need two consecutive successful runs.
........Found 10 possible keys. Trying to authenticate with each of them ...

Authentication failed. Trying again...
...........Found 27 possible keys. Trying to authenticate with each of them ...

Authentication failed. Trying again...
.....................Found 5 possible keys. Trying to authenticate with each of them ...

Authentication failed. Trying again...
............................
```

iceman1001, 2018-02-05:
Five times strange prng,  not a problem.

```
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
.....Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
....Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
.......Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
......Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
........Card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
generating polynomial with 16 effective bits only, but shows unexpected behaviour.
```

iceman1001, 2018-02-05:
Five time NACK/WEAK , not a problem
```
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
............Key not found (lfsr_common_prefix list is null). Nt=7a38a870
This is expected to happen in 25% of all cases. Trying again with a different reader nonce...
..........Key not found (lfsr_common_prefix list is null). Nt=7a38a870
This is expected to happen in 25% of all cases. Trying again with a different reader nonce...
.........Found a possible key. Trying to authenticate...

Found valid key:fc00018778f7

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
..................Key not found (lfsr_common_prefix list is null). Nt=e19dca3d
This is expected to happen in 25% of all cases. Trying again with a different reader nonce...
..........Key not found (lfsr_common_prefix list is null). Nt=e19dca3d
This is expected to happen in 25% of all cases. Trying again with a different reader nonce...
..........Found a possible key. Trying to authenticate...

Found valid key:fc00018778f7

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
......................Found a possible key. Trying to authenticate...

Found valid key:fc00018778f7

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
..............Found a possible key. Trying to authenticate...

Found valid key:fc00018778f7

proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
...............Found a possible key. Trying to authenticate...

Found valid key:fc00018778f7

proxmark3>
```

iceman1001, 2018-02-06:
well,  the PR solves the double running, so its merged,  and this issue is resolved.
Time will tell if the residue behavior is based on difference in setup or a bug.

For those who experiences this wdt crash behavior can verify it by also running iceman fork on same setup since iceman fork doesn't have this behavior.

closed 2018-02-06 by iceman1001

pwpiwi, 2018-02-06:
I have now seen WDT crashes with Fudan clone (always NACK) as well. Not so often than you do, but they happen. Will open another issue...

iceman1001, 2018-02-07:
This PR https://github.com/Proxmark/proxmark3/pull/568 solves some of the problems.
However long running crashes still might exist,  we just will find a correct key much faster now :)