Proxmark/proxmark3 #513: EMV emulator

merlokk, 2017-12-07:
Im looking into code and think about EMV card emulator.
I think about something like HCE in Android. ARM side will just transfers commands to client side and the client side will send responses
something like:
```
CLIENT                                    ARM
Setup(UID, ATC)      ----->      
                              <------       OK
GetState                 ----->     
                              <------     CardState
                               ............
GetState                 ----->     
                              <------     CardState, GotDataFromField
SendData               ----->     
                              <------     CardState, OK
                               ............
ExitEmulator          ----->     
                              <------     OK
```

What do you think? If it possible to do with current USB code?

iceman1001, 2017-12-07:
shouldn't be any problem at all.  The transfer speed is ok.

merlokk, 2017-12-07:
Its good.
Can the arm side work with Xilinx and with USB simultaneously?

iceman1001, 2017-12-07:
??  xilinx and usb

merlokk, 2017-12-07:
emulator must work with 2 parts of proxmark simultaneously:
1. USB
2. Antenna-analog part-fpga-cpu serial

If I will transfer some data via USB is it will affect on data transfer to fpga (antenna)?

iceman1001, 2017-12-07:
nothing of the proxmark3 currently is done in parallell or via interrupts.
so when it sends usb,  then it sends usb.  The most fancy is  'hf mf sim'   which tries to do both.

The other kinds is, 'hf 14a sim'  which only does logging until it runs out of memory..

merlokk, 2017-12-07:
thanks) so it will needs to check some thoughts how to do that

merlokk, 2017-12-08:
for now i see 2 possibilities:
1. Send command to ARM and infinite wait data from ARM with response (as in `hf mf mifare` and etc)
2. make DMA config with 2 buffers and check data from DMA. And do connection as in my first message

Maybe as for first part - i will do emulator with way 1 and then refactor to way 2 if it would needs.

iceman1001, 2017-12-08:
have a look at  **hf mf sniff**
https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c#L2433

that is most likely what you want.

merlokk, 2017-12-08:
yes)   thanks)))  will try to connect it with usb logic)

iceman1001, 2017-12-08:
of course, look at the client side code ...

iceman1001, 2017-12-11:
Cool!

merlokk, 2017-12-12:
OK, proof of concept works. 
Maximum buffer length like 147 bytes. It seems 1kb buffer will be enough)

```
#db# dlen=1 maxdlen=147 count=3338889
All is OK. Returned.
#db# Command done.
```

merlokk, 2017-12-13:
Concept works)

closed 2017-12-13 by merlokk