Proxmark/proxmark3 #514: Lf snoop trigger threshold does nothing

Tom5ive, 2017-12-09:
Hi All,

Testing with the latest client / firmware compiled on OS X High Sierra. I can't get lf snoop to work at all, the command executes and completes regardless of what threshold is set. If I use the brew tap it works fine but that repo seems to have other bugs.

Cheers,
TH

iceman1001, 2017-12-09:
...so... no debug info,  no showing which commands you did.... 
Not very helpful.

Tom5ive, 2017-12-16:
My apologies. Below is the output - the command seems to complete straight away regardless of threshold set.

tomharkness@SilverBox:~$ proxmark3 /dev/cu.usbmodem1421 
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-209-g0fb3e45-suspect 2017-12-05 21:05:12
os: master/v3.0.1-209-g0fb3e45-suspect 2017-12-05 21:05:14
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 199936 bytes (76). Free: 62208 bytes (24).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hw tune 

Measuring antenna characteristics, please wait.........          
# LF antenna: 30.52 V @   125.00 kHz          
# LF antenna: 23.65 V @   134.00 kHz          
# LF optimal: 30.52 V @   125.00 kHz          
# HF antenna:  0.07 V @    13.56 MHz          
# Your HF antenna is unusable.          
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
          

          
No GUI in this build!
proxmark3> lf conf b 8 L t 64
#db# LF Sampling config:           
#db#   [q] divisor:           95           
#db#   [b] bps:               8           
#db#   [d] decimation:        1           
#db#   [a] averaging:         1           
#db#   [t] trigger threshold: 64           
proxmark3> lf snoop 
proxmark3>

iceman1001, 2017-12-16:
you seem to miss some commands and their output.

```
lf snoop
--- do the snooping part ---
lf read
data plot 
```

marshmellow42, 2017-12-16:
lf read is not needed (it will overwrite the snoop)
But a plot of what you see would help us confirm if your antenna is actually satisfying the threshold, or if someone broke something recently.

iceman1001, 2017-12-16:
I stand corrected

```
data samples
data plot
```

pwpiwi, 2017-12-16:
@marshmellow42 wrote:
> But a plot of what you see would help us confirm if your antenna is actually satisfying the threshold, or if someone broke something recently.

@Tom5ive wrote:
> No GUI in this build!

marshmellow42, 2017-12-17:
Well without a GUI there is little you can do but save a trace and submit it for us to review.  The lf snoop doesn't auto decode anything so a GUI (plot window) is pretty important.

iceman1001, 2017-12-17:
true indeed..

data sample
data save mytrace.pm3

upload file via a filesharing service.

Still, even if,  I have a sneaky suspicion this is not a bug in pm3 but a user who don't understand who to use the pm3.  I also wonder what this lf snoop is expected to do..   Since a 'lf snoop' is one of the more complexer task to analyse...

Tom5ive, 2017-12-21:
Work has been keeping life busy..

I will upload a trace this evening once I return home from work. I don't think the issue is with my antennas (yes they are custom made but are tuned very well). 

Regarding my understanding of lf snoop - I have used the command previously with the home-brew build which has the gui, and I have managed to save multiple traces that apear correctly as a waveform. The reason for using lf snoop is to find the password set by an unknown Chinese cloner.

As I understand it - lf config has a threshold setting, this setting refers to the threshold of RF energy? (I may be wrong here but this is my understanding) that the proxmark coil needs to sense before beginning the snoop process. 

As I explained above, the problem with the latest firmware (in my experience) is that the lf snoop command completes instantly regardless of threshold value set. Meaning that the proxmark does not wait until it senses RF energy based on the threshold value  before completing the command. Thus I cannot snoop between a cloner / card because the command executes instantly.  This behaviour is NOT present in the home-brew client / firmware. One I get home this evening I will upload anything required to help everyone understand.

marshmellow42, 2017-12-21:
I see the issue in the code.  I will attempt to fix this as soon as I can.  Ultimately the cancel-count code needs brackets around the if statement in lf-sampling.c. Afk atm.

iceman1001, 2017-12-21:
@marshmellow42   I added your fix for you.

@Tom5ive  Interesting to hear,  its not often users end up actually learn to look at the lf snoop and decode it to bits & bytes.  Sorry I offended you.

marshmellow42, 2017-12-21:
Please try the latest code now as a fix has been pushed.

Tom5ive, 2017-12-21:
@iceman1001 - No offence taken I completely understand that most people do come into these things not actually understanding whats really happening behind the scenes - I should have been clearer in my original post but was a but rushed at the time.

@marshmellow42 Thanks for sorting this so quickly. I'll give it a test later today.

Tom5ive, 2017-12-22:
The behaviour seems different but it is still not waiting based on threshold. Even with the threshold set to 128 the command hangs for a moment then completes. If I have a tag on the antenna and a cloner ready in hand and write to the tag just after executing the command - the command indeed does execute faster once it detects the rf pulse. But if I execute lf snoop without any RF near the PM3 antenna and wait more than a second, the command completes. 

If there is anything I can upload to help let me know.

iceman1001, 2017-12-22:
that sounds like a wait timeout issue in the client side of LF SNOOP..

Tom5ive, 2017-12-22:
It does but I am unsure of what could be causing it. All other client functionality seems to work as expected, both LF and HF R/W operations. The client compiles just fine and I have all the correct libraries installed.

marshmellow42, 2017-12-22:
Hopefully I'll get a little time this weekend to have a closer look

pwpiwi, 2017-12-23:
There is a fix in the @micolous usb PR. The bug is in the waiting loop on client side.

iceman1001, 2017-12-23:
would that be the remake of pm3 comm PR?  the one that has been hanging for awhile since nobody seem understand the need to remake it?   If its that PR, don't count it will be merged in some time.  Since we have EMV before,  than a release, and then if ppl has time will be looked into.  So fix for this indiviually would be faster.

pwpiwi, 2017-12-23:
No, I don't want to merge this PR now. But it includes the required fix. Just pick this:
https://github.com/Proxmark/proxmark3/pull/463/commits/4acceb76a494388dcac4c09478567fc81d163993

iceman1001, 2017-12-23:
ok,  I cherrypick it, 
https://github.com/Proxmark/proxmark3/pull/528

marshmellow42, 2017-12-23:
Thanks to @iceman1001 and @pwpiwi for the id and fix for the second bug related to this issue.  It should be fixed now

Tom5ive, 2017-12-24:
Thanks all - lf snoop is now working as expected!

closed 2017-12-24 by iceman1001