Proxmark/proxmark3 #517: mf 14a sim broken with Android

ikarus23, 2017-12-11:
Hi,
I wanted to simulate a 7 byte UID of a MIFARE DESFire tag and noticed it worked fine for some reader (I don't know the manufacturer/vendor) but I was not able to detect the card with my Android phone (Google Nexus 5X).

Is there anyone else with this issue? Is there something wrong with Android and simulated tags (as mentioned in [an issue at the MCT repo](https://github.com/ikarus23/MifareClassicTool/issues/154))?

ikarus23, 2017-12-13:
Did some further tests: It seams like reading a emulated tag with and Android device is fine. It just depends on the Android device! Nexus 7 (2012) works great, Nexus 5X does not work. So probably this is just an NFC bug of the Nexus 5X. It would not be the first on this device...

I would appreciate if some proxmark users with  Android phones can test this too. Just to find out if the Nexus 5X is "something special".

micolous, 2017-12-25:
The underlying Android OS will attempt to authenticate with the card using the MIFARE Application Directory (MAD) key first, before any app on the phone is allowed to talk to the card.  I think this is used to allow different applications on the phone to listen to particular MAD entries.

Both the Nexus 5X and the Nexus 7 (2012) have NXP chipsets so I'm not sure why one would work with MIFARE Classic but not the other. I'd test it myself on a 5X, but the one I have is bootlooping and waiting on Google Store to repair it.

In the past when I've tried this with a Nexus 5X, I've been attempting some of the MFC key recovery attacks, and not actually doing a full read with the card.

pwpiwi, 2017-12-27:
@ikarus23 : Can you please provide traces of both simulations? I.e. after you have stopped `hf mf sim`, please run `hf list 14a` and post the results for both Nexus 5X and Nexus 7.

ikarus23, 2017-12-28:
Hmm, that is strange. I updated my PM3 and now my Nexus 7 (2012) and my Nexus 5X can't enumerate the tag. Interesting enough, `hf mf sim u 11223344556677` works for both devices...

Nexus 7 (2012), I touched the tag several times:
```
proxmark3> hf 14a sim 3 11223344556677
Emulating ISO/IEC 14443 type A tag with 7 byte UID (11223344556677)          
#db# Received unknown command (len=1):          
#db# 01          
#db# Received unknown command (len=1):          
#db# 00          
#db# Button press          
#db# 0 0 19       
proxmark3> hf list 14a                
Recorded Activity (TraceLen = 511 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
iso14443a - All times are in carrier periods (1/13.56Mhz)          
iClass    - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |        224 | Rdr | 01                                                              |     | ?          
     109074 |     110130 | Rdr | 26                                                              |     | REQA          
     111302 |     113670 | Tag | 44  03                                                          |     |           
    5813580 |    5814636 | Rdr | 26                                                              |     | REQA          
    5815808 |    5818176 | Tag | 44  03                                                          |     |           
   11416480 |   11416768 | Rdr |00!                                                              |     | ?          
   11520786 |   11521842 | Rdr | 26                                                              |     | REQA          
   11523014 |   11525382 | Tag | 44  03                                                          |     |           
   17221488 |   17222544 | Rdr | 26                                                              |     | REQA          
   17223716 |   17226084 | Tag | 44  03                                                          |     |           
   34333768 |   34334824 | Rdr | 26                                                              |     | REQA          
   34335996 |   34338364 | Tag | 44  03                                                          |     |           
   40038214 |   40039270 | Rdr | 26                                                              |     | REQA          
   40040442 |   40042810 | Tag | 44  03                                                          |     |           
   45736100 |   45737156 | Rdr | 26                                                              |     | REQA          
   45738328 |   45740696 | Tag | 44  03                                                          |     |           
   45747730 |   45750194 | Rdr | 93  20                                                          |     | ANTICOLL          
   45751366 |   45757190 | Tag | 88  11  22  33  88                                              |     |           
   51534244 |   51535300 | Rdr | 26                                                              |     | REQA          
   51536472 |   51538840 | Tag | 44  03                                                          |     |           
   68647430 |   68648486 | Rdr | 26                                                              |     | REQA          
   68649722 |   68652090 | Tag | 44  03                                                          |     |           
   74349168 |   74350224 | Rdr | 26                                                              |     | REQA          
   74351396 |   74353764 | Tag | 44  03                                                          |     |           
   80045552 |   80046608 | Rdr | 26                                                              |     | REQA          
   80047780 |   80050148 | Tag | 44  03                                                          |     |           
   85743268 |   85744324 | Rdr | 26                                                              |     | REQA          
   85745496 |   85747864 | Tag | 44  03                                                          |     |           
   85754642 |   85757106 | Rdr | 93  20                                                          |     | ANTICOLL          
   85758278 |   85764102 | Tag | 88  11  22  33  88                                              |     |           
   91542602 |   91543658 | Rdr | 26                                                              |     | REQA          
   91544830 |   91547198 | Tag | 44  03                                                          |     |           
   97244324 |   97245380 | Rdr | 26                                                              |     | REQA          
   97246552 |   97248920 | Tag | 44  03                                                          |     |           
  102944752 |  102945808 | Rdr | 26                                                              |     | REQA          
  102946980 |  102949348 | Tag | 44  03                                                          |     |           
[... cut ...]
  185643612 |  185649436 | Tag | 44  55  66  77  00                                              |     |           
  185656138 |  185666602 | Rdr | 95  70  44  55  66  77  00  7a  b8                              |  ok | ANTICOLL-2          
  185667838 |  185671422 | Tag | 20  fc  70                                                      |     |           
  185679506 |  185684274 | Rdr | e0  80  31  73                                                  |  ok | RATS          
  185685446 |  185692486 | Tag | 04  58  80  02  13  ce                                          |  ok |
```

pwpiwi, 2018-01-12:
Hmmm, didn't you say that Nexus 7 works and Nexus 5 doesn't? The traces look like the oposite.

ikarus23, 2018-01-13:
>Hmm, that is strange. I updated my PM3 and now my Nexus 7 (2012) and my Nexus 5X can't read the enumerated tag. Interesting enough, `hf mf sim u 11223344556677` works for both devices...

But, yes, from the dumps it looks like the Nexus 5X is working.  However, I never saw any response from the Android system. Maybe Android tries to read some AID by default, if it detects a DESFire tag (emulated with `hf 14a sim`).

It would be great if some of you guys could try the same. Just emulate a DESFire tag with `hf 14a sim` and see if it gets detected by an Android device. If it too dosn't work, I'm pretty sure Android is the issue and not the Proxmark3 emulation.

pwpiwi, 2018-01-13:
What app do you want us to use on Android?

ikarus23, 2018-01-13:
It dosn't mater. Any app that will display the UID should be fine. At least UID emulation is the only goal I want to achieve here. I tested it with Mifare Classic Tool (MCT).

pwpiwi, 2019-04-25:
@ikarus23, if you are still around you may test if PR #812 has solved your issue as well.

ikarus23, 2019-04-25:
@pwpiwi of course I'm still around! However, I was not able to test it, because my proxmark thinks the HF antenna is broken (0.07V). If I use the RDV4.0 fork my antenna is just fine.

Do you know if #812 was merged into the RDV4.0 as well? Because Mifare Classic simulation now works with RDV4.0 software base.

pwpiwi, 2019-04-26:
Yeah, kind of. The issue with wrong Parity Bit encryption was raised by @McEloff in the forum. Fix was then implemented in parallel (😦) for each repository.

pwpiwi, 2019-04-26:
> However, I was not able to test it, because my proxmark thinks the HF antenna is broken (0.07V). If I use the RDV4.0 fork my antenna is just fine.

Huh? Same hardware with different forks giving different voltages?

ikarus23, 2019-04-26:
>Huh? Same hardware with different forks giving different voltages?

Exactly! Very strange.

pwpiwi, 2019-04-27:
Was this some time ago? The RDV4.0 voltage measurement has been introduced with PR #678 in September 2018. Works perfectly today:
```
~/github/proxmark3$ client/proxmark3.exe com6 -w
Waiting for Proxmark to appear on com6 .
Prox/RFID mark3 RFID instrument
bootrom: Version information not available
os: update_mfu/v3.1.0-88-g9d3be09-dirty-suspect 2019-04-24 18:48:29
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 204940 bytes (39%). Free: 319348 bytes (61%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hw tune

Measuring antenna characteristics, please wait.........
# LF antenna: 68.89 V @   125.00 kHz
# LF antenna: 40.01 V @   134.00 kHz
# LF optimal: 69.30 V @   126.32 kHz
# HF antenna: 49.64 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```

ikarus23, 2019-05-05:
No, it does not work with the latest commit:
```
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-401-g53edb04-suspect 2018-09-13 18:14:59
os: master/v3.0.1-542-g9ebbfd8-suspect 2019-05-05 10:12:41
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: not available
          
uC: AT91SAM7S512 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 205007 bytes (39%). Free: 319281 bytes (61%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hw tune

Measuring antenna characteristics, please wait.........          
# LF antenna:  0,00 V @   125.00 kHz          
# LF antenna:  0,00 V @   134.00 kHz          
# LF optimal:  0,00 V @ 12000,00 kHz          
# HF antenna:  0,07 V @    13.56 MHz          
# Your LF antenna is unusable.          
# Your HF antenna is unusable.
```
But it works with the RDV4.0 repository.

pwpiwi, 2019-05-06:
> os: master/v3.0.1-542-g9ebbfd8-suspect 2019-05-05 10:12:41

This should be v3.1.0 instead of v3.0.1. Maybe your local repository is incomplete?

> SmartCard Slot: not available

Which PM3 is this? Is it RDV4 or older?

Is it only the voltage measurements which are not working, or is your antenna indeed unusable?

ikarus23, 2019-05-06:
>This should be v3.1.0 instead of v3.0.1. Maybe your local repository is incomplete?

No, I'm on 9ebbfd898ca6ce4f9e32b64aca3f3972a3a74e2c which is the top of the master right now. Did someone forget to push?

>Which PM3 is this? Is it RDV4 or older?

It's an old classic PM3 but with 512K flash.

>Is it only the voltage measurements which are not working, or is your antenna indeed unusable?

It is only the measurement. My antenna works perfectly once I flash the RDV4.0 fork on the very same setup.

pwpiwi, 2019-05-08:
I did a fresh `git clone` followed by `git tag -l`. The tag v3.1.0 is there.

> It is only the measurement. My antenna works perfectly once I flash the RDV4.0 fork on the very same setup

Sorry, I was not clear enough. Is the antenna usable with this repository, despite the fact that it says 0,00 V ? E.g. does `hf 14a info` work with a respective tag?

ikarus23, 2019-05-08:
>I did a fresh git clone followed by git tag -l. The tag v3.1.0 is there.

Interesting, I never head to use `git tag -l` in the past on any repository I've ever used... However, regardless of the tag, I was up to date (9ebbfd898ca6ce4f9e32b64aca3f3972a3a74e2c). Just a tag mismatch. 

But again, after `git tag -l` the info when starting the client was still 3.0.1.... So I had a look at my origin of the repository. It was still using SSH from the time I head push rights here. Therefore I decided to remove the whole repository and to check it out again using HTTPS.

Aaaaaand it works! I've got 3.1.0 as tag and my antenna is back to live. Still don't know what exactly was causing this.

But: If I just enter `hf mf sim u 11223344` my Nexus 5x is not able to detect anything.

pwpiwi, 2019-05-08:
`git tag -l` just lists the existing tags. There is no need to do it, it is just for information.

Regarding `hf mf sim` and Nexus 5x: Please open a new issue, we have hijacked this one already 😃. Please run `hf mf sim` for a while, then stop it with the PM3 button and post the result of `hf list mf`.

pwpiwi, 2019-05-08:
I have read on your github repository that the Nexus 5x is not a very reliable NFC tag reader...

ikarus23, 2019-05-08:
Ok, didn't had a look at the man page :). The 3.1.0 tag was there. It looks like something different was the issue.

> I have read on your github repository that the Nexus 5x is not a very reliable NFC tag reader...

That is true. But as I tried with the PM3, not even the UID was detected.

Now I unplugged everything and set it up again and it started to work. Don't know why I didn't do this in the first place. This is usually the first thing I try if my PM3 is not working ;) Now the UID gets detected. I sill can't read the tag, but I think this is because the emulator memory is not initialized with a factory formatted tag.

pwpiwi, 2019-05-15:
> I still can't read the tag, but I think this is because the emulator memory is not initialized with a factory formatted tag.

Yes, sometimes it is important that you have the correct data in block 0. Can you now confirm that the simulation works (and we can close this issue)? Otherwise please provide a trace ('hf list mf' after a simulation) here.

ikarus23, 2019-05-21:
So I did a `hf mf eclr` to init the emulator with an empty tag and then `hf mf sim u 11223344`. It did work on my Nexus 7 (2012) and sometimes on my Nexus 5X. Maybe my antenna (or the antenna in the Nexus 5X) is bad.

closed 2019-05-21 by ikarus23