Proxmark/proxmark3 #520: EMV work.

merlokk, 2017-12-13:
I put here some plaining work in EMV area.
This work will be very interesting for people who work with bank cards and terminals. Now there are many commercial card emulators and sample terminals but they costs from $10000USD and plus yearly subscription...
This work will help expand the number of proxmark users and maybe attract new contributors.

EMV1-3 - merged)
EMV4 - SDA,DDA,fDDA,CDA - completed #512 
EMV5 - Separate commands from EMV processing and transaction parameters in `hf emv exec` - I need argtable (#490) to implement them
EMV6 - EMV emulator - started here #513 (https://github.com/merlokk/proxmark3/commits/EMVemulator). Now concept works. Here (EMV6) will be a concept with sending errors on each APDU command. (correctly working prototype)
EMV7 -  working card emulator. I will need to integrate JSON parser (and choose one...).

iceman1001, 2017-12-14:
Wow!  @merlokk , I love your plan,  

EMV4 - with some luck if @pwpiwi finds time,   it would be nice still. 
--------
EMV5 - based on argtable,  hm, that might take some time.
EMV6 - 7  can be made individually.

With the amount of new EMV functions, I doubt there are many who tried them out more than yourself.
Its a bit of learning to do.  Just running commands and look at the output is not enough.
Reading up on it and understand whats happening will take time.

merlokk, 2017-12-21:
As for JSON. I like to try https://github.com/akheron/jansson
What do you think? license MIT

iceman1001, 2017-12-21:
don't know if the MIT license goes with GPL 2.0...

pwpiwi, 2017-12-21:
Yes, it does. And BTW: GPL3 doesn't!

iceman1001, 2017-12-21:
MIT goes with GPL2.0,  but not GPL3.0??  strange,  all these licenses..  Maybe ask Akheron for a license that fits for proxmark3 project?  I doubt that they dislike it.

pwpiwi, 2017-12-21:
I meant: GPLv3 doesn't go with GPLv2.

iceman1001, 2017-12-21:
Not sure about that,  
_if a user wants to combine code licensed under different versions of GPL, then this is only allowed if the code with the earlier GPL version includes an "or any later version" statement_

problem is all proxmark3 files has a somewhat incoherent GPL2 statements,  there not just one license file.   And then, the licens upholder (Westhues)  or (akheron) would need to talk with eachother complaining that the proxmark3 project is violating the license.    Long time I saw Westhues here.
It would come down to akheron,  to complain to Westhues to change the problems which might occured.

I strongly doubt someone is interested that.   And as a suggestion earlier,  if the worry of the open source license,  someone could ask the license holder of  **jansson** json lib and ask for a version of their license which would work with GPL2.0,  just like Adam Laure granted us.     

These discussions tends to be theoretical threats more than realistic ones.    So if you feel the GPL 3.0 parts of the current repo doesn't go with GPL2.0,  do please bring it up with Jonathan Westhues.

pwpiwi, 2017-12-22:
Off topic. Opened new issue #527.

iceman1001, 2017-12-27:
@merlokk   suggestion;  https://github.com/peterfillmore/emvkeytool  ,  dumping  found keys into a *.pem file?   And test key for ROCA  weakness,   https://github.com/crocs-muni/roca

merlokk, 2018-01-03:
ill add dumping keys to EMV8. as for test ROCA - i see that all the code written on python( so maybe it have no sense to rewrite on C and include it to proxmark. it just needs to have this python file to check key.

merlokk, 2018-10-11:
online test key for ROCA weakness https://keychest.net/roca#/
and its OK to put there key from json file in plain HEX format

pwpiwi, 2019-04-25:
What about your plans? All done? Can we close this "todo list"?

merlokk, 2019-04-25:
lets close. ill make emulator later( now i have no time(

closed 2019-04-25 by merlokk