Proxmark/proxmark3 #527: GPL License compatibility issues

pwpiwi, 2017-12-22:
Copied from #520:

@merlokk wrote:
> As for JSON. I like to try https://github.com/akheron/jansson
> What do you think? license MIT

@iceman1001 wrote:
> don't know if the MIT license goes with GPL 2.0...

@pwpiwi wrote:
> Yes, it does. And BTW: GPL3 doesn't!

@iceman1001 wrote:
> MIT goes with GPL2.0, but not GPL3.0?? strange, all these licenses.. Maybe ask Akheron for a license that fits for proxmark3 project? I doubt that they dislike it.

@pwpiwi wrote:
> I meant: GPLv3 doesn't go with GPLv2.

@iceman1001 wrote:
> Not sure about that,
> if a user wants to combine code licensed under different versions of GPL, then this is only allowed if the code with the earlier GPL version includes an "or any later version" statement
> 
> problem is all proxmark3 files has a somewhat incoherent GPL2 statements, there not just one license file. And then, the licens upholder (Westhues) or (akheron) would need to talk with eachother complaining that the proxmark3 project is violating the license. Long time I saw Westhues here.
> It would come down to akheron, to complain to Westhues to change the problems which might occured.
> 
> I strongly doubt someone is interested that. And as a suggestion earlier, if the worry of the open source license, someone could ask the license holder of jansson json lib and ask for a version of their license which would work with GPL2.0, just like Adam Laure granted us.
>
> These discussions tends to be theoretical threats more than realistic ones. So if you feel the GPL 3.0 parts of the current repo doesn't go with GPL2.0, do please bring it up with Jonathan Westhues.

pwpiwi, 2017-12-22:
> if a user wants to combine code licensed under different versions of GPL, then this is only allowed if the code with the earlier GPL version includes an "or any later version" statement

It is not that easy. See the compatibility matrix in https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility

> problem is all proxmark3 files has a somewhat incoherent GPL2 statements

This is not a problem but is to be expected in open source development. There are many authors in a project and everyone can choose the license conditions for his/her piece of code.

>  there not just one license file

There is one license for the proxmark project. See LICENSE.txt. The Proxmark3 repository is currently licensed under GPLv2. The README.md allows distribution under GPLv2 or later.

> I strongly doubt someone is interested that. 

I AM interested. I am not a lawyer but I would think that repository admin @W8M2Hg9lLmWqXSGC and even more those merging code are accountable and responsible for license violations. Therefore checking license conditions is part of my code reviews.

Coming back to the original question, see https://www.gnu.org/licenses/license-list.en.html
> Expat License (#Expat)
> This is a lax, permissive non-copyleft free software license, compatible with the GNU GPL. It is sometimes ambiguously referred to as the MIT License.

iceman1001, 2017-12-22:
So pm3 project allows for GPL3.0 with the current statement of allowing later ones.  
Of course the GPL3.0 must befollowed when it come to the GPL3.0 code.

Introducing new code under a laxer license,  well,  that might not work as you already suspect.
If you want to use MIT for your contributions,  for that concern, another license,  why not just have a free software license?  Released under as free software in total.

However,  sadly, as I mentioned,  the licenses is not that easy going in pm3.  For example the loclass source code is only GPL2.0 without the extra notion  "of any later version",  effectivly blocking everything down to GPL2.0.  https://github.com/Proxmark/proxmark3/blob/41f08b7c08928060a8dd84916fddf294fab7f26c/client/loclass/cipher.h
Thus making all GPL3.0 additions since that code,  impossible.  

Then we have  roel license here,
https://github.com/Proxmark/proxmark3/blob/master/common/usb_cdc.c

I am sure there are more,  maybe compile a list of the different licenses used already in PM3 would actually be helpful in your task of finding compatible licenses for new source code to be added.

iceman1001, 2017-12-22:
And I also belive you have to compile a list of all copyright owners in the project,  in order to know which ones to contact when the violations occure,  which also needs to be maintained.   

And there are public domain code aswell.    This will a busy Christmas.

iceman1001, 2017-12-22:
My approach is that the indivual copyright holders are responisble to keep track of when violations are made and notify according to the license used in order for compliancy.  

Its none of my business.  

You can also start complaining to RyscCorp for breaking the PM3 GPL2.0,  by not sharing their source code for their pm3 firmware they make / sell their devices with.   We have seen posts on the forum about non-working firmware,  Not to mention those oscbur chinese firmware which other pm3 devices are sold with.   So, are the maintainers responsible for that too?    In my view no.  You might have different view of it.

merlokk, 2017-12-22:
as I see:

```
MIT License
A short and simple permissive license with conditions only requiring preservation of copyright and license notices. Licensed works, modifications, and larger works may be distributed under different terms and without source code.

```

so we can do anything with this code except of deleting copyright in the files. 
so we can use code with this license what do you think?

as for code written by contributors - if someone writes code into repository with license - he(she) accepted this license. if he dont write his license into PR.

iceman1001, 2017-12-22:
If you show me a legal predujacte stating that the legal acceptans of a code-commit overrules your own license,  I will belive it.  Still, that would be only valid in one country and might as well be different in other countries.   

This makes the open-sourced idea quite funny,   it was suppose to take away closed-source and enable end users to own the right to change the source code themself,   but it turned out to be a more complex legal matter than before.  Every time ppl talk about their concerns of open-source makes me chuckle inside.   The real complexity in the legal perspective is huge.   Try to claim your rights,  as the original open source license holder against companies in different countries,   try taking them to court.   It can and most likely take years of effort and costs.   But it also gives the original holder the possibilty to say,  I don't have to offer support,  you do it yourself.   

We are not even talking about a software patent either.

merlokk, 2017-12-22:
https://en.wikipedia.org/wiki/MIT_License
 The MIT license is also compatible with many copyleft licenses, such as the GNU General Public License (GPL); MIT licensed software can be integrated into GPL software, but not the other way around.

so its ok

iceman1001, 2017-12-22:
We already have MIT-licensed source code in PM3..
Liblua is released under MIT.  

https://github.com/promark/proxmark3/blob/master/liblua/lua.h

iceman1001, 2017-12-22:
The idea that liblua now would be accepted as GPL2.0  because it was commited into PM3 is clearly not valid.

pwpiwi, 2017-12-22:
@iceman1001 wrote:
> You can also start complaining to RyscCorp for breaking the PM3 GPL2.0, by not sharing their source code for their pm3 firmware they make / sell their devices with. We have seen posts on the forum about non-working firmware, Not to mention those obscure chinese firmware which other pm3 devices are sold with. So, are the maintainers responsible for that too? In my view no. You might have different view of it.

No, I don't think that the maintainers are responsible if other people violate the PM3 license conditions. And I didn't say that.

I said that I think that maintainers are responsible if they merge code written by other authors and thereby violate the other author's license conditions. Therefore it is absolutely correct to assess the JSON libraries license conditions. I agree with @merlokk that @akheron's code can be legally included because its MIT license is compatible with GPLv2.

> The idea that liblua now would be accepted as GPL2.0 because it was commited into PM3 is clearly not valid.

Indeed, this is not valid. liblua still remains licensed under MIT license. But the MIT license allows to use the code in a GPLv2 licensed program.

pwpiwi, 2017-12-22:
> For example the loclass source code is only GPL2.0 without the extra notion "of any later version", effectivly blocking everything down to GPL2.0.

Correct. We cannot choose the "or later" option for PM3 any more without getting rid of the loclass code.

> Thus making all GPL3.0 additions since that code, impossible.

Not only since that code. Even without that code we cannot add GPLv3 code to PM3 because it is not compatible with the current GPLv2.

micolous, 2017-12-26:
I ran a reporting tool (scancode-toolkit), but this doesn't seem to pick up some of the GPLv2 only vs. GPLv2+ distinction in many cases.  It will reduce the sample size a fair bit though if you want to audit this stuff.  However, `client/reveng` is all GPLv3+ code which contradicts both the GPLv2+ license stated for the rest of the project, and the GPLv2 only license stated for `loclass`.

This also identifies all the files that are missing license/copyright headers.

I've taken the action for now to remove all the offending (`loclass`) code from my Android port (AndProx) with the two commits above.  That would make it GPLv3+ again (because `client/reveng` makes Proxmark3 GPLv3+, and AndProx itself is GPLv3+).

There are some bitwise operations used from that `loclass` module used in non-IClass contexts. However, they appear to have been partially included by the author directly in PM3 in a file with "GPLv2+" license header, and then later removed.  I've "recovered" that functionality from those older git revisions, but this is not complete.

I think that bitwise operations and IClass support are two separate concerns, so it would be good if we can at least get the license changed to GPLv2+ for at least the bitwise operations.

`uart_win32.c` is flagged as a GPLv3 file, because I found some version of this code licensed under (L)GPLv3+ in libnfc and added a note about it when doing the POSIX/Windows split of uart modules (in #341).  It looks like this was added to the project by Roel under BSD-3-Clause in https://github.com/Proxmark/proxmark3/commit/5bcc76c427b680e28c001783c29c695e3a8d0c17.  The other hit I got for this was https://github.com/nfc-tools/libnfc/commit/b110b5b5a904fc48db27edc7d71569e5fe17acd7 (as GPLv3+), but was later relicensed under LGPLv3+.

iceman1001, 2017-12-26:
I have spoken with @holiman  and he is willing to modify his license for the proxmark3 project,  to use the term  "any later version".   At least his part will not lock down the pm3. Very kind of him.

holiman, 2017-12-26:
Done, LMK if that's alright. 

Also, fwiw, I hereby grant the proxmark project perpetual license on all the loclass stuff.

pwpiwi, 2017-12-28:
@micolous: thanks for the analysis. Didn't notice the reveng issue yet.

After loclass has been sorted out I do see reveng as remaining issue and the following options to mitigate:
1. Remove the command completely
2. Remove the sources only. Change the command to call reveng as an external program.
3. Ask the reveng author to change reveng's license conditions
4. Ask all contributing authors to change the Proxmark license conditions to "GPLv3 or later"

marshmellow42, 2017-12-28:
I doubt reveng is a widely used option inside the pm3 client, so I see nothing wrong with removing it.  The second option would be a nice option but then the binary would have to be present.

iceman1001, 2018-01-02:
Not really finished with iClass.
@holiman   it turns out you have the armsrc/optimized_cipher.c  / .h  files  also as GPL2..

micolous, 2018-01-02:
I never build armsrc bits in my fork. It's only relevant for the firmware,
which is not distributed in the APK.

iceman1001, 2018-01-25:
not yet,  see my previous post.

pwpiwi, 2018-01-31:
Updated armsrc/optimized_cypher.[ch] as well (PR #554).

iceman1001, 2018-01-31:
not really...
Either you use the **I hereby grant the proxmark project perpetual license on all the loclass stuff."**  license or you use the old one.   You can not make up your own.

pwpiwi, 2018-01-31:
It is not my own. It is holiman's.

iceman1001, 2018-01-31:
No,  its not.  Holimn didn't change that file,  hence it follows its licens gpl2.0 
We are not granted the right to create our own license, like you just did.

pwpiwi, 2018-01-31:
Please have a closer look at 26f202e.

iceman1001, 2018-01-31:
And I don't see any text stating that we are granted the right to change license anywhere.

pwpiwi, 2018-01-31:
Edited my last post. See holiman's commit.

iceman1001, 2018-01-31:
link not working.

pwpiwi, 2018-01-31:
Mobile travelling. Check holimans loclass github repository.

iceman1001, 2018-01-31:
The idea of editing posts is also interesting,  it makes whole conversations quite hard to follow,  since the answers/comments now are totally out of context.   Still,  in the commit on loclass,  nothing that I have asked about has been answered.

pwpiwi, 2018-01-31:
That's because you are too fast with reading and answering.  :smile: Will answer when back at PC.

iceman1001, 2018-01-31:
To recapture:   https://github.com/holiman/loclass/commit/d04d51a25b673caf9af1310f774e1981bfec7350
or in this thread. 

Where in that commit do you:

- see holiman changed those files you changed license for?
- see a message stating it should include all relevant files if forgotten?
- see any text stating that we are granted the right to change license regarding loclass?

iceman1001, 2018-01-31:
Speaking of changing licenses,  remember this one?   http://www.proxmark.org/forum/viewtopic.php?pid=19460#p19460

holiman, 2018-01-31:
Hey!

See https://github.com/holiman/loclass/issues/1  and the commit: https://github.com/holiman/loclass/commit/d04d51a25b673caf9af1310f774e1981bfec7350#diff-6d4aaeff70a9b9ed500a162854a5355b .

iceman1001, 2018-01-31:
@holiman  you need to merge that PR so we can put this one to the history books.

holiman, 2018-01-31:
I did, on December  26. Don't know what all the confusion is about... The license update covered all files.

holiman, 2018-01-31:
So

>Where in that commit do you:
>
> * see holiman changed those files you changed license for?
https://github.com/holiman/loclass/commit/d04d51a25b673caf9af1310f774e1981bfec7350#diff-6d4aaeff70a9b9ed500a162854a5355b . 

>  *see a message stating it should include all relevant files if forgotten?

None were forgotten :)

iceman1001, 2018-01-31:
Well, if you conside your file under "loclass/folder" ,   to be the ones related to armsrc/ ones,  we almost be fine,   with a message saying "Swapped to the files with updated license from holiman/loclass/ folder,  given refence [insert link here]."  However, since those two files doesn't exist in PM3 client/loclass folder,  the PR #554   just look like we are changing licenses without granted rights. 

About finding the granted right under a closed issue where posts can be edited,  is less optimal than what I would prefer. 

As I see it, possible solutions is:
- insert PR message which points to your commit stating its right to update license
- you merge  PR #554
- you post a message to PR #554 saying this PR is ok.

I'm fine with any of them.   The last one is the simplest one.

iceman1001, 2018-01-31:
Perfect!  Thanks   @holiman!  problem resolved.  
@pwpiwi   you can close this thread now :)

closed 2018-01-31 by pwpiwi