Proxmark/proxmark3 #530: hf mf chk dump issue

Tom5ive, 2017-12-26:
Hi All,

I think I found a problem when trying to dump keys to a .bin file while using a known key with hf mf chk. See below output. The file is not actually locked as it can be written with the first command.

proxmark3> hf mf chk *1 ? d
No key specified, trying default keys          
chk default key[ 0] ffffffffffff          
chk default key[ 1] 000000000000          
chk default key[ 2] a0a1a2a3a4a5          
chk default key[ 3] b0b1b2b3b4b5          
chk default key[ 4] aabbccddeeff          
chk default key[ 5] 1a2b3c4d5e6f          
chk default key[ 6] 123456789abc          
chk default key[ 7] 010203040506          
chk default key[ 8] 123456abcdef          
chk default key[ 9] abcdef123456          
chk default key[10] 4d3a99c351dd          
chk default key[11] 1a982c7e459a          
chk default key[12] d3f7d3f7d3f7          
chk default key[13] 714c5c886e97          
chk default key[14] 587ee5f9350f          
chk default key[15] a0478cc39091          
chk default key[16] 533cb6c723f6          
chk default key[17] 8fd0a4f256e9          

To cancel this operation press the button on the proxmark...          
--o          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|001|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|002|  a0a1a2a3a4a5  | 1 |  ffffffffffff  | 0 |          
|003|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|004|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|005|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|006|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|007|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|008|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|009|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|010|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|011|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|012|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|013|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|014|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|015|  ffffffffffff  | 0 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|          
Found keys have been dumped to file dumpkeys.bin. 0xffffffffffff has been inserted for unknown keys.          
          
proxmark3> hf mf chk * A 2a2c13cc242a
chk key[ 0] 2a2c13cc242a          

To cancel this operation press the button on the proxmark...          
--o          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|001|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|002|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|003|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|004|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|005|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|006|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|007|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|008|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|009|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|010|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|011|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|012|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|013|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|014|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|015|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |          
|---|----------------|---|----------------|---|          
          
proxmark3> hf mf chk * A 2a2c13cc242a d
chk key[ 0] 2a2c13cc242a          
File: d: not found or locked.          
proxmark3>

Tom5ive, 2017-12-26:
Ignore this please I was unaware of how exactly the command worked

closed 2017-12-26 by Tom5ive

iceman1001, 2017-12-26:
the command parsing of chk is nasty.   So there are quirks when some params doesnt get to do what you think it should. Still, if you use the establish params in samples,  it should work as expected.