Proxmark/proxmark3 #535: Ubuntu 12

peseta3, 2017-12-30:
Hallo together work the PR3 with Ubuntu 12 together?

iceman1001, 2017-12-31:
You would need a GCC with a version of v4.9 or higher in order to compile.  Not sure if that exists on Ubuntu12.

peseta3, 2017-12-31:
Hallo Iceman,

Thank you for your fast post.
I installed the GCC v4.9.
after I make this:
sudo apt-get install p7zip git build-essential libreadline5 libreadline-dev libusb-0.1-4 libusb-dev libqt4-dev perl pkg-config wget libncurses5-dev gcc-arm-none-eabi libstdc++-arm-none-eabi-newlib

I get This:

root@peseta3-Lenovo-B50-50:~# sudo apt-get install p7zip git build-essential libreadline5 libreadline-dev libusb-0.1-4 libusb-dev libqt4-dev perl pkg-config wget libncurses5-dev gcc-arm-none-eabi libstdc++-arm-none-eabi-newlib
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut       
Statusinformationen werden eingelesen... Fertig
E: Paket gcc-arm-none-eabi kann nicht gefunden werden
E: Paket libstdc++-arm-none-eabi-newlib kann nicht gefunden werden
E: Mittels regulärem Ausdruck »libstdc++-arm-none-eabi-newlib« konnte kein Paket gefunden werden
root@peseta3-Lenovo-B50-50:~#

some thing is wrong.........

peseta3, 2017-12-31:
oh the Text is line through, I don know why.

iceman1001, 2017-12-31:
well,  did you read the output?   

E: Paket gcc-arm-none-eabi kann nicht gefunden werden
E: Paket libstdc++-arm-none-eabi-newlib kann nicht gefunden werden

You will need to install a working gcc-arm-none-eabi tool chain

peseta3, 2017-12-31:
Yes I read it.
I install it with

> sudo apt-get install gcc-arm-none-eabi
sudo apt-get install libstdc++-arm-none-eabi-newlib

> 
 but  the packet is not found

iceman1001, 2017-12-31:
And there is your problem

peseta3, 2017-12-31:
yes thats right.
Thank you.
happy new year.

closed 2017-12-31 by peseta3