Proxmark/proxmark3 #541: hf mf sniff - only pick up halt and anticoll

iceman1001, 2018-01-10:
Since the posts on forum about **hf mf sniff** not working I decided to test it.
I'm running on latest source from today, 20180110,   

The two below outputs,  I have tried sniffing between a chameleon mini and a valid reader.
It 

`proxmark3> hw ver
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v2.2.0-739-g91e0384-dirty-suspect 2017-12-27 14:16:43
os: master/v2.2.0-755-g2d51da3-dirty-suspect 2018-01-10 13:47:03
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 199860 bytes (38%). Free: 324428 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
`
**hf mf sniff l d e**
`anticollision: 2018-01-10 14:17:00
RDR: 50 00 57 cd 
TAG: 50 00 57 cd 

anticollision: 2018-01-10 14:17:00
RDR: 50 00 57 cd 

anticollision: 2018-01-10 14:17:00
RDR: 50 00 57 cd 

anticollision: 2018-01-10 14:17:00
RDR: 50 00 57 cd 

anticollision: 2018-01-10 14:17:00`

`anticollision: 2018-01-10 14:17:57
RDR: 50 00 57 cd 

anticollision: 2018-01-10 14:17:57
RDR: 50 00 57 cd 

anticollision: 2018-01-10 14:17:57
RDR: 50 00 57 cd 
TAG: 50 00 57 cd 

anticollision: 2018-01-10 14:21:59
anticollision: 2018-01-10 14:21:59
anticollision: 2018-01-10 14:21:59
anticollision: 2018-01-10 14:21:59`

Compared with the output with which collects much more.
**hf 14a snoop c**
https://pastebin.com/EWWBWGmf

merlokk, 2018-01-11:
my sniff cant pick any command at all

merlokk, 2018-01-12:
i have checked it and see that all is ok
```
received trace len: 67 packages: 1
tag select uid:01 02 03 04  atqa:0x0002 sak:0x18
RDR(0):61 00 2d 62
TAG(1):87 3e 4c 6b b3 a8 aa 40
RDR(2):88 78 4b 12
............>
received trace len: 42 packages: 1
TAG(3):b8 ec 05 c1
RDR(4):8d 9d d7 ea d3 4d 81 59 82 99 e9 d0 71 e1 70 6a ec 15
.........>
received trace len: 13 packages: 1
TAG(5):88 b7 67 e9
```

Maybe this behavior because of fast issuing commands

pwpiwi, 2018-01-12:
Tried as well with current release (genuine card and genuine reader). No issues:
```
proxmark3>  hf mf sniff
-------------------------------------------------------------------------
Executing command.
Press the key on the proxmark3 device to abort both proxmark3 and client.
Press the key on pc keyboard to abort the client.
-------------------------------------------------------------------------
.Waiting for a response from the proxmark...
Don't forget to cancel its operation first by pressing on the button
.....>
received trace len: 201 packages: 1
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(0):e0 87 8e 07
TAG(1):04
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(2):60 00 f5 7b
TAG(3):49 bc 30 01
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(4):60 05 58 2c
TAG(5):ee 8e ff 60
RDR(6):4c ef c8 e9 a4 fb 1b c6
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(7):50 00 57 cd
.....>
received trace len: 621 packages: 2
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(8):e0 87 8e 07
TAG(9):04
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(10):60 00 f5 7b
TAG(11):88 09 97 92
RDR(12):2c 72 d7 3a 82 54 f8 67
TAG(13):35 ad 38 a2
RDR(14):fb 68 20 85
TAG(15):b7 16 aa 4f 32 59 49 7d 11 0f c5 82 41 12 29 5e 54 f6
RDR(16):dd bf 7e 7c
TAG(17):3d d1 29 c0 fb 5c df bf 47 4a 3a 77 bd 6c 1c e5 4d 84
RDR(18):5a 55 cd c3
TAG(19):17 48 34 61 49 8f 09 df c0 bf 73 03 3a 85 f7 2b 98 de
RDR(20):6e 62 e1 04
TAG(21):1e 59 be 5f af 9f bd f1 4e 4f e6 f4 2f fd 56 c0 8c 1a
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(22):60 05 58 2c
TAG(23):70 bd 69 a8
RDR(24):2d 7d 67 1a dc 7a 0d 62
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(25):50 00 57 cd
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(26):50 00 57 cd
```
[...]
```
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(34):60 00 f5 7b
TAG(35):3a d3 0a 6b
RDR(36):45 ad 9c fb 6c 29 f6 e8
TAG(37):fd 35 4e 95
RDR(38):03 bd d1 0f
TAG(39):ec 35 06 d1 87 62 6f 27 03 56 e5 b0 e9 99 8a 54 e8 f8
RDR(40):c0 84 80 3a
TAG(41):48 ad b3 10 81 42 f8 29 31 d9 a6 ac 65 a6 0c 12 ac 07
RDR(42):3d 61 f3 d5
TAG(43):e2 c4 4a 48 90 f0 80 3c 37 97 7b 0e 9a 7f 1b ad 0b c3
RDR(44):34 9c e9 9d
TAG(45):ea 04 87 54 f8 b4 e2 c7 3e 1c bc f4 f6 ce dd 86 c4 3b
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(46):60 05 58 2c
TAG(47):2d 03 9a 13
RDR(48):8a 06 31 3c d5 f8 ba 84
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(49):50 00 57 cd
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(50):50 00 57 cd
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(51):50 00 57 cd
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(52):50 00 57 cd
tag select uid:xx xx xx xx  atqa:0x0004 sak:0x08
RDR(53):50 00 57 cd
.Waiting for a response from the proxmark...
Don't forget to cancel its operation first by pressing on the button
...
aborted via keyboard!
proxmark3>
```

merlokk, 2018-01-12:
I see that not works NXP taginfo from Android phone. It seems it have  not a big pauses between commands. 
I look on monday but it seems it cant work with current hardware. It too slow. 
Or maybe current setup have too much pauses between transfers

pwpiwi, 2018-01-12:
Don't mix up `hf mf sim` with `hf mf sniff`. If you use both, you cannot tell which is broken.

merlokk, 2018-01-12:
I see this behavior in trying to read generation card from Android taginfo.
P. S. I thought that maybe it because not so strong field...

merlokk, 2018-01-15:
and i see another strange behavior....   if i send good auth command - all is ok, but if i send auth command with wrong key - `hf mf sniff` dont show any trace....

iceman1001, 2018-01-15:
That would be the behavior on the clientside  **mfTraceDecode**  where all failure sets the traceState = TRACE_ERROR  and it will no reset that state until an new anticollision is done with success...

Its a bit strict in that sense, it doesn't just sniff and log it.  It tries to follow the statemachine aswell and act there after.   The  **hf 14a sniff**  is in that sense better at sniffing and not interpret the data.

pwpiwi, 2018-01-15:
Yes, @iceman1001 is correct. hf mf sniff is mostly used to decode commands and responses after a successfull authentication. The rest is ignored.

It might be a good idea to connect it to mfkey...

iceman1001, 2018-01-15:
Wasn't it @merlokk  who commited this code 2012?!? Why am I trying to explain it to him? 

**hf mf sniff** has had recovery of key functionality since long

pwpiwi, 2018-01-16:
Indeed. I am getting old. I had forgotten.

pwpiwi, 2018-01-16:
Back to the original topic: This seems not to be a PM firmware bug. Antenna, Android, Chameleon Mini, whatever....

merlokk, 2018-01-16:
I dont understand why it works not as should.
it should transfer all data except of anticollision.
it transfers it if authentication ok, but not transfers if auth is failed. I dont see this behavior in code. in code it just have to transfer all the commands.

and maybe core topic is not a PM bug, but this bug looks very strange and may connected with topic

iceman1001, 2018-01-16:
My opinition when comparing between the two implementations
**hf 14a snoop**
**hf  mf sniff**
They uses both same data collecting dmabuffer but where the 14a snoop works as expected,  and mf sniff doesnt, it rules out other possible idea like antenna, card or chameleon mini.
Same setup,  where one work and the other doesn't.

I have look at the difference,  which is that mf sniff interupts and sends data to client,  then restarts collecting,  and 14a snoop uses only logtrace to save data.

So 14 snoop works,  hence,  logtrace isn't it,   the difference is in the interupt & send.

iceman1001, 2018-01-16:
Not to forget,  that once you run the mf sniff , and rerun it,  it almost never works.  Not even getting anticollisions & halt data.

iceman1001, 2018-01-16:
@merlokk  didn't work.  Still anti + halt.

merlokk, 2018-01-16:
fixed.
7b UID fixed too )))

iceman1001, 2018-01-16:
I'm so happy @merlokk  took this one seriously,  instead of opting for the easy way out.
Your latest fixes brings back the life in my  **hf mf sniff** command.  

Now the output to both screen and logfile look like crap and could do for a overhaul aswell but I am happy with a working command.

iceman1001, 2018-01-16:
@merlokk    while you are in a flow,   https://github.com/iceman1001/proxmark3/issues/155 ,  some more bugs that exits ;)

iceman1001, 2018-01-17:
btw,  **hf mf sniff** looks like it has only partial command support.
Doesn't deal with Increment, Decrement, Restore, Transfer,    or (Classic Ev1 Set_Mode_type, Personalize_uid_usage) 

nor does it nicely deal with not offical commands like the magic commands..

Since readers nowdays both uses ClassicEv1 cmds and clone detections,  these commands shouldn't "break" the transactions.

When looking today at **hf mf sniff** it kind of sucks at doing what it should do.   Who would use the command in its current state and think its good or even working well?

pwpiwi, 2018-01-17:
It supports only authentication, Read, and Write. But that is all on client side and can easily be extended. Let's finish the mods on device side first.

iceman1001, 2018-01-17:
Those doesn't mutually exclude eachother.  I see no reason either of them should be postponed.

iceman1001, 2018-01-17:
hm, the more I look and play with **hf mf sniff** I can't see why. 
Lets play with an idea.  The  **hf 14a snoop** works the same.  And we get a nice tracelog.
**hf mf sniff**  can decode part of the trace and save to a eml file..   But all of that code is on clientside.

What about we make a **hf list mf**  based on the clientside **hf mf sniff** code.  It can find uid,  decrypt key,  list,  save to eml,  even try decrypting nested auth?

merlokk, 2018-01-17:
as for eml i dont think that it `hf list` functionality. maybe separate command?
lets merge this and then try to get work nested and then think

closed 2018-01-17 by iceman1001

iceman1001, 2018-01-17:
the save to eml functionality is also an odd option.  Thinking in a use-case scenario,  once you sniff you usually have access to the card anyway.  You sniff to get one key,  in order to get the rest.

Making a eml-dumpfile from parts of a sniffed trace seems not optimal.  How often does the reader read the whole card?  
Saving a individual log trace functionality,  is interesting when its a decrypted session. 
otherwise we have the normal tracelog. 

So **hf mf sniff**  ->  **hf list mf**  can take away eml-option,  and have a option where it "decrypts" the session instead.  With key found from trace (normal auth and nested auth)  and supplied key?...

I'm going to remove **hf mf sniff** from iceman fork. Will save some space on the device side.

merlokk, 2018-01-17:
is this https://github.com/iceman1001/mf_nonce_brute/blob/master/mf_nonce_brute.c#L240 code brute-forces all the range?
or it need to use hardnested?

iceman1001, 2018-01-17:
mf_nonce_brut is based on the weak prng.  
you will not be able to use hardnested against collected nonces from a sniffed trace,  since they are too few. 
That is why I suggest a optional key parameter for **hf list mf**  in order to decrypt it,    not only from authentications found from the sniffed trace.

pwpiwi, 2018-01-18:
> mf_nonce_brut is based on the weak prng

after a short glance there is code vor EV1 (strong prng) cards. Did you test?

iceman1001, 2018-01-18:
No, never tested against a hardend prng card.

merlokk, 2018-01-18:
EV1 just here https://github.com/iceman1001/mf_nonce_brute/blob/master/mf_nonce_brute.c#L140
```
	if (!ev1) {
		//1st (1st nt)
		byte = (nt >> 24) & 0xFF;
		check = odd_parity(byte) ^ ((nt > 16) & 1) ^ ((xored >> 9) & 1);
		if(check) return false;
			
		//2nd (2nd nt)
		byte = (nt >> 16) & 0xFF;
		check = odd_parity(byte) ^ ((nt >> 8) & 1) ^ ((xored >> 8 ) & 1);
		if(check) return false;
	}
```

Its just several check. hardnested needs to have another bruteforce rules? not just checks?

iceman1001, 2018-01-18:
The hardned prng will not apply to the weaked prngs which are looped in mf_nonce_brute.c  
well, some of them are,  but I belive the weakend is just a subset of the hardned.

pwpiwi, 2018-01-18:
Correct. There are only 2^16 different "random" numbers in the weak PRNG but 2^32 in a "good" PRNG. However, the reader's answer (ar) and the tag's answer (at) are defined as PRNG successor states of the initial tag nonce (nt). These successor states are always from the "weak" subset. The mf_nonce_brute therefore has to try 2^16 states only - regardless of the tag having a weak or good PRNG.

BTW: I am quite sure that there is room for improvement in mf_nonce_brute.

iceman1001, 2018-01-18:
Were where you when I was looking at it to explain? 
if j-runs attack finds its way into the pm3 client,  I'm sure you will find great improvements.
I was quite happy to get it down to 4sec and verify crc, commands in order to find correct solution.

J-run made a two-part attack.  mf_nonce_brute is the first part.  Sometimes you get lucky and find the key. No need for the  second part,  which is an online attack against the card where it tried to find the last 0xFFFF possible keys.    This part is also in iceman fork client.   It should actually benefit from the improved checkkeys implementation I did...      But I never saw any usuage of J-runs attack on the forum...

Until yesterday,  did was kind of empty... but with a  **hf list mf** it could bring usefulness to the PM3 client.

iceman1001, 2018-01-18:
...I also had an idea of take advantage of acid's solver,  alas I am not good enough to adapt it.

pwpiwi, 2018-01-19:
> Were where you when I was looking at it to explain?

Didn't see a use case

>  Sometimes you get lucky and find the key. No need for the second part

Probably the difference between broken and hardened PRNG

iceman1001, 2018-01-19:
...it was a rethorical question

...could be like that, could also be because ar/nr is 32bits and key is 48

pwpiwi, 2018-01-19:
Did you ever run the second part? Is it feasible to try 65536 keys online? Is this faster than hardnested?

iceman1001, 2018-01-19:
I think current speed is 65key / s.    Which would give 65536  / 65 / 60 = 17min   Doable but way to long compare with hardnested.  
It raises the question if hardnested can be adjusted to take advantage of the know 32bits?

commit 3544b997 (2018-01-17): fixed #541