Proxmark/proxmark3 #542: Compiling on RPi ARM fails

magdesign, 2018-01-11:
Trying to compile proxmark3 on RaspberryPi3 on Debian Jessie, get following error:

https://gist.github.com/magdesign/05baa05c1b4c20258e42ac4aa0e25abf

Is there anything special I have to do to get it done on ARM?

iceman1001, 2018-01-11:
It would be a compiler directive #define which doesn't match Raspberry Jessie when decide which cpu op-codes to use.    RPI3,  is ARM isn't it?    So it isn't __linux__  but __arm__

pwpiwi, 2018-01-12:
I don't think so. Obviously hardnested_bitarry_core.c still compiles. I would therefore assume it is @merlokk's modified dispatch functions in hardnested_bf_core.c?

iceman1001, 2018-01-12:
it compiles but it doesn't link?
btw,  I don't think you can run hardnested on RPi3..  to little memory.   Will be interesting to hear if you manage to run it.

magdesign, 2018-01-12:
I could not make it run so far on a RPi....Will try it  with merlokk's version.
For the moment using iceman's version on a Fedora notebook.

merlokk, 2018-01-12:
may you check if this commit compiles?
 https://github.com/Proxmark/proxmark3/commit/91ca74f0ed2a6db99df87e7b318660f45b6858f9

merlokk, 2018-01-12:
this lines can show this error
https://github.com/Proxmark/proxmark3/blob/master/client/hardnested/hardnested_bf_core.c#L618

if the error shows)
maybe it needs to add this ifdef
https://github.com/Proxmark/proxmark3/blob/master/client/hardnested/hardnested_bf_core.c#L618
`#if defined (__i386__) || defined (__x86_64__)`