Proxmark/proxmark3 #557: nested in sniff

merlokk, 2018-01-31:
I look into https://github.com/iceman1001/mf_nonce_brute

and see something like:
```
check = oddparity(nt[1]) ^ nt[0] & 1 ^ nt_par[from field at nt[1] ] ^ nt_enc[0] & 1
...
check = oddparity(at[2]) ^ at[1] & 1 ^ at_par[from field at at[2] ] ^ at_enc[1] & 1
check = oddparity(at[1]) ^ at[0] & 1 ^ at_par[from field at at[1] ] ^ at_enc[0] & 1
...
if (check) exit
```

What sense is here? 
Is it can be used as at (and etc) check?

merlokk, 2018-01-31:
It xor:
- nt byte 1 parity
- nt byte 1 parity from field
- nt byte 0 bit 1
- nt encoded byte 0 bit 1 (from field)

merlokk, 2018-01-31:
the crypto ciphers 1 byte and then parity of this byte. so as i remembered this crypto bit the same in nt[0]&1 and in parity(nt[0])
Where i wrong?

iceman1001, 2018-01-31:
hm,  I think the user who updated that part wrote on a commit or issue about it.
Something about partity is linked with first bit of byte since the same crypto bit was used to encrypt it..
and all together they would cancel out eachother...

It should be mentioned inside the darkside papers etc also..

pwpiwi, 2018-02-01:
Did someone try this piece of code? Does it work?

Maybe it is better to write it from scratch? This would eliminate copyright questions as well.

merlokk, 2018-02-01:
Im trying to write it from scratch but dont understand how it works. And if it works(

iceman1001, 2018-02-01:
... don't trust j-run,  jbfuzier or me,  ignore the forum thread,   and try it yourself.

It is good to verify yourself,  its not hard to test it,  since the readme has all need data to run.   Or get a nested auth trace from somewhere and run it.   It compiles out-the-box.

After that, the level of discussion will be better.

iceman1001, 2018-02-01:
btw,  what copyright questions would arise to start with?

iceman1001, 2018-02-01:
@merlokk   I saw you added an issue on j-run repo.  Let hope you get an answer,  that repo is quite stale.
There are major differences between j-run's repo and iceman,   mostly because of this commit.
There are another user ( @andrew892)  who made the improvements. 

https://github.com/iceman1001/mf_nonce_brute/commit/35bd507fa4e97b4eb73fb494d2c0b0b67145ed95

merlokk, 2018-02-01:
https://nethemba.com/resources/mifare-classic-slides.pdf
```
the parity bit is encrypted, but the internal state 
will not shift   the first bit of the next byte will  →
be encrypted by the same keystream bit
```
and here
http://www.proxmark.org/forum/viewtopic.php?id=223

so it looks ok. ill test it)

pwpiwi, 2018-02-01:
> btw, what copyright questions would arise to start with?

The author didn't allow copying and distribution. At least it is not mentioned in the code. Therefore the copyright/copyleft right is unclear at least.

iceman1001, 2018-02-01:
Do feel free to make a GPLv2 verson of it....   but the derivative nature of GPLv2 means that all of the code that is build upon it,  also becomes GPLv2. The only file which is not expressed is the main file which clearly builds on the others.  We are pretty much in the clear, like 99,999%,  however if you have time and effort to make a better and fast implementation that will be just great!

merlokk, 2018-02-02:
looks like it works
https://github.com/merlokk/proxmark3/blob/mfsniff_pl/client/mifarehost.c#L872

merlokk, 2018-02-05:
```
					if (
						(oddparity8(nt1 >> 8 & 0xff) ^ (nt1 & 0x01) ^ ((nt_enc_par >> 5) & 0x01) ^ (nt_enc & 0x01)) ||
						(oddparity8(nt1 >> 16 & 0xff) ^ (nt1 >> 8 & 0x01) ^ ((nt_enc_par >> 6) & 0x01) ^ (nt_enc >> 8 & 0x01)) ||
						(oddparity8(nt1 >> 24 & 0xff) ^ (nt1 >> 16 & 0x01) ^ ((nt_enc_par >> 7) & 0x01) ^ (nt_enc >> 16 & 0x01))
						)
						printf("check nt error\n");
					
					if (
						(oddparity8(ar >> 8 & 0xff) ^ (ar & 0x01) ^ ((ar_enc_par >> 5) & 0x01) ^ (ar_enc & 0x01)) ||
						(oddparity8(ar >> 16 & 0xff) ^ (ar >> 8 & 0x01) ^ ((ar_enc_par >> 6) & 0x01) ^ (ar_enc >> 8 & 0x01)) ||
						(oddparity8(ar >> 24 & 0xff) ^ (ar >> 16 & 0x01) ^ ((ar_enc_par >> 7) & 0x01) ^ (ar_enc >> 16 & 0x01))
						)
						printf("check ar error\n");

					if (
						(oddparity8(ar & 0xff) ^ (at >> 24 & 0x01) ^ ((ar_enc_par >> 4) & 0x01) ^ (at_enc >> 24 & 0x01)) ||
						(oddparity8(at >> 8 & 0xff) ^ (at & 0x01) ^ ((at_enc_par >> 5) & 0x01) ^ (at_enc & 0x01)) ||
						(oddparity8(at >> 16 & 0xff) ^ (at >> 8 & 0x01) ^ ((at_enc_par >> 6) & 0x01) ^ (at_enc >> 8 & 0x01)) ||
						(oddparity8(at >> 24 & 0xff) ^ (at >> 16 & 0x01) ^ ((at_enc_par >> 7) & 0x01) ^ (at_enc >> 16 & 0x01))
						)
						printf("check at error\n");
```

closed 2018-02-09 by merlokk