Proxmark/proxmark3 #559: Proxmark multi sector AUTH

maxben14, 2018-02-03:
How use command proxmark do multi sector AUTH ?
I want after comand "hf mf rdbl 0 A FFFFFFFFFFFF" send " hf mf rdbl 1 A FFFFFFFFFFFF" comand AUTH in 1 sector with multi AUTH and then show log.
How me this do with proxmark ?

pwpiwi, 2018-02-03:
If you mean nested authentication: not yet possible.

pwpiwi, 2018-02-03:
BTW: again a question better suited for the forum.

closed 2018-02-03 by pwpiwi

maxben14, 2018-02-03:
@pwpiwi, i try decrypt trace from proxmark.

I try decrypt trace multi sector AUTH with mfkey64.

	415348140 |  415349132 | Rdr | 52                                                              |     | WUPA
  415350384 |  415352752 | Tag | 04  00                                                          |     |
  415361932 |  415372460 | Rdr | 93  70  4e  7b  b7  0a  88  21  10                              |  ok | SELECT_UID     
  415373648 |  415377168 | Tag | 08  b6  dd                                                      |     |
  415548620 |  415553388 | Rdr | 60  08  bd  f7                                                  |  ok | AUTH-A(8)      
  415555344 |  415560016 | Tag | 85  96  40  a5                                                  |     |
  415564156 |  415573532 | Rdr |e3!  5d  78  90  70 2d! d0! 59!                                  | !crc| ?
  415574720 |  415579392 | Tag | 9d  fb  8c ec!                                                  |     |
  415944524 |  415949292 | Rdr |cb! 04! 9d!  2f                                                  | !crc| ?
  415950480 |  415971344 | Tag | 7c  81  86 ae!  1c 7e! 84! 83! ab! 32!  26  fd 84! a0! 16!  39  |     |
            |            |     | 88  51                                                          | !crc|
  416102812 |  416107516 | Rdr |69! 9b!  a5 19!                                                  | !crc| ?
  416109536 |  416114272 | Tag | e3 27!  bf  9b                                                  |     |
  416117820 |  416127132 | Rdr |84!  41 19! 9a! 63!  2b  4b  8d                                  | !crc| ?
  416128384 |  416133120 | Tag |3c!  17 b0!  cb                                                  |     |
  416252316 |  416257020 | Rdr |95!  71 4a!  ff                                                  | !crc| ANTICOLL-2     
  416258272 |  416279072 | Tag | 9e 38! c9! 56! 5a! dd! aa! 5c! 69! ba!  70  bc 80! 78! bb! 0b!  |     |
            |            |     | 23 89!                                                          | !crc|
  416516812 |  416521580 | Rdr |fc!  17  2d  d0
  
mfkey64 4e7bb70a 859640a5 e35d7890 702dd059 9dfb8cec cb049d2f 7c8186ae1c7e8483ab3226fd84a016398851 699ba519 e327bf9b
MIFARE Classic key recovery - based on 64 bits of keystream
Recover key from only one complete authentication!

Recovering key for:
   uid: 4e7bb70a
    nt: 859640a5
  {nr}: e35d7890
  {ar}: 702dd059
  {at}: 9dfb8cec
{enc0}: cb049d2f
{enc1}: 7c8186ae1c7e8483ab3226fd84a016398851
{enc2}: 699ba519
{enc3}: e327bf9b

LFSR successors of the tag challenge:
  nt' : 4ea81662
  nt'': c11de701
Time spent in lfsr_recovery64(): 0.17 seconds

Keystream used to generate {ar} and {at}:
   ks2: 3e85c63b
   ks3: 5ce66bed

Decrypted communication:
{dec0}: 30084a24
{dec1}: 298106213013402500900000bb3de6d051f0
{dec2}: 600c99b1
{dec3}: cb3b98b9

I correct do decrypt next encrypt Nt ?
encrypt Nt = e3 27 bf 9b
decrypt Nt = cb 3b 98 b9

Next i try decrypt trace after second AUTH.

C:\prox\ProxSpace-master\pm3\tools\mfkey>mfkey64 4e7bb70a cb3b98b9 8441199a 632b4b8d 3c17b0cb 95714aff 9e38c9565addaa5c69ba70bc8078bb0b2389 fc172dd0
MIFARE Classic key recovery - based on 64 bits of keystream
Recover key from only one complete authentication!

Recovering key for:
   uid: 4e7bb70a
    nt: cb3b98b9
  {nr}: 8441199a
  {ar}: 632b4b8d
  {at}: 3c17b0cb
{enc0}: 95714aff
{enc1}: 9e38c9565addaa5c69ba70bc8078bb0b2389
{enc2}: fc172dd0

LFSR successors of the tag challenge:
  nt' : 1e76eaf6
  nt'': 3ac2e23c
Time spent in lfsr_recovery64(): 0.14 seconds

Keystream used to generate {ar} and {at}:
   ks2: 7d5da17b
   ks3: 06d552f7

Decrypted communication:
{dec0}: 95714aff
{dec1}: 9e38c9565addaa5c69ba70bc8078bb0b2389
{dec2}: fc172dd0

Found Key: [28bbc9a40cb3]

Not corect key and incorrect decrypted date.

pwpiwi, 2018-02-03:
I am happy to answer in the forum.

maxben14, 2018-02-03:
@pwpiwi, ok i send this question on forum http://www.proxmark.org/forum/viewtopic.php?pid=31089#p31089