Proxmark/proxmark3 #561: Different behaviour between 2 Ultralight C tags

jpclair, 2018-02-05:
Hi,

I have a different behaviour between 2 tags with the proxmark commands.
The only information I have is that these 2 tags are provided by different companies.

When I do a simple hf mfu info

Tag A:
```
--- Tag Information ---------
-------------------------------------------------------------
      TYPE : MIFARE Ultralight C (MF0ULC)
       UID : 04 35 66 12 00 15 89
    UID[0] : 04, NXP Semiconductors Germany
      BCC0 : DF, Ok
      BCC1 : 8E, Ok
  Internal : 48, default
      Lock : 00 00  - 0000000000000000
OneTimePad : 00 00 00 00  - 00000000000000000000000000000000
```

Tag B:
```
iso14443a card select failed
```

So, it looks like the Tag B is broken.
But there is one thing I can't understand:
I can easily read this tag with the Android App NXP TagInfo (see below), and I can read without any problem the UID with my Springcard Proxnroll Reader.

And I can sniff the communications between my NFC Reader and this tag with the Proxmark (see below)

What can be the problem? How can I solve it?

**NXP TagInfo**

Tag A
```** TagInfo scan (version 4.23) 2018-02-05 15:36:07 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
NXP Semiconductors

# IC type:
MIFARE Ultralight C (MF0ICU2)

-- NDEF ------------------------------

# No NDEF data storage present:
Maximum NDEF storage size after format: 137 bytes

-- EXTRA ------------------------------

# Memory size:
192 bytes total memory
* 48 pages, with 4 bytes per page
* 144 bytes user memory (36 pages)

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 14443-3 (Type A) compatible
ISO/IEC 14443-2 (Type A) compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcA, android.nfc.tech.MifareUltralight, android.nfc.tech.NdefFormatable]
* Maximum transceive length: 253 bytes
[... 210 lines cut ...]
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       2676 | Tag | 04                                                              |     |
       7168 |       8160 | Rdr | 52                                                              |     | WUPA
      14080 |      15072 | Rdr | 52                                                              |     | WUPA
      31376 |      32432 | Rdr | 26                                                              |     | REQA
      33620 |      34004 | Tag |00!                                                              |     |
      40064 |      41120 | Rdr | 26                                                              |     | REQA
      53936 |      54992 | Rdr | 26                                                              |     | REQA
      56180 |      56628 | Tag | 04                                                              |     |
     380544 |     381536 | Rdr | 52                                                              |     | WUPA
     382788 |     385156 | Tag | 44  00                                                          |     |
    1640784 |    1641776 | Rdr | 52                                                              |     | WUPA
    1643028 |    1645396 | Tag | 44  00                                                          |     |
    1656496 |    1658960 | Rdr | 93  20                                                          |     | ANTICOLL
    1660148 |    1665972 | Tag | 88  04  16  56  cc                                              |     |
    1674800 |    1685264 | Rdr | 93  70  88  04  16  56  cc  0e  a3                              |  ok | SELECT_UID
    1686516 |    1689460 | Tag | 04  da  07                                                      |     |
    1699392 |    1701856 | Rdr | 95  20                                                          |     | ANTICOLL-2
    1703044 |    1708932 | Tag | 32  93  4f  80  6e                                              |     |
    1717488 |    1728016 | Rdr | 95  70  32  93  4f  80  6e  0f  c2                              |  ok | ANTICOLL-2
    1729204 |    1731636 | Tag | 00  fe                                                          |     |
    1739168 |    1743936 | Rdr | 50  00  57  cd                                                  |  ok | HALT
```

iceman1001, 2018-02-05:
Output from 

- **hf mfu info**
- **hf mfu dump**

jpclair, 2018-02-05:
Hi,

**hf mfu info**

Tag A
```
--- Tag Information ---------
-------------------------------------------------------------
      TYPE : MIFARE Ultralight C (MF0ULC)
       UID : 04 35 66 12 00 15 89
    UID[0] : 04, NXP Semiconductors Germany
      BCC0 : DF, Ok
      BCC1 : 8E, Ok
  Internal : 48, default
      Lock : 00 00  - 0000000000000000
OneTimePad : 00 00 00 00  - 00000000000000000000000000000000

Trying some default 3des keys
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
#db# Cmd Error: 00
#db# Authentication failed
```

Tag B
```
iso14443a card select failed
```

**hf mfu dump**

Tag A
```
TYPE : MIFARE Ultralight C (MF0ULC)
Reading tag memory...
#db# Cmd Error: 00
#db# Read block 4 error

 Block#  | Data        |lck| Ascii
---------+-------------+---+------
  0/0x00 | 04 35 66 df |   |
  1/0x01 | 12 00 15 89 |   |
  2/0x02 | 8e 48 00 00 |   |
  3/0x03 | 00 00 00 00 | 0 | ....
---------------------------------
Dumped 4 pages, wrote 16 bytes to 04356612001589.bin
```

Tag B
```
iso14443a card select failed
```

iceman1001, 2018-02-05:
standard questios,   
do you use latest source for github?
have you tried different position between tag and antenna?

output for:
hw version
hw status

uzlonewolf, 2018-02-17:
You sure that 2nd snoop is Tag B?  The UID is completely different then what NXP TagInfo is reporting.

jpclair, 2018-02-17:
I think you are right. I have multiple cards which have the same problem. So maybe, I tried to read one with TagInfo and the other one with the PM3.
I will check with Tag Info the good one when I'll go back from abroad.

Sorry about that

jpclair, 2018-02-25:
Hi @uzlonewolf The report of the scan with NXP TagInfo for the tag B has been updated.
Sorry for my mistake

marshmellow42, 2018-04-18:
Sorry for being late to the party, but what does your `hw tune` look like?

Also is the physical size of tag B different?  Like a small keyfob or something?

jpclair, 2018-07-10:
Hi @marshmellow42 after further investigation, the tag with an anormal behavior is not a chip manufactured by NXP but a counterfeint. That can explain that the tag communicates so weirdly.

So, this issue could be closed. Thanks for your help

closed 2018-07-10 by jpclair