Proxmark/proxmark3 #563: "Untrusted kexts are not allowed"

faul8, 2018-02-05:
I am following the Wiki for installation on MacOS, running 10.13.3.
At step 11, the output I get from "sudo make install_kext" is:
`mkdir -p /System/Library/Extensions/Proxmark3.kext/Contents
cp Info.plist /System/Library/Extensions/Proxmark3.kext/Contents
chown -R root:wheel /System/Library/Extensions/Proxmark3.kext
chmod 755 /System/Library/Extensions/Proxmark3.kext /System/Library/Extensions/Proxmark3.kext/Contents
chmod 644 /System/Library/Extensions/Proxmark3.kext/Contents/Info.plist
rm -rf /System/Library/Caches/com.apple.kext.caches
rm: /System/Library/Caches/com.apple.kext.caches/Directories/Library/Extensions/KextIdentifiers.plist.gz: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/Library/Extensions: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/Library: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/System/Library: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories/System: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Directories: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup/IOKitPersonalities_x86_64.ioplist.gz: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup/kernelcache: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup/KextPropertyValues_OSBundleHelper_x86_64.plist.gz: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup/KextPropertyValues_PGO_x86_64.plist.gz: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup/loadedkextmt.plist: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches/Startup: Operation not permitted
rm: /System/Library/Caches/com.apple.kext.caches: Operation not permitted
make: *** [install_kext] Error 1`

And the output I get from "sudo kextcache -system-caches" is:
`Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/009DE355-6641-4345-AD2C-BD9A27A06EFF.kext
Bundle (/System/Library/Extensions/Proxmark3.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/009DE355-6641-4345-AD2C-BD9A27A06EFF.kext
Unable to stage kext (/System/Library/Extensions/Proxmark3.kext) to secure location.`

How can I get around these issues?

pwpiwi, 2018-02-05:
Are you sure that you need these steps? Your PM3 came with HID bootrom? What was the result of `system_profiler SPUSBDataType` ?

faul8, 2018-02-05:
```
PM3:

              Product ID: 0x4b8f
              Vendor ID: 0x9ac4
              Version: 0.01
              Speed: Up to 12 Mb/sec
              Manufacturer: proxmark.org
              Location ID: 0xfa130000 / 6
              Current Available (mA): 500
              Current Required (mA): 150
              Extra Operating Current (mA): 0
```

pwpiwi, 2018-02-05:
I am not completely sure but the 150mA "Current Required" had been introduced recently. It should be a CDC device, not an HID device. The steps in section "Upgrading HID to CDC" should not be required.

Are there any Mac Guru'S who can tell how to definitely distinguish between the two?

faul8, 2018-02-05:
Got it working. Looks like it's CDC, but wasn't working without upgraded firmware.

closed 2018-02-05 by faul8