Proxmark/proxmark3 #564: antenna is unusable

peseta3, 2018-02-06:
Hallo Together,

do you have an Idea from this?

```
C:\Users\Joerg\Desktop\Proxmark\pm3-bin-2.4.0\pm3-bin-2.4.0\win32 (client+GUI)>proxmark3 COM6
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-294-g380e0d0-suspect 2018-01-25 15:40:39
os: master/v3.0.1-294-g380e0d0-suspect 2018-01-25 15:40:40
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 197743 bytes (38%). Free: 326545 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hw tune

Measuring antenna characteristics, please wait...
# LF antenna:  0.00 V @   125.00 kHz
# LF antenna:  0.00 V @   134.00 kHz
# LF optimal:  0.00 V @ 12000.00 kHz
# HF antenna:  0.00 V @    13.56 MHz
# Your LF antenna is unusable.
# Your HF antenna is unusable.
proxmark3>
```

iceman1001, 2018-02-06:
You seem to mix firmware and client for different builds.  
The recommendation is to always run client / firmware from same build.

peseta3, 2018-02-06:
@iceman1001 
Thank you, but on witch did you can see this?

iceman1001, 2018-02-06:
Your top four rows..

> pm3-bin-2.4.0\pm3-bin-2.4.0\win32 (client+GUI)>proxmark3 COM6
>
> bootrom: master/v3.0.1
> os: master/v3.0.1

peseta3, 2018-02-06:
Have I work to this pm3-bin-2.5.0 ?

peseta3, 2018-02-06:
No thats not the reason.
It must be any other thing.

iceman1001, 2018-02-06:
no.. you would use  v3.0.1 client if you use v3.0.1 firmware

peseta3, 2018-02-07:
hmm I understand this not realy.
But on the Desktop PC VMWare it works but on the Notebook with all the same didn`t work.

peseta3, 2018-02-08:
Ah Ok now I check it.

This was the realy important link for me
[https://drive.google.com/drive/folders/0B03wIb-JZ6VQbFRmWnlKOHVKM0k](url)
20180206

Thank you Iceman for your help.

closed 2018-02-08 by iceman1001

peseta3, 2018-02-08:
sorry Iceman.......

I tested in the moment, in the Kali VM doesn`t work , in The windows Host it run.
he has Problems with the Driver and USB Port

iceman1001, 2018-02-08:
sorry,  can't help you,  your virtual os setup is different story.

peseta3, 2018-02-08:
yes i think so, too!