Proxmark/proxmark3 #567: hf 14a snoop don't work.

maxben14, 2018-02-07:
I try sniff between my device and android.
proxmark3> hf 14a snoop
#db# cancelled by button
#db# COMMAND FINISHED
#db# maxDataLen=3, Uart.state=0, Uart.len=0
#db# traceLen=3592, Uart.output[0]=00000022
proxmark3> hf list 14a
Recorded Activity (TraceLen = 3592 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |       2368 | Tag | 04  00                                                          |     |
      14064 |      19952 | Tag | 01  02  03  04  04                                              |     |
      35696 |      39216 | Tag | 08  b6  dd                                                      |     |
     337536 |     339904 | Tag | 04  00                                                          |     |
     360672 |     364192 | Tag | 08  b6  dd                                                      |     |
     496448 |     498816 | Tag | 04  00                                                          |     |
     519584 |     523104 | Tag | 08  b6  dd                                                      |     |
     624960 |     627328 | Tag | 04  00                                                          |     |
     648112 |     651632 | Tag | 08  b6  dd                                                      |     |
     712688 |     717360 | Tag | 00  00  00  00                                                  |     |
     737840 |     742512 | Tag | f0  e7  2c  22                                                  |     |
     839536 |     841904 | Tag | 04  00                                                          |     |
     862688 |     866208 | Tag | 08  b6  dd                                                      |     |
     964320 |     966688 | Tag | 04  00                                                          |     |
     987488 |     991008 | Tag | 08  b6  dd                                                      |     |
    1090432 |    1092800 | Tag | 04  00                                                          |     |
    1113584 |    1117104 | Tag | 08  b6  dd                                                      |     |
    1221408 |    1223776 | Tag | 04  00                                                          |     |
    1244560 |    1248080 | Tag | 08  b6  dd                                                      |     |
    1346208 |    1348576 | Tag | 04  00                                                          |     |
    1369344 |    1372864 | Tag | 08  b6  dd                                                      |     |
    1481056 |    1483424 | Tag | 04  00                                                          |     |
    1504192 |    1507712 | Tag | 08  b6  dd                                                      |     |
    1603408 |    1608080 | Tag | 00  00  00  00                                                  |     |
    1628624 |    1633360 | Tag | 4d  47  ee  bc                                                  |     |
    1726592 |    1728960 | Tag | 04  00                                                          |     |
    1749728 |    1753248 | Tag | 08  b6  dd                                                      |     |
    1898512 |    1900880 | Tag | 04  00                                                          |     |
    1921648 |    1925168 | Tag | 08  b6  dd                                                      |     |
    2069808 |    2072176 | Tag | 04  00                                                          |     |
    2092960 |    2096480 | Tag | 08  b6  dd                                                      |     |
    2455072 |    2457440 | Tag | 04  00                                                          |     |
    2478224 |    2481744 | Tag | 08  b6  dd                                                      |     |
    4325504 |    4327872 | Tag | 04  00                                                          |     |
    4348688 |    4352208 | Tag | 08  b6  dd                                                      |     |
    6101980 |    6106748 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    6151200 |    6153568 | Tag | 04  00                                                          |     |
    6174352 |    6177872 | Tag | 08  b6  dd                                                      |     |
    7981980 |    7982972 | Rdr | 52                                                              |     | WUPA
    7984224 |    7986592 | Tag | 04  00                                                          |     |
    8007392 |    8010912 | Tag | 08  b6  dd                                                      |     |
    9814560 |    9816928 | Tag | 04  00                                                          |     |
    9837728 |    9841248 | Tag | 08  b6  dd                                                      |     |
   11686912 |   11689280 | Tag | 04  00                                                          |     |
   11710080 |   11713600 | Tag | 08  b6  dd                                                      |     |
   13622928 |   13625296 | Tag | 04  00                                                          |     |
   13646096 |   13649616 | Tag | 08  b6  dd                                                      |     |
   15473472 |   15475840 | Tag | 04  00                                                          |     |
   15496640 |   15500160 | Tag | 08  b6  dd                                                      |     |
   17320028 |   17321020 | Rdr | 52                                                              |     | WUPA
   17322272 |   17324640 | Tag | 04  00                                                          |     |
   17345456 |   17348976 | Tag | 08  b6  dd                                                      |     |
   19223776 |   19226144 | Tag | 04  00                                                          |     |
   19246960 |   19250480 | Tag | 08  b6  dd                                                      |     |
   21149248 |   21151616 | Tag | 04  00                                                          |     |
[... 221 lines cut ...]
  190092768 |  190095136 | Tag | 04  00                                                          |     |
  190115968 |  190119488 | Tag | 08  b6  dd                                                      |     |
  190180832 |  190185504 | Tag | 00  00  00  00                                                  |     |
  190206064 |  190210736 | Tag | c9  ec  35  8d                                                  |     |
  192330160 |  192332528 | Tag | 04  00                                                          |     |
  192353360 |  192356880 | Tag | 08  b6  dd                                                      |     |
  194185312 |  194187680 | Tag | 04  00                                                          |     |
  194208512 |  194212032 | Tag | 08  b6  dd                                                      |     |
  196060220 |  196061212 | Rdr | 52                                                              |     | WUPA
  196062464 |  196064832 | Tag | 04  00                                                          |     |
  196085664 |  196089184 | Tag | 08  b6  dd                                                      |     |
  197965440 |  197967808 | Tag | 04  00                                                          |     |
  197988640 |  197992160 | Tag | 08  b6  dd                                                      |     |
  199747772 |  199748188 | Rdr | 01                                                              |     | ?
  199748412 |  199749276 | Rdr |22!                                                              |     | ?
  199794544 |  199796912 | Tag | 04  00                                                          |     |
  199817744 |  199821264 | Tag | 08  b6  dd                                                      |     |
  201621184 |  201623552 | Tag | 04  00                                                          |     |
[... cut ...]
  203450896 |  203453264 | Tag | 04  00                                                          |     |
  203474096 |  203477616 | Tag | 08  b6  dd                                                      |     |
  205277440 |  205279808 | Tag | 04  00                                                          |     |
  205300640 |  205304160 | Tag | 08  b6  dd                                                      |     |
  207108976 |  207111344 | Tag | 04  00                                                          |     |
  207132192 |  207135712 | Tag | 08  b6  dd

pwpiwi, 2018-02-07:
Would you please ask questions not related to the source code in the forum instead of the code repository.

closed 2018-02-07 by pwpiwi

maxben14, 2018-02-07:
@pwpiwi , this is not a question, it's a bug report in the proxmark sniffer. I do not understand why the proxmark often does not want to capture data from the reader and captures only from the card. Can you fix this bug in the proxmark?

marshmellow42, 2018-02-07:
your snoop sometimes captured the reader.  it all depends on your antenna and the positioning to capture the best snoop.  is it possible to improve the sensitivity with code, with the pm3's cpu realtime? not likely..  but i don't know.  but overall i wouldn't consider this a bug, more of a requested enhancement...

maxben14, 2018-02-07:
@marshmellow42 , problem that proxmark often don't capture answer from my Android. How increase timing waiting in proxmark software ?

marshmellow42, 2018-02-07:
android phones have terrible interference and signal strength.  i'm impressed we can read anything at all from them...