Proxmark/proxmark3 #573: Sending raw data from Graphbuffer

yifanz5356, 2018-02-18:
Hi all,
Does the Proxmark3 have a command to send the data in the graphbuffer out directly using the 125kHz antenna (Act more like a reader than a tag)?

Background:
We used lf snoop to capture the 125kHz signal from the car's keyless entry system. And we want to replay the signal using another proxmark3. So after we did the lf snoop, we used data save to save the data to a text file. Then we used load data to fill the another Proxmark3's graphbuffer with the same data.

We are trying to find an appropriate command to replay the signal. lf sim will not work because it only transmit when there's a reader signal exist, we want the proxmark3 actively sends the signal.

This is the hardware we used:
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-307-ga37725f-dirty-suspect 2018-02-10 00:48:56
os: master/v3.0.1-307-ga37725f-dirty-suspect 2018-02-10 00:49:41
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 196768 bytes (38%). Free: 327520 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

marshmellow42, 2018-02-18:
This is a question better asked on the forum.  But the cmd you are looking for is lf cmdread.  I recently added a bitbang mode so be sure to update to the latest firmware

marshmellow42, 2018-02-18:
But no you can't just use the graphbuffer.

yifanz5356, 2018-02-18:
Thank you for your help! I actually did not find how to post the new question on the forum. 
In order to use lf cmdread to send our data located in the graphbuffer, we need to demodulate the data to binary from in order to be sent, right? 
What should be the '0' and '1' period in the cmdread and how to get the binary form from the data in the graphbuffer? 
Sorry for the lengthy question, but I am very new using the proxmark3.

iceman1001, 2018-02-18:
post thread on forum as @marshmellow42  says.   this is not an Github issue.

closed 2018-02-18 by iceman1001