Proxmark/proxmark3 #574: Issue with parity bits and lf_bulk_program.lua

Tom5ive, 2018-02-20:
Hi,

There is an issue with the lua wherein the parity is around the wrong way. I can confirm this issue after programming 200+ cards that had issues with certain systems because they expected the parity to be correct. 

The fix is to change 

high = evenparity(string.sub(stream,0,12)) and 1 or 0
	low =  not evenparity(string.sub(stream,13)) and 1 or 0

To:

high = not evenparity(string.sub(stream,0,12)) and 1 or 0
	low =  evenparity(string.sub(stream,13)) and 1 or 0

I would love to commit this to master.

Thanks,

iceman1001, 2018-02-22:
yes,   verifed behavior,  why don't you make a PR for it?

Tom5ive, 2018-02-22:
Done cheers!

closed 2018-02-22 by iceman1001

Tom5ive, 2018-03-22:
Hi Guys, there is still an issue with this code. In our organisation we have discovered that once you go above  CARD 8000 there are again issues with parity bits. Not sure what the fix is but I will continue digging.