Proxmark/proxmark3 #575: Build a diagnostic test to ensure PM3 is getting enough power

micolous, 2018-02-21:
On testing on Android (using mainline firmware), I notice some phones are skimpy on supplying power to PM3, which typically impacts use with `hf` commands.  The symptom is that commands appear to sort-of work, but the PM3 will see incorrect commands coming in the CDC-ACM interface, or that the entire PM3 will lock up.

I suspect this also impacts other platforms with limited USB current.

I'm currently able to mostly work around this with a USB-Y cable (with two USB-A connectors), plugging one into the device, and one into a battery pack.

I'd like it if there was some sort of firmware-level test that could thrash the Proxmark3's MCU, while also running the HF radio transmitting at the same time.  This test shouldn't need any card to be in the field, or even have a HF antenna plugged in.  It should also test to see if it is getting good results.

This could also help to identify manufacturing defects.

This should help to identify the difference between "this is an actual bug in the firmware" or "the PM3 is glitching because of insufficient power".

pwpiwi, 2018-02-22:
Maximum current I had measured when in hf operation was 120mA. This is without a battery and you definitely don't want to use your Android phone as a PowerPack to charge your PM3. Are there any data sheets indicating how much current can be drawn from smartphone's USB connectors?

merlokk, 2018-02-22:
Here is no datasheets( minimum 500ma. And in enumeration phase device send needed power to Host and Host enables it or not. And I dont remember where I see, but in enumeration phase device should not get more than 500ma current

pwpiwi, 2018-02-22:
I would be astonished if a smartphone would be able to deliver the standard 500mA. Even during heavy use this would probably be far more than its own needs?

micolous, 2018-02-24:
Enumeration phase [requests 150mA](https://github.com/Proxmark/proxmark3/blob/master/common/usb_cdc.c#L81), which seems pretty reasonable to me.

My PM3 doesn't have an internal battery pack, and I was using the Y-cable to supply extra power to the PM3.  When operating, my USB current meter was reporting that the phone wasn't supplying any power at that point, and the USB battery pack was reporting ~250mA.  I suspect in this case the phone was also trying to charge, and if I disconnect `Vbus` on the phone-side, it may give me some better numbers... I'll report back later about this.

_USB OTG_ (for phones with a Micro-A/B connector) [specifies 100mA](http://www.usb.org/developers/onthego/OTG_Compliance_DocumentRC1.doc) (checklist 2.10, Bus Power Consumption).

_USB-C_ has multiple power levels it can do, and I haven't been able to find a definitive spec on what even a single phone model does.  My presumption is that phones won't do the 12v or 20v power levels (used by laptops), and may even limit themselves to 100mA when there's a USB-C to A dongle.  Though USB-C phones also let you charge another device over USB-C, they're using different USB-C modes.

The concern I have is being able to do a quick test without needing special hardware (other than PM3).  This would also identify power consumption variation between parts/devices by different manufacturers.