Proxmark/proxmark3 #582: fpga image version

galenzhao, 2018-03-13:
Hi，
I saw the code was updated 4 month ago in fpga folder,
but when I build with the newest code 
it shows like 👍 

GalenZhaos-MacBook-Air:proxmark3 galenzhao$ ./client/proxmark3 /dev/cu.usbmodem1421 
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-351-g51d51c6-suspect 2018-03-13 08:33:35
os: master/v3.0.1-351-g51d51c6-suspect 2018-03-13 08:33:36
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
          
uC: AT91SAM7S256 Rev D          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 200083 bytes (76). Free: 62061 bytes (24).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hw ver

the fpga image seem likes still very old,

is that right?

iceman1001, 2018-03-13:
Seems ok to me. 

**HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59**

look at history for FPGA code:  https://github.com/Proxmark/proxmark3/commits/master/fpga/fpga_hf.bit
which shows latest commit was 24 october 2017.

pwpiwi, 2018-03-14:
... and LF FGPA code indeed didn't change since years.

closed 2018-03-14 by pwpiwi