Proxmark/proxmark3 #583: Flashing Issues: Can not flashing fpga image

linsir, 2018-03-14:
I have a PM3, which has been working fine.
```
#db# Prox/RFID mark3 RFID instrument                 
#db# bootrom: /-suspect 2015-04-02 15:12:04                 
#db# os: /-suspect 2015-04-02 15:12:11                 
#db# HF FPGA image built on 2015/03/09 at 08:41:42                 
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory   
```

But unfortunately I have a issue during firmware upgrade - it doesnt matter if I build from sources with Linux or Windows - nor if I try to flash with Linux or Windows.The bootrom flashed fine, but the fullimage can not flashed.  like this:
```
Note: 0x5-byte hole created

.Waiting for Proxmark to appear on /dev/ttyACM0 Found.

Flashing...
Writing segments for file: bins/2.0/fullimage.elf
 0x00102000..0x00131e97 [0x2fe98 / 384 blocks]................Error: Unexpected reply 0x00fe NACK (expected ACK)
       Lock Error
       Lock Bits:      0x000a
 ERROR
Error writing block 16 of 384
```
I try flash old version (848), get:
```
Waiting for Proxmark to appear...
Found proxmark on /dev/ttyACM1

Loading ELF file 'bins/848/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000b60->0x00000b60) [RWX] @0x298

Waiting for Proxmark to appear on /dev/ttyACM1. Found.

Flashing...
Writing segments for file: bins/848/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100d5f [0xb60 / 6 blocks]...... OK

Resetting hardware...
All done.

Have a nice day!
Waiting for Proxmark to appear...
Found proxmark on /dev/ttyACM1

Loading ELF file 'bins/848/fpgaimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0000a4bc->0x0000a4bc) [R  ] @0xb4

Waiting for Proxmark to appear on /dev/ttyACM1. Found.

Flashing...
Writing segments for file: bins/848/fpgaimage.elf
 0x00102000..0x0010c4bb [0xa4bc / 83 blocks]................Error: Unexpected reply 0x00fe (expected ACK)
 ERROR
Error writing block 16 of 83
Waiting for Proxmark to appear...
Found proxmark on /dev/ttyACM1

Loading ELF file 'bins/848/osimage.elf'...
Loading usable ELF segments:
1: V 0x00110000 P 0x00110000 (0x00014c82->0x00014c82) [R X] @0xb8
2: V 0x00200000 P 0x00124c88 (0x000029b8->0x000029b8) [RWX] @0x14d40
Note: Extending previous segment from 0x14c82 to 0x17640 bytes
Note: 0x6-byte hole created

Waiting for Proxmark to appear on /dev/ttyACM1. Found.

Flashing...
Writing segments for file: bins/848/osimage.elf
 0x00110000..0x0012763f [0x17640 / 188 blocks]............................................................................................................................................................................................ OK

Resetting hardware...
All done.

Have a nice day!

```
Then I connect PM3 with client get:

```
#db# Prox/RFID mark3 RFID instrument                 
#db# bootrom: svn 848 2014-02-28 07:25:33                 
#db# os: svn 848 2014-02-28 07:25:37                 
#db# FPGA image built on 2014/02/25 at 07:43:59                 
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory 
```
and it doesn't work now.
```
proxmark3> hw tune

Measuring antenna characteristics, please wait...#db# Measuring antenna characteristics, please wait...                 
......#db# Measuring complete, sending report back to host                 
          
# LF antenna:  0.00 V @   125.00 kHz          
# LF antenna:  0.00 V @   134.00 kHz          
# LF optimal:  0.00 V @ 12000.00 kHz          
# HF antenna:  0.00 V @    13.56 MHz          
# Your LF antenna is unusable.          
# Your HF antenna is unusable.          

```
I try to search about the problem, can find any useful infomation but this

<http://www.proxmark.org/forum/viewtopic.php?id=1509>

Did you have a solution for this? 

Thanks

pwpiwi, 2018-03-14:
Well, the flashing seemed to be successful. You need to downgrade your client software to SVN 848 as well to get a working client/firmware pair (although this version is even older than your original one).

Please continue in the forum. This is not the correct place to ask handling questions.

closed 2018-03-14 by pwpiwi

linsir, 2018-03-14:
@pwpiwi  I try that, and it doesn't work .
```
# LF antenna:  0.00 V @   125.00 kHz          
# LF antenna:  0.00 V @   134.00 kHz          
# LF optimal:  0.00 V @ 12000.00 kHz          
# HF antenna:  0.00 V @    13.56 MHz          
# Your LF antenna is unusable.          
# Your HF antenna is unusable. 
```

linsir, 2018-03-14:
Okay.