Proxmark/proxmark3 #586: More Issue with parity bits and lf_bulk_program.lua

Tom5ive, 2018-03-22:
Hi Guys, there is still an issue with this code. In our organisation we have discovered that once you go above CARD 8000 there are again issues with parity bits - even after the previous fix. Not sure what the fix is but I will continue working to find out.

pwpiwi, 2018-03-22:
32Bits issue? What is your facility code?

pwpiwi, 2018-03-22:
```
	high = evenparity(string.sub(stream,0,12)) and 1 or 0
```
The first character in a LUA string has index 1, the last has index -1. I have no clue what an index of 0 means?

Tom5ive, 2018-03-23:
We use a 26bit standard. Facility code is 10.

Tom5ive, 2018-03-23:
Just an update. I just dis some testing and generated a heap of cards ranging from CC 1000 -> CC 12000 and the issue begins at card 4096 - the parity flips.

pwpiwi, 2018-03-23:
In
```
stream=toBits(id,26)
```
change to 24.

Tom5ive, 2018-03-25:
Ok I will give this a go. Why would that help though? We are definitely using wg26.

pwpiwi, 2018-03-25:
The wg26 includes two paritiy bits. `stream` is without parity bits.

Tom5ive, 2018-03-25:
After making this change the issue still presents. When I write tag 4095 the code still writes 002006141ffe when it should be writing 002006141fff. Its one hex value off.

pwpiwi, 2018-03-27:
In fact it should be writing 00200**4**141fff. Looks like Evenparity and Oddparity are mixed up. We need to change to
```
	high = evenparity(string.sub(stream,1,12)) and 1 or 0
	low =  not evenparity(string.sub(stream,13)) and 1 or 0
```
in addition to the 24 Bit fix above.

Tom5ive, 2018-03-29:
Ok I will give this a try, iceman just implemented my fix for parity earlier on with commit #576 which fixed issues with parity earlier on. Maybe swapping back in combination with the 24bit fix will do it.

Tom5ive, 2018-04-03:
Confirmed working.

closed 2018-04-10 by iceman1001