Proxmark/proxmark3 #594: lf search exits early with no error

patrickp2, 2018-04-17:
When I run lf search, or lf search u, it exits out of the search early with no error.

Picture of issue
http://www.evernote.com/shard/s509/sh/eaea4607-d6b9-47ce-b866-8a104164384e/40eef20c64ea26e121f480b1ca479940

Here's my hw version 
http://www.evernote.com/shard/s509/sh/71e392de-3d9b-49cf-98a8-cbf96118c147/119e8d0c5fdbf57de05d775c7ade45a1

Using windows 8 and ELECHOUSE PROXMARK3 DEV KIT 2. Please advise.

marshmellow42, 2018-04-18:
Can you post the results of
`hw tune`
And 
`lf read`
`data save trace.pm3`

iceman1001, 2018-04-18:
- what size of the tag ( cardsize,  keyfob?) picture of it?
- what antenna are you using?  
- which pm3 device do you have?

pwpiwi, 2018-04-18:
* your Operating System Version?

patrickp2, 2018-04-20:
hw tune 
http://www.evernote.com/shard/s509/sh/4355249c-0c83-4277-a184-39ebd564d61c/08cebbeaa51112716aebbf774232b75c

Is the default mode to exit out if it's not picking up a card? I assumed it would at least say "card not found". It is entirely possible the card im scanning is UHF not LF. (It's not HF).

what size of the tag ( cardsize, keyfob?) picture of it? Ski pass
what antenna are you using? LF
which pm3 device do you have? Elechouse proxmark3 dev kit 2

lf read 
and trace brings up lots of -13's

marshmellow42, 2018-04-20:
That would indicate your tag was not lf or is a lf tag that requires a wakeup cmd that is unknown.

iceman1001, 2018-04-20:
Isn't skipasses usually 14B / 15 ?  That would be a HF tag.

marshmellow42, 2018-04-20:
Ski passes are generally hf or uhf not lf.  But , I haven't seen every ski resort...

jacopo-j, 2018-04-20:
I can confirm that skipasses are ISO 15693 as they are *vicinity* cards, not *proximity* cards. You usually keep them in your pocket while you access the gate.

patrickp2, 2018-04-20:
I thought that the software was bugging as it didn't show completion of the lf search. I also have other weird issues like when I try to dump a mifare card (hf mf dump), I get "Could not find file dumpkeys.bin"
http://www.evernote.com/shard/s509/sh/d2062a2a-e309-4384-b0d7-e9735fe0d3bb/0646e709e5f0c43eb8db92f8a08c52d4

Some older ski passes are ISO 15963 (I wrote emulation code on the chameleon mini). This one is not ISO 15963 (it's a newer one, skidata also announced moving to UHF). It does not respond in the 13.56 MHz range. I'm assuming its in the 860-960MHz range.