Proxmark/proxmark3 #599: hf mf chk, possible bug

TomHarkness, 2018-05-04:
Hi team,

I think that I found a bug in the 'hf mf chk' function. See below:

proxmark3> hf mf chk *1 482eb20350e0 2a2c13cc242a
chk key[ 0] 2a2c13cc242a

To cancel this operation press the button on the proxmark...
--o
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|000|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|001|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |
|002|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|003|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|004|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|005|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|006|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|007|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|008|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|009|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|010|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|011|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|012|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|013|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|014|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|015|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|---|----------------|---|----------------|---|

proxmark3> hf mf chk *1 ? 482eb20350e0 2a2c13cc242a
chk key[ 0] 482eb20350e0
chk key[ 1] 2a2c13cc242a

To cancel this operation press the button on the proxmark...
--o
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|000|  482eb20350e0  | 1 |  ffffffffffff  | 0 |
|001|  2a2c13cc242a  | 1 |  ffffffffffff  | 0 |
|002|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|003|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|004|  482eb20350e0  | 1 |  ffffffffffff  | 0 |
|005|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|006|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|007|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|008|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|009|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|010|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|011|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|012|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|013|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|014|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|015|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|---|----------------|---|----------------|---|

proxmark3>

I assume that the command 'hf mf chk *1 <some_keys_here>' should check the keys specified. However it only seems to work when I have the '?' argument. The output from 'hf mf chk h' is not clear on what the '?' argument actually does.

pwpiwi, 2018-05-09:
Your assumptions are wrong. `hf mf chk h` tells you
```
<key type (A/B/?)>
```
i.e. the `?` means to test both A and B keys.
```
[<key (12 hex symbols)>] 
```
i.e. you can provide one additional key only. If you need to add more keys, use a dictionary file. Check the existing one (default_keys.dic) for the format.

closed 2018-05-09 by pwpiwi

TomHarkness, 2018-05-15:
Oh, I see now! Thanks!