Proxmark/proxmark3 #601: Firmware upgrade issuse (pm3 V3.0)

user203, 2018-05-07:
I received my ELECHPUSE Proxmark3 Easy lately. I plugged it on Ubuntu and it works fine. When I flashed firmware I got an successful update message.
> **sudo ./flasher /dev/ttyACM0 ../armsrc/obj/fullimage.elf**
>Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002d3a0->0x0002d3a0) [R X] @0x94
1: V 0x00200000 P 0x0012f3a0 (0x00001a40->0x00001a40) [RW ] @0x2d434
Note: Extending previous segment from 0x2d3a0 to 0x2ede0 bytes
Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/ttyACM0....... Found.
Flashing...
Writing segments for file: ../armsrc/obj/fullimage.elf
 0x00102000..0x00130ddf [0x2ede0 / 375 blocks]....................................................................................................................................................................................................................................................................................................................................................................................... OK
Resetting hardware...
All done.
Have a nice day!

After that, my device has stopped working and is not recognized as USB. I have noticed that the 2 LED (A and C) is always ON. I was upgrading only firmware without Bootloader because it has recognized as CDC device.

Firmware information before upgrade firmware:
>**./proxmark3 /dev/ttyACM0**
>#db# Prox/RFID mark3 RFID instrument          
#db# bootrom: /-suspect 2016-11-09 00:59:56          
#db# os: /-suspect 2016-12-08 12:45:38          
#db# HF FPGA image built on 2015/03/09 at 08:41:42          
#db#   Modify by Willok(willok@163.com)
#db#      proxmark3.taobao.com
Prox/RFID mark3 RFID instrument          
uC: AT91SAM7S256 Rev D          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 0 bytes ( 0%). Free: 262144 bytes (100%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory

I'm wondering what goes wrong. Any ideas on what is causing this and what I could do to try and fix it?  

Thank you for your help.

iceman1001, 2018-05-07:
...its a firmware made by a chinese dude called @willok   We can at the best answer questions to official firmware.

pwpiwi, 2018-05-09:
There is a small chance that the bootloader is based on the old memory layout (changed in March 2015) and therefore jumps to a wrong memory location. If this is the case, updating the bootloader would help. Good luck!

mondjunge, 2018-05-10:
I got the exact same problem, but I did not save any console logs.

mondjunge, 2018-05-10:
I got it running:
clone latest proxmark3 repo (resp. pull the latest changes). Build.
I than ran the flasher like described in wiki (proxmark still unplugged!)
When the flasher waits for proxmark3 to appear on ttyACM0, HOLD DOWN button (on proxmark) and plug it in. HOLD DOWN Button until flasher is completed.
DONE.
Your proxmark3 should get recognised now, only the power LED will be on.

mondjunge, 2018-05-10:
For the record: I "bricked" the device with Kali, since the Kali wiki states to only flash the firmware (with CDC interface), while the ubuntu wiki says to flash both at once: bootrom and firmware.

pwpiwi, 2018-05-10:
Usualy it is wise to flash the firmware only. If anything goes wrong you would still have a working bootrom for another try.

There had been very few changes in the last years which required flashing the bootrom.

iceman1001, 2018-05-10:
usually it is as @pwpiwi  when you are on the same firmware source.  Like trying to update offical firmware and you already have offical firmware on your device.

When swapping between different firmware sources,   like  going from offical -> iceman,   or unknown (like willok) to official,   I found the best way for me was to run flasher with bootrom/fullimage in the same command.

**_flashers com3 -b bootrom.elf fullimage.elf_**

user203, 2018-05-10:
Thank you very much for your help. An not full upgrade has proved to be the right cause.
I flashed bootrom with fullimage according to the [flashing procedure](https://github.com/Proxmark/proxmark3/wiki/flashing) and everything works fine again.
>**sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf**
>Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000cd4->0x00000cd4) [R X] @0x298
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002d3a0->0x0002d3a0) [R X] @0x94
1: V 0x00200000 P 0x0012f3a0 (0x00001a40->0x00001a40) [RW ] @0x2d434
Note: Extending previous segment from 0x2d3a0 to 0x2ede0 bytes
Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Flashing...
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100ed3 [0xcd4 / 7 blocks]....... OK
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x00130ddf [0x2ede0 / 375 blocks]....................................................................................................................................................................................................................................................................................................................................................................................... OK
Resetting hardware...
All done.
Have a nice day!

Now the device is recognized as J. Westhues ProxMark-3 RFID Instrument
>**lsusb**
Bus 001 Device 011: ID 9ac4:4b8f J. Westhues ProxMark-3 RFID Instrument

before
>Bus 001 Device 006: ID 2d2d:504d

Regards! I'm going back to the joy.

closed 2018-05-11 by pwpiwi

DEFCON3187, 2019-05-10:
Hi everyone!!

Im a complete beginner using this tech, I just purchased the Proxmark 3 V3.0 I was wondering if someone could advise me of the correct location of the installation files. What is the latest software that will work on this device?

Many thanks.

pwpiwi, 2019-05-10:
If you came here, you already found the source code for the latest software. If you are on Windows, you can use one of the precompiled versions (http://www.proxmark.org/forum/viewtopic.php?id=3975).

hom1476, 2021-12-24:
> Thank you very much for your help. An not full upgrade has proved to be the right cause. I flashed bootrom with fullimage according to the [flashing [...]

Execute me, Would you please tell me the process with simple method.
I tried for couple but still didn't work.  
Many thanks