Proxmark/proxmark3 #618: lf indala clone not working

dygos2, 2018-06-07:
I've seen around some people trying to make the indala clone to work, but my version/firmware that i have cannot write propertly.
Here is my version:

Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-298-g298e1a2-suspect 2018-04-11 17:40:28
os: master/-suspect 2018-03-17 20:20:11
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

Any magic i should do?
Thanks

marshmellow42, 2018-06-07:
What did you try?
What tag are you cloning to?

I haven't seen any real issues with the cmd, just how ppl try to use it.

dygos2, 2018-06-07:
Thanks for your quick answer my friend!
This is the original Tag that im reading...

proxmark3> lf search

NOTE: some demods output possible binary  if it finds something that looks like a tag          False Positives ARE possible          Checking for known tags:          BitLen: 64          Indala UID=1010000000000000000000000000000010110011001000111010000000110001 (a0000000b323a031)          Valid Indala ID Found!          
==============================
and... i'm adding the new tag and trying -> lf indala clone a0000000b323a031
then i do lf search again and nothing seems to copy.
Rodrigo

marshmellow42, 2018-06-07:
What tag are you cloning to?

dygos2, 2018-06-07:
into a T5577 125Khz

marshmellow42, 2018-06-07:
What form factor?  Card, keyfob, glass injectable?

dygos2, 2018-06-07:
Exactly this guy over here.
https://www.aliexpress.com/item/50pcs-Amtel-T5577-key-fob-T5577-key-125Khz-rewritable-RFID-Proximity-ID-Token/32828167791.html?spm=a2g0s.9042311.0.0.3eef4c4dGjaZDr

marshmellow42, 2018-06-07:
Key fob t55x7 often require a custom small antenna for the pm3 in order for the fob to hear the special  write cmd from the pm3.

What antenna are you using?

dygos2, 2018-06-07:
sorry @marshmellow42 but i never used any other antenna, besides the sensor that the PM3 already comes with. I never had any issues reading and writing other tags, and indala can be read no problem, my problem is the write function. So... im assuming that indala writing, needs to have a different antena rather the one that already comes with the device? I'm months using the lf hid clone XXX and never had any issues with the antenna.

lohcm88, 2018-06-07:
I have done a lot of cloning on this type of fobs... probably u got a bad tag.  Try using another to check.  Or u can using the command lf sea u , if the tag is bad, it won't return any binaries after the command.

marshmellow42, 2018-06-07:
Depends on the device.  There are many different pm3 packages now.  Some have better antennas, and others mount the antenna so close to the pm3 pcb it interferes with the antenna and some lf functions.  

These are just possibilites I've seen.  Certainly could be a bad tag, or a locked tag.  
If you can lf t5 detect and lf t5 dump and get valid responses from your t55xx tag then the tag and antenna are fine.

dygos2, 2018-06-14:
whats up guys!?
Im back on my indala studies here...
Ok so.. this is what i've got so far...
My PM3 does recognize the indala tag (original).. here is the answer ive got...
============================
lf searchNOTE: some demods output possible binary  if it finds something that looks like a tag          False Positives ARE possible          Checking for known tags:          BitLen: 64          Indala UID= 1010000000000000000000000000000010110011001000111010000000110001(a0000000b323a031)          Valid Indala ID Found!          
----------------------------------------
after that.. ive tryed...
lf indala clone 1010000000000000000000000000000010110011001000111010000000110001
tested 'lf sea' and nothing came back
then  tryed
lf indala clone a0000000b323a031
'lf sea' and nothing as well...
when i clone this new tag with -> lf hid clone XXXXX , then lf sea answers fine!
also.. after if hid clone, it shows -> Valid T55xx Chip Found .. not sure if that helps ...

Still looking for answers here... 
Also... didnt know thats lots of differents PM3 on the market... and reference for a legit/good one to buy? 
Thanks

dygos2, 2018-06-14:
Another maybe useful information to add...
im trying to write blocks like lf t55xx wr b 0 d 00148040
but after that... lf t55xx dump doesnt show that nothing is there...

marshmellow42, 2018-07-30:
i've done a bit of testing on the latest firmware and can only find that tag position on the antenna is sometimes very critical (depending on antenna strength and tag size).  
i can repeatedly have the `lf indala clone xxxxxx` cmd work as expected, if the tag is placed such that it hears the t55xx cmds from the pm3.  

for small tags: position and antenna strength are very important.  usually centering the tag on the antenna and place the tag directly on the antenna works best.

there are also reports of some pm3s with their builtin antenna are weak when transmitting, but i don't have the hardware in question to test.  my 3 pm3's work fine.

dygos2, 2018-07-30:
Hmmm that’s interesting...
Can you please send me a link of the exactly pm3 you bought and I can buy it and test it? Some from eBay by any chance?
Thanks,
Rodrigo

dygos2, 2018-07-30:
I guess I’m using a pm3 that’s not really legitimate! I bought it from eBay and I guess was a fake one? 
What’s the source to buy a good one?
Rodrigo

dygos2, 2018-08-01:
> there are a number of sources, most buy from a source relative to their location. I've 2 original boards from RyscCorp, it is hard to beat the tried and true original board... i've heard decent things about the rv2 board from elechouse. and there will be the new rv4 board available soon, but i haven't had time to test that board out too much yet...
> i trust iceman's site. past that check out http://www.proxmark.org/forum/viewforum.php?id=17 for other reviewers.
>
> instead of a new board you might just need to take the time to investigate if you can build a new antenna and find a way to attach it to whatever board you have.

Thanks for the info my friend! I’ll see what I can try over here!
Rodrigo

marshmellow42, 2018-08-03:
I don't believe this is a bug so I think we can close this issue.

Any arguments?

closed 2018-08-03 by marshmellow42

samturner3, 2019-06-05:
I can confirm this is a bug. It is only on this firmware.
On below (iceman fork) it works perfectly:
```
Prox/RFID mark3 RFID instrument
bootrom: iceman/master/ 2018-01-04 11:52:17
os: master/v3.0.1-213-gbc3b2f7-suspect 2017-12-07 14:03:27
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 199407 bytes (38%). Free: 324881 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
```

After reflash and on this firmware it will not write an indala tag, only read.
```
Prox/RFID mark3 RFID instrument
bootrom: /-suspect 2019-06-05 03:40:35
os: /-suspect 2019-06-05 03:40:36
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2018/09/12 at 15:18:46
SmartCard Slot: not available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 197222 bytes (38). Free: 327066 bytes (62).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
```

Same hardware.
Same tag.

I had to search everywhere for a working version where indala write actually works.

marshmellow42, 2019-06-05:
You didn't include any of the commands you tried.

What hardware version/type do you have?

Do the std t55xx write cmds work for you on both as the iceman fork has made adjustments to these?

marshmellow42, 2019-06-05:
# of bits alone won't make it work, it has to be valid data as well.  We don't provide a valid tags data as an example.