Proxmark/proxmark3 #624: Proxmark3 RDV recognition Problems

Vepox, 2018-07-04:
So I've set up the Proxmark3. Everything I wanted it to do worked, except for the Darkside Attack on MIFARE Classic (It took 40+, then I cancelled). I've read some forums and ppl suggested to update the Firmware. Said -> Done (flasher.exe com5 -b ..\firmware_win\bootrom\bootrom.elf).

Now Windows doesn't recognize the Proxmark3 anymore **at all**.

If I unplug the Proxmark, then plug it back in again holding this little Button, Windows recognizes the Proxmark for a second, then throws the disconnect sound (so its gone again).

Rn, if it just sits there connected to Windows (not showing up in device manager however), it does this clicky sound every few seconds.

Any suggestions? Im totally lost.

pwpiwi, 2018-07-04:
You have flashed the bootrom only. Try flashing the firmware as well.

Vepox, 2018-07-04:
But since the Proxmark is not showing up in device manager anymore, there is no COM Port for me to select in the update process.

pwpiwi, 2018-07-05:
You hear a disconnect sound. Therefore it was connected before. Therefore it must have been in the device manager. Keep the button pressed all the time and check the device manager again. You may need a third hand.

Vepox, 2018-07-05:
If I keep the Button pressed for a few seconds, the Proxmark will connect indeed. However, it will disconnect again after a split second. Not enough time to start updating anything.

pwpiwi, 2018-07-05:
If you can exclude hardware failures you can try the advises in the Wiki De-Bricking sections.

Vepox, 2018-07-05:
I managed to fix it, thanks for your help.

Solution was to disconnect it from the computer, hold the button, keep holding it and then reconnect it to the computer. Just pressing the Button, without powering it off didnt do the job.

Whilst holding the button all the time, I then could update the Software with the flasher.

closed 2018-07-05 by Vepox

mac3d0, 2021-11-05:
> I managed to fix it, thanks for your help.
>
> Solution was to disconnect it from the computer, hold the button, keep holding it and then reconnect it to the computer. Just pressing the Button, without [...]

could you tell me how you did it? I have the same problem but I can't solve it...

mac3d0, 2021-11-05:
could you tell me how you did it? I have the same problem but I can't solve it...

BasePointer, 2021-11-08:
> could you tell me how you did it? I have the same problem but I can't solve it...

Uhm I guess I already answered it in my previous comment.
I held down the button during the whole process. I was pressing it before even pluggin the Proxmark in. I didnt just press the button once. Thats all I did. It's been a few years tho so I cant say if anything has changed since.