Proxmark/proxmark3 #626: lf t55xx write command sometimes results in bitshifted data

ss23, 2018-07-09:
```
proxmark3> lf hid clone 20050c5d8f
Cloning tag with ID 20050c5d8f          
#db# DONE!          
proxmark3> lf t55xx read b 2
Reading Page 0:          
blk | hex data | binary          
----+----------+---------------------------------          
  2 | AAB32AD2 | 10101010101100110010101011010010          
proxmark3> lf t55xx write b 2 d AAB32AD2
Writing page 0  block: 02  data: 0xAAB32AD2           
proxmark3> lf t55xx read b 2
Reading Page 0:          
blk | hex data | binary          
----+----------+---------------------------------          
  2 | 55599569 | 01010101010110011001010101101001  
```

As the `lf t55xx write` command only accepts one block at a time, I can't test whether this is a result of simply attempting to write a single block of data at once.
From what I can see, there's no way to call `WriteT55xx()` from the client directly either, which would help narrow down the issue.

marshmellow42, 2018-07-09:
The issue is on the read cmd.  The t55x7 chips have no way to accurately sync up the start of the transmission.  So the result is for some bit rates and modulations the pm3 syncs off one bit.  Feel free to attempt to improve this.

ss23, 2018-07-09:
@marshmellow42 How does it return the 'correct' data the first time in that case? Immediately after a HID clone, the correct data is always read, no matter how many times I try, and `lf search` repeatedly detects it as a HID device. Once I've written this block, I can never get it to be detected as a HID device again.
Is there perhaps two issues here?

ss23, 2018-07-09:
Here is some more verbose output which makes me think it's not an issue with the reading - https://gist.github.com/ss23/fd74f5f0002a027a965f087d8ef09aa1
As I understand it, after the write, there should be no change to the data on the card, but it seems to trigger the issue I'm seeing 100% of the time.

marshmellow42, 2018-07-09:
AAB32AD2 is not a valid hid block.

ss23, 2018-07-09:
@marshmellow42 Are you saying that the initial read is the erroneous one (e.g. the output of `lf t55xx read b 2` after the first `lf hid clone 20050c5d8f` is incorrect?)

marshmellow42, 2018-07-09:
Yes

marshmellow42, 2018-07-09:
In fact all the reads are affected by the same bit off error

marshmellow42, 2018-07-09:
Change the distance of the tag to antenna, it usually changes the results.

ss23, 2018-07-09:
@marshmellow42 Changing the distance to the antenna hasn't made any difference, but still, I understand now. I'm able to do a t55xx write with the *correct* value and have it still work (even though the read is still off by a little).

Thank you for the help!

closed 2018-07-09 by ss23